lovyan03/LovyanGFX #507: Do you support M5stack CoreS3?? can't see it in the listed devices

cohen-sydney, 2024-01-20:
**Carefully written requests are more likely to be given priority.**
**丁寧に記述された要望は優先して対応される可能性が高くなります。**

Do you support M5stack CoreS3?? can't see it in the listed devices "How to use" the Define is not there.
also M5Paper??

### Device Name (デバイスの名称・型番等)

M5Stack CoreS3

### URL of Device Specifications document (仕様書等のURL)
https://docs.m5stack.com/en/core/CoreS3

### URL of the store where we can purchase (商品を購入できるURL)

cohen-sydney, 2024-01-20:
I used Arduino to couple this ClockSample.ino for M5CoreS3 and it errors.
/Users/andrew/Documents/Arduino/ClockSimple/ClockSample/ClockSample.ino:6:8: error: 'LGFX' does not name a type
 static LGFX lcd;
        ^~~~
/Users/andrew/Documents/Arduino/ClockSimple/ClockSample/ClockSample.ino:7:28: error: 'lcd' was not declared in this scope
 static LGFX_Sprite canvas(&lcd);       // オフスクリーン描画用バッファ
                            ^~~
/Users/andrew/Documents/Arduino/ClockSimple/ClockSample/ClockSample.ino: In function 'void setup()':
/Users/andrew/Documents/Arduino/ClockSimple/ClockSample/ClockSample.ino:27:3: error: 'lcd' was not declared in this scope
   lcd.init();
   ^~~
/Users/andrew/Documents/Arduino/ClockSimple/ClockSample/ClockSample.ino: In function 'void drawClock(uint64_t)':
/Users/andrew/Documents/Arduino/ClockSimple/ClockSample/ClockSample.ino:166:3: error: 'lcd' was not declared in this scope
   lcd.display();
   ^~~

exit status 1

Compilation error: 'LGFX' does not name a type

tobozo, 2024-01-20:
- Declare this **before** including LovyanGFX

```
#define LGFX_AUTODETECT  
```

- If using Arduino IDE, make sure "M5Stack CoreS3" is selected from the Arduino boards menu
- If using platformio, make sure you add those build flags in your environment section:

```ini
build_flags =  -DARDUINO_M5STACK_CORES3
```

cohen-sydney, 2024-01-20:
@tobozo thanks I am using Arduino IDE.

cohen-sydney, 2024-01-20:
Yep got it working thanks, just now need to set the time. should I connect to WiFi and use NTP, thoughts?