lovyan03/LovyanGFX #514: support for hx8396a lcd？

zhjygit, 2024-01-24:
Now， Panel_HX8357.hpp is ok, however my lcd is hx8396a,how could the lovyanGFX support the hx8396a?
Waiting for you reply.

tobozo, 2024-01-24:
> how could the lovyanGFX support the hx8396a?

if two panel products are from the same line it can be achieved by implementing a variant in the Panel_HX8357 according to the product specs sheet of the X8357 device

however this product reference doesn't seem to exist on aliexpress, google or amazon

zhjygit, 2024-01-24:
> > how could the lovyanGFX support the hx8396a?
> 
> if two panel products are from the same line it can be achieved by implementing a variant in the Panel_HX8357 according to the product specs sheet of the X8357 device
> 
> however this product reference doesn't seem to exist on aliexpress, google or amazon

Sorry sir, it is hx8369a.
Here is the document:
[HX8369-A_DS_preliminary_v01_100713.pdf](https://github.com/lovyan03/LovyanGFX/files/14038320/HX8369-A_DS_preliminary_v01_100713.pdf)
[esp32-s3-wroom-1_wroom-1u_datasheet_cn.pdf](https://github.com/lovyan03/LovyanGFX/files/14038324/esp32-s3-wroom-1_wroom-1u_datasheet_cn.pdf)
[屏幕规格书.pdf](https://github.com/lovyan03/LovyanGFX/files/14038327/default.pdf)

tobozo, 2024-01-24:
thanks, found more results with "HX8369-A" term indeed :+1: 

[this](https://github.com/muojie/spec_mtk/blob/095744d6c1f8ab07010eaa60f000fd2936c0c52b/MTK_driver/spec/LCM/HX8369-A/YT50F51J0%E8%B0%83%E8%AF%95%E8%B5%84%E6%96%99/5051%E4%BB%A3%E7%A0%81.txt#L4) appears to be the most readable example for this model, if you've already set the wiring, then you can start poking around with Panel_HX8357 files to see what happens, if they're from the same device family  it should be as simple as replacing the start initial sequence

closed 2024-03-01 by github-actions[bot]