lovyan03/LovyanGFX #525: ILI9488 working with arduinoIDE bu not working with espIDF

paxo-project, 2024-02-11:
## Environment ( 実行環境 )

- MCU or Board name: ESP32 WROVER B
- Panel Driver IC:    ILI9488
- Bus type:           SPI (3 wires)
- LovyanGFX version:  latest
- FrameWork version:   ESP-IDF
- Build Environment:   PlatformIO
- Operating System:   Linux

## Problem Description ( 問題の内容 )
I'm not able to connect the screen on espIDF, even if i use the same working configuration from arduinoIDE

### Expected Behavior ( 期待される動作 )
White screen

### Steps to reproduce ( 再現のための前提条件 )
Just run a simple code to show color on the screen;

### Code to reproduce this issue ( 再現させるためのコード )

Working on ArduinoIDE
<script src="https://gist.github.com/paxo-project/f1cedce5fb5cd6a02ac5291613fef2a8.js"></script>

Not working on espIDF
<script src="https://gist.github.com/paxo-project/158519d397871860696c70295bef02d0.js"></script>

This code is run on a pcb made for the project paxo.

Thank you in advance!

Sincerely Gabriel Rochet

tobozo, 2024-02-11:
hi,

please prodive esp-idf version, the content of sdkconfig or anything overloading its values may be helpful too

paxo-project, 2024-02-11:
Thank you for your reply, I will give you all you need but I'm a beginner with espIDF, so I don't know the useful information, so can you tell me what do you need? (I will send you all when I would be available)

paxo-project, 2024-02-15:
here is my .ini:

```[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf
monitor_speed = 115200
lib_deps =
    lovyan03/LovyanGFX@^1.1.9
build_flags =
    -DPLATFORM_ESP32DEV
    -DCONFIG_FATFS_MAX_LFN=255
    -I include
    -fexceptions```
    
LovyanGFX version: ```LovyanGFX@^1.1.9```
pio version: ```PlatformIO 6.1.13 is currently the newest version available.```
espIDF version: ```6.4.0```

If you need more information, ask me or follow the repository link?
[repo](https://github.com/paxo-phone/PaxOS-9)

Thank you in advance!

tobozo, 2024-02-16:
esp-idf version is written at the top of your sdkconfig file e.g. a commit shows you recently switched from [5.1.1 to 5.1.2](https://github.com/paxo-phone/PaxOS-9/commit/25d6035bcc4fe392d4111d236613ca1e5b26e506)

I couldn't compile your project (some unrelated errors) so I created a simple example from it:

- deleted everything except the files visible in the screenshot:

[image]

- copied this [main.cpp](https://gist.github.com/tobozo/2410c4712de2fa0305cba779f729ac43) in the src folder
- `pio run -e esp32dev`

please confirm the problem is still present when using this minimal example, and provide the full console output of the startup sequence

paxo-project, 2024-02-16:
Thank you for your reply:

I confirm the problem is still present and here is the full console output:
```ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7076
load:0x40078000,len:15584
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3876
entry 0x4008064c
␛[0;32mI (30) boot: ESP-IDF 5.1.1 2nd stage bootloader␛[0m
␛[0;32mI (30) boot: compile time Feb 16 2024 18:38:05␛[0m
␛[0;32mI (30) boot: Multicore bootloader␛[0m
␛[0;32mI (34) boot: chip revision: v1.1␛[0m
␛[0;32mI (38) boot.esp32: SPI Speed      : 40MHz␛[0m
␛[0;32mI (43) boot.esp32: SPI Mode       : DIO␛[0m
␛[0;32mI (47) boot.esp32: SPI Flash Size : 4MB␛[0m
␛[0;32mI (52) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (57) boot: Partition Table:␛[0m
␛[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (83) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (91) boot: End of partition table␛[0m
␛[0;32mI (95) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1ec7ch (126076) map␛[0m
␛[0;32mI (149) esp_image: segment 1: paddr=0002eca4 vaddr=3ffb0000 size=01374h (  4980) load␛[0m
␛[0;32mI (151) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=4eaa8h (322216) map␛[0m
␛[0;32mI (271) esp_image: segment 3: paddr=0007ead0 vaddr=3ffb1374 size=01020h (  4128) load␛[0m
␛[0;32mI (273) esp_image: segment 4: paddr=0007faf8 vaddr=40080000 size=0dca4h ( 56484) load␛[0m
␛[0;32mI (307) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (307) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (319) cpu_start: Multicore app␛[0m
␛[0;32mI (319) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (319) cpu_start: Starting app cpu, entry point is 0x40082184␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (337) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (337) cpu_start: cpu freq: 160000000 Hz␛[0m
␛[0;32mI (337) cpu_start: Application information:␛[0m
␛[0;32mI (342) cpu_start: Project name:     os9_test_screen␛[0m
␛[0;32mI (347) cpu_start: App version:      1␛[0m
␛[0;32mI (352) cpu_start: Compile time:     Feb 16 2024 18:37:40␛[0m
␛[0;32mI (358) cpu_start: ELF file SHA256:  46c3eb2a583a3139...␛[0m
␛[0;32mI (364) cpu_start: ESP-IDF:          5.1.1␛[0m
␛[0;32mI (369) cpu_start: Min chip rev:     v0.0␛[0m
␛[0;32mI (373) cpu_start: Max chip rev:     v3.99 ␛[0m
␛[0;32mI (378) cpu_start: Chip rev:         v1.1␛[0m
␛[0;32mI (383) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (390) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM␛[0m
␛[0;32mI (396) heap_init: At 3FFB4778 len 0002B888 (174 KiB): DRAM␛[0m
␛[0;32mI (403) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (409) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (415) heap_init: At 4008DCA4 len 0001235C (72 KiB): IRAM␛[0m
␛[0;32mI (423) spi_flash: detected chip: gd␛[0m
␛[0;32mI (426) spi_flash: flash io: dio␛[0m
␛[0;33mW (430) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.␛[0m
␛[0;33mW (443) DAC: legacy driver is deprecated, please migrate to `driver/dac_oneshot.h`, `driver/dac_cosine.h` or `driver/dac_continuous.h` instead␛[0m
␛[0;32mI (460) app_start: Starting scheduler on CPU0␛[0m
␛[0;32mI (461) app_start: Starting scheduler on CPU1␛[0m
␛[0;32mI (461) main_task: Started on CPU0␛[0m
␛[0;32mI (471) main_task: Calling app_main()␛[0m
connected```

tobozo, 2024-02-16:
> ␛[0;33mW (443) DAC: legacy driver is deprecated, please migrate to `driver/dac_oneshot.h`, `driver/dac_cosine.h` or `driver/dac_continuous.h` instead␛[0m

Strange, apart from this deprecation warning I don't see anything hinting at a driver or bus problem :thinking: 

Although it is recommended to use `framework = arduino, espidf` (esp-idf with Arduino as a component), I just tried both with a M5CoreS3 and it appears to work in both situations.

So here's the other recipe you can test, with Arduino as a component:

- use `framework = arduino, espidf` instead of  `framework = espidf` in the platformio.ini
- copy the contents of [this file](https://gist.github.com/paxo-project/f1cedce5fb5cd6a02ac5291613fef2a8) in main.cpp (it's the Arduino version you linked earlier)
- uncomment/set/edit the following values in the sdkconfig file:
```
CONFIG_AUTOSTART_ARDUINO=y
# CONFIG_WS2812_LED_ENABLE is not set
CONFIG_FREERTOS_HZ=1000
CONFIG_MBEDTLS_PSK_MODES=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK=y
```

paxo-project, 2024-02-21:
Ok, thank you very much, I completely changed to arduino on pio because idf has a lot of problems. Now everything works perfectly

closed 2024-02-21 by paxo-project