lovyan03/LovyanGFX #530: Support of emulated ARDUINO Projects with Portdunio using native SDL2 Lib for X11 support

marelab, 2024-02-22:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**
ATM the lib seems to work for Arduino OR Linux Support with SDL2. But with projects that are based on Arduino C Code and get ported to Linux via  portduino (https://github.com/meshtastic/framework-portduino) you run into a problem when both are defined and needed. I think its quite unusable to have a ARDUINO defines and SDL2 because its a native Linux lib. So your lib works perfect if you just have one of its tragets defined. I the case you want both ARDUINO & SDL you run into problem with the .pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/common.hpp because for elif hits first ARDUINO define and never includes SDL2. The build that wants SDL2 as Arduino under Linux fails then because wrong headers.

## Environment ( 実行環境 )

- MCU or Board name: Linux Native X86 & ARM
- Panel Driver IC:    SDL2 X11
- Bus type:           [other]
- LovyanGFX version:  [latest develop|other]
- FrameWork version:   [Portdunio]
- Build Environment:   [PlatformIO|Eclipse]
- Operating System:   [Linux|macOS]

## Problem Description ( 問題の内容 )
If we got a Situation where its a ARDUINO build for linux with the portdunio lib that wants to use also SDL2 the include should be made before ARDUINO

### Expected Behavior ( 期待される動作 )

Use of including common.hpp for SDL sould be possible even if a define for ARDUINO is set.

### Actual Behavior ( 実際の動作 )
If a define for ARDUINO is used a SDL2 lib can't because the common.hpp for arduino ist included. 

### Steps to reproduce ( 再現のための前提条件 )

1. Checkout the https://github.com/marelab/firmware.git 
2. Under arch/portduino/portduino.ini change the lovyan03 lib_deps back to the original
3. Build it with VScode & platformio.

tobozo, 2024-02-22:
I'm not sure this requires a pull request, compilation succeeded just by using the develop branch instead of the master branch:

```ini
lib_deps =
  ${env.lib_deps}
  ${networking_base.lib_deps}
  rweather/Crypto@^0.4.0
  git+https://github.com/lovyan03/LovyanGFX.git#develop
```

[image]

marelab, 2024-02-22:
Thank you are right !!! I checked it now againt the develop branch and the problem has gone. Sorry I had to check that by myself before but was focused on the version meshtastic used.

closed 2024-02-22 by marelab