lovyan03/LovyanGFX #531: LVGL drawing freezes on Sunton ESP32-S3 8048S050 doard.

alexander-sorokin, 2024-02-25:
## Environment ( 実行環境 )

- MCU or Board name:  [ESP32-8048S050 - Sunton ESP32-S3 5" 800*480 IPS with Capacitive Touch]
- Panel Driver IC:    [ST7262]
- Bus type:           [I2C]
- LovyanGFX version:  [1.1.12]
- FrameWork version:   [ESP-IDF v5.1] - [ESP-IDF v5.3]
- Build Environment:   [VS Code with ESP-IDF extension]
- Operating System:   [Windows]

## Problem Description ( 問題の内容 )
While using LVGL, display freezes after drawing one or few first frames. I added logging to touch callback but it never shows in console output, while display flushing reports for first half a second. lv_timer_handler() is called each 30 milliseconds.

When I inserted and adopted code from LovyanGFX readme, it worked fine and showed animation.

I do not want to use Arduino library for ESP32.

### Expected Behavior ( 期待される動作 )
See changes on screen from LVGL animation or touch responses.

### Actual Behavior ( 実際の動作 )
Display freezes while rendering LVGL.

<details>
<summary>Console output</summary>

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x18dc
load:0x403c9700,len:0x4
load:0x403c9704,len:0xda4
load:0x403cc700,len:0x3084
entry 0x403c9934
I (27) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (27) boot: compile time Feb 24 2024 16:17:16
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (48) boot.esp32s3: SPI Flash Size : 16MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=2a730h (173872) map
I (131) esp_image: segment 1: paddr=0003a758 vaddr=3fc92300 size=03304h ( 13060) load
I (134) esp_image: segment 2: paddr=0003da64 vaddr=40374000 size=025b4h (  9652) load
I (139) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=4ebf8h (322552) map
I (194) esp_image: segment 4: paddr=0008ec20 vaddr=403765b4 size=0bcd0h ( 48336) load
I (210) boot: Loaded app from partition at offset 0x10000
I (210) boot: Disabling RNG early entropy source...
I (222) cpu_start: Multicore app
I (222) octal_psram: vendor id    : 0x0d (AP)
I (222) octal_psram: dev id       : 0x02 (generation 3)
I (225) octal_psram: density      : 0x03 (64 Mbit)
I (231) octal_psram: good-die     : 0x01 (Pass)
I (236) octal_psram: Latency      : 0x01 (Fixed)
I (241) octal_psram: VCC          : 0x01 (3V)
I (246) octal_psram: SRF          : 0x01 (Fast Refresh)
I (252) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (258) octal_psram: BurstLen     : 0x01 (32 Byte)
I (263) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (270) octal_psram: DriveStrength: 0x00 (1/1)
I (275) esp_psram: Found 8MB PSRAM device
I (279) esp_psram: Speed: 40MHz
I (283) cpu_start: Pro cpu up.
I (287) cpu_start: Starting app cpu, entry point is 0x4037540c
0x4037540c: call_start_cpu1 at D:/DEV/esp/v5.1.2/esp-idf/components/esp_system/port/cpu_start.c:157

I (0) cpu_start: App cpu up.
I (1024) esp_psram: SPI SRAM memory test OK
I (1033) cpu_start: Pro cpu start user code
I (1033) cpu_start: cpu freq: 240000000 Hz
I (1033) cpu_start: Application information:
I (1037) cpu_start: Project name:     lovyangfx-esp
I (1042) cpu_start: App version:      b70a495-dirty
I (1048) cpu_start: Compile time:     Feb 24 2024 16:16:43
I (1054) cpu_start: ELF file SHA256:  8ac53a1d17efe62f...
I (1060) cpu_start: ESP-IDF:          v5.1.2
I (1065) cpu_start: Min chip rev:     v0.1
I (1070) cpu_start: Max chip rev:     v0.99
I (1075) cpu_start: Chip rev:         v0.1
I (1080) heap_init: Initializing. RAM available for dynamic allocation:
I (1087) heap_init: At 3FCAE700 len 0003B010 (236 KiB): DRAM
I (1093) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (1100) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1106) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (1113) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (1121) spi_flash: detected chip: gd
I (1124) spi_flash: flash io: qio
I (1129) sleep: Configure to isolate all GPIO pins in sleep state
I (1135) sleep: Enable automatic switching of GPIO sleep configuration
I (1143) app_start: Starting scheduler on CPU0
I (1148) app_start: Starting scheduler on CPU1
I (1148) main_task: Started on CPU0
I (1158) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1168) main_task: Calling app_main()
I (1388) my-tag: LVGL task started.
I (1388) my-tag: Flushing display.
I (1398) my-tag: Flushing display.
I (1408) my-tag: Flushing display.
I (1408) my-tag: Flushing display.
I (1408) my-tag: Flushing display.
I (1418) my-tag: Flushing display.
I (1418) my-tag: Flushing display.
I (1428) my-tag: Flushing display.
I (1428) my-tag: Flushing display.
I (1428) my-tag: Flushing display.
I (1438) my-tag: Flushing display.
I (1438) my-tag: Flushing display.
I (1438) my-tag: Flushing display.
I (1448) my-tag: Flushing display.
I (1448) my-tag: Flushing display.
I (1458) my-tag: Flushing display.
I (1458) my-tag: Flushing display.
I (1458) my-tag: Flushing display.
I (1468) my-tag: Flushing display.
I (1468) my-tag: Flushing display.
I (1468) my-tag: Flushing display.
I (1478) my-tag: Flushing display.
I (1478) my-tag: Flushing display.
I (1478) my-tag: Flushing display.
I (1488) my-tag: Flushing display.
I (1488) my-tag: Flushing display.
I (1498) my-tag: Flushing display.
I (1498) my-tag: Flushing display.
I (1498) my-tag: Flushing display.
I (1508) my-tag: Flushing display.
I (1508) my-tag: Flushing display.
I (1518) my-tag: Flushing display.
I (1518) my-tag: Flushing display.
I (1528) my-tag: Flushing display.
I (1528) my-tag: Flushing display.
I (1528) my-tag: Flushing display.
I (1538) my-tag: Flushing display.
I (1538) my-tag: Flushing display.
I (1548) my-tag: Flushing display.
I (1548) my-tag: Flushing display.
I (1548) my-tag: Flushing display.
I (1558) my-tag: Flushing display.
I (1558) my-tag: Flushing display.
I (1568) my-tag: Flushing display.
I (1568) my-tag: Flushing display.
I (1568) my-tag: Flushing display.
I (1578) my-tag: Flushing display.
I (1578) my-tag: Flushing display.
I (1588) my-tag: LVGL suggested redraw in 30 milliseconds.
I (1588) main_task: Returned from app_main()
I (1618) my-tag: LVGL suggested redraw in 30 milliseconds.
I (1648) my-tag: LVGL suggested redraw in 30 milliseconds.
I (1678) my-tag: LVGL suggested redraw in 30 milliseconds.
I (1708) my-tag: LVGL suggested redraw in 30 milliseconds.
I (1738) my-tag: LVGL suggested redraw in 30 milliseconds.
I (1768) my-tag: LVGL suggested redraw in 30 milliseconds.
I (1798) my-tag: LVGL suggested redraw in 30 milliseconds.
I (1828) my-tag: LVGL suggested redraw in 30 milliseconds.
I (1858) my-tag: LVGL suggested redraw in 30 milliseconds.
I (1888) my-tag: LVGL suggested redraw in 30 milliseconds.
I (1918) my-tag: LVGL suggested redraw in 30 milliseconds.
```
</details>

### Steps to reproduce ( 再現のための前提条件 )

1. ``` git clone --recursive https://github.com/alexander-sorokin/lovyangfx-esp-lvgl.git ```
2. Open in [VS Code](https://code.visualstudio.com/) with [ESP-IDF](https://github.com/espressif/esp-idf) extension installed.
3. Configure (my configuration is included), build and upload.

### Code to reproduce this issue ( 再現させるためのコード )

I created [sample project](https://github.com/alexander-sorokin/lovyangfx-esp-lvgl/) which causes the issue.

mhaberler, 2024-02-25:
this is just a gut feeling, anyway:

I'm using a similar board ( Sunton ESP32-8048S070) in code based on @sukesh-ak 's https://github.com/sukesh-ak/ESP32-TUX, and never had a lockup like you describe

have a look at the [lvgl_acquire/lvgl_release lock handling](https://github.com/sukesh-ak/ESP32-TUX/blob/4eea9bfd0c82c5b6a7c9b207b9e76c9f7b202844/main/helpers/helper_display.hpp#L217-L231) - my understanding is that you surround any lvgl code with lvgl_acquire/lvgl_release

st3m0r, 2024-02-25:
increase the memory in the lvgl.conf file  #define LV_MEM_SIZE (50* 1024U)

sukesh-ak, 2024-02-26:
@alexander-sorokin LVGL does not support multi-tasking. So you need to take care of it yourself like my code sample mentioned above.

alexander-sorokin, 2024-02-26:
The issue was I am not calling [lv_tick_inc(uint32_t tick_period)](https://docs.lvgl.io/8.3/porting/tick.html?highlight=lv_tick_inc). I followed [this example](https://github.com/lovyan03/LovyanGFX/blob/4fd713ebd3019d17c6c89c6e2f7171d2cafeff23/examples/Advanced/LVGL_PlatformIO/src/main.cpp), but it is for Arduino. There is [configuration](https://github.com/lovyan03/LovyanGFX/blob/4fd713ebd3019d17c6c89c6e2f7171d2cafeff23/examples/Advanced/LVGL_PlatformIO/src/lv_conf.h#L88) for automatic calling of this function.

closed 2024-02-26 by alexander-sorokin