lovyan03/LovyanGFX #533: drawJPG does not work

daverathbone, 2024-02-28:
## Environment ( 実行環境 )

- MCU or Board name:  [ESP-WROOM-32r]
- Panel Driver IC:    [ILI9488]
- Bus type:           [SPIr]
- LovyanGFX version:  [latest developr]
- FrameWork version:   |ESP-IDF v5.0]
- Build Environment:   [Visual Studio Code IDE]
- Operating System:   [Windows]

## Problem Description ( 問題の内容 )
My code works for BMP and PNG but Not JPG

### Expected Behavior ( 期待される動作 )
Should take buffer of JPEG file and convertto image

### Actual Behavior ( 実際の動作 )
No error but blank

### Steps to reproduce ( 再現のための前提条件 )
step 1
  lcd.drawBmp(buffer, fileSize, x, y,0, 0, 0, 0, scalex, scaley);   //  works
Step 2
  lcd.drawPng(buffer, fileSize, x, y,0, 0, 0, 0, scalex, scaley);   // works
Step 3
  lcd.drawJpg(buffer, fileSize, x, y,0, 0, 0, 0, scalex, scaley);

// If possible, attach a picture of your setup/wiring here.
The display has text as well  that works but no image if jpg

// main text underlogo
lcd.setCursor(0, 128);
lcd.setTextColor(TFT_GREEN);	
lcd.setFont( &fonts::FreeMonoBold9pt7b);
lcd.println("Select settings with OK button (Not Fitted)");
lcd.println("");

### Code to reproduce this issue ( 再現させるためのコード )
```cpp
// place bmp image from file at x,y 
void displayJPG(const char *filename, int32_t x, int32_t y, float scalex , float scaley) {
    
        FILE *file = fopen(filename, "r");
        if (!file) {
            ESP_LOGE(TAG, "Failed to open file for reading");
            return;
        }

        fseek(file, 0, SEEK_END);
        long fileSize = ftell(file);
        fseek(file, 0, SEEK_SET);

        uint8_t *buffer = (uint8_t *)malloc(fileSize);
        if (!buffer) {
            ESP_LOGE(TAG, "Failed to allocate memory for file content");
            fclose(file);
            return;
        }

        size_t bytesRead = fread(buffer, 1, fileSize, file);
        if (bytesRead != fileSize) {
            ESP_LOGE(TAG, "Failed to read the file into buffer");
            free(buffer);
            fclose(file);
            return;
        }

        fclose(file);

        lcd.startWrite(); 
        lcd.drawJpg(buffer, fileSize, x, y,0, 0, 0, 0, scalex, scaley);
        lcd.endWrite();  

        free(buffer);
    }

```
**Please submit complete source code that can reproduce your problem.**
**あなたの問題を再現できる完全なソースコードを提示してください。**
 Program is to large but you have the code

lovyan03, 2024-02-28:
Progressive JPEG is not supported. Please check the JPEG data specifications.

daverathbone, 2024-02-28:
What JPEG data specification is used if not Progressive JPEG ?

lovyan03, 2024-02-28:
I don't know the details, but drawJpg is working for me.

daverathbone, 2024-02-28:
My jpg is just from Windows Paint.net. how have you made your JPG's?

lovyan03, 2024-02-28:
find out for yourself

lovyan03, 2024-02-28:
> Progressive is the normal JPG used in 89% of most applications today.

It is difficult to support progressive JPG with a microcontroller with little memory. Try it yourself.

closed 2024-02-28 by lovyan03