lovyan03/LovyanGFX #542: ST7701 / GC9503 ESP32-S3 wrong colors behaviour

FigueiredoStable, 2024-03-14:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  [ESP32-S3-WROOM]
- Panel Driver IC:    [ST7701 / GC9503]
- Bus type:           [SPI]
- LovyanGFX version:  [v1.1.12]
- FrameWork version:   [ArduinoESP32]
- Build Environment:   [PlatformIO]
- Operating System:   [macOS]

## Problem Description ( 問題の内容 )

The screen appears to have inverted colors, as if it were a negative.

### Expected Behavior ( 期待される動作 )

Correct colors.
[image]

### Actual Behavior ( 実際の動作 )

All screens generated are wrong colors.

### Steps to reproduce ( 再現のための前提条件 )

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

#screen
[image]

# board
[image]

### Code to reproduce this issue ( 再現させるためのコード )

# board pin configuration:
``` c
#define LGFX_USE_V1
#include <LovyanGFX.hpp>
#include <lgfx/v1/platforms/esp32s3/Panel_RGB.hpp>
#include <lgfx/v1/platforms/esp32s3/Bus_RGB.hpp>
#include <driver/i2c.h>

class LGFX : public lgfx::LGFX_Device
{
public:
	lgfx::Bus_RGB _bus_instance;
	lgfx::Panel_GC9503 _panel_instance;
	lgfx::Touch_GT911 _touch_instance;
	lgfx::Light_PWM _light_instance;

	LGFX(void)
	{
		{
			auto cfg = _panel_instance.config();

			cfg.memory_width  = 480;
			cfg.memory_height = 480;
			cfg.panel_width  = 480;
			cfg.panel_height = 480;

			cfg.offset_x = 0;
			cfg.offset_y = 0;

[... 967 lines cut ...]
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

When I use the latest version of Arduino_GFX with the configuration below it shows normally.

```c
#include <Arduino_GFX_Library.h>

#define GFX_BL 38

Arduino_ESP32RGBPanel *bus = new Arduino_ESP32RGBPanel(
  39 /* CS */, 48 /* SCK */, 47 /* SDA */,
  18 /* DE */, 17 /* VSYNC */, 16 /* HSYNC */, 21 /* PCLK */,
  11 /* R0 */, 12 /* R1 */, 13 /* R2 */, 14 /* R3 */, 0 /* R4 */,
  8 /* G0 */, 20 /* G1 */, 3 /* G2 */, 46 /* G3 */, 9 /* G4 */, 10 /* G5 */,
  4 /* B0 */, 5 /* B1 */, 6 /* B2 */, 7 /* B3 */, 15 /* B4 */
);
    Arduino_ST7701_RGBPanel *gfx = new Arduino_ST7701_RGBPanel(
    bus, GFX_NOT_DEFINED /* RST */, 0 /* rotation */,
    true /* IPS */, 480 /* width */, 480 /* height */,
    st7701_type1_init_operations, sizeof(st7701_type1_init_operations),     true /* BGR */,
    10 /* hsync_front_porch */, 8 /* hsync_pulse_width */, 50 /* hsync_back_porch */,
    10 /* vsync_front_porch */, 8 /* vsync_pulse_width */, 20 /* vsync_back_porch */);

```

FigueiredoStable, 2024-03-16:
After better evaluating the version for the arduino_GFX library that was sent by the manufacturer, there are changes in the instructions, in any version just change the following points that work, as follows:

### /* Changed by manufacturer */

```c
static const uint8_t st7701_type1_init_operations[] = {
    BEGIN_WRITE,
    WRITE_COMMAND_8, 0xFF,
    WRITE_BYTES, 5, 0x77, 0x01, 0x00, 0x00, 0x10,

    WRITE_C8_D16, 0xC0, 0x3B, 0x00,
    WRITE_C8_D16, 0xC1, 0x0D, 0x02,
    WRITE_C8_D16, 0xC2, 0x31, 0x05,
    WRITE_C8_D8, 0xCD, 0x00, //0x08 /* Changed by manufacturer */

    WRITE_COMMAND_8, 0xB0, // Positive Voltage Gamma Control
    WRITE_BYTES, 16,
    0x00, 0x11, 0x18, 0x0E,
    0x11, 0x06, 0x07, 0x08,
    0x07, 0x22, 0x04, 0x12,
    0x0F, 0xAA, 0x31, 0x18,

    WRITE_COMMAND_8, 0xB1, // Negative Voltage Gamma Control
    WRITE_BYTES, 16,
    0x00, 0x11, 0x19, 0x0E,
    0x12, 0x07, 0x08, 0x08,
    0x08, 0x22, 0x04, 0x11,
    0x11, 0xA9, 0x32, 0x18,

    // PAGE1
    WRITE_COMMAND_8, 0xFF,
    WRITE_BYTES, 5, 0x77, 0x01, 0x00, 0x00, 0x11,

    WRITE_C8_D8, 0xB0, 0x60, // Vop=4.7375v
    WRITE_C8_D8, 0xB1, 0x32, // VCOM=32
    WRITE_C8_D8, 0xB2, 0x07, // VGH=15v
    WRITE_C8_D8, 0xB3, 0x80,
    WRITE_C8_D8, 0xB5, 0x49, // VGL=-10.17v
    WRITE_C8_D8, 0xB7, 0x85,
    WRITE_C8_D8, 0xB8, 0x21, // AVDD=6.6 & AVCL=-4.6
    WRITE_C8_D8, 0xC1, 0x78,
    WRITE_C8_D8, 0xC2, 0x78,

    WRITE_COMMAND_8, 0xE0,
    WRITE_BYTES, 3, 0x00, 0x1B, 0x02,

    WRITE_COMMAND_8, 0xE1,
    WRITE_BYTES, 11,
    0x08, 0xA0, 0x00, 0x00,
    0x07, 0xA0, 0x00, 0x00,
    0x00, 0x44, 0x44,

    WRITE_COMMAND_8, 0xE2,
    WRITE_BYTES, 12,
    0x11, 0x11, 0x44, 0x44,
    0xED, 0xA0, 0x00, 0x00,
    0xEC, 0xA0, 0x00, 0x00,

    WRITE_COMMAND_8, 0xE3,
    WRITE_BYTES, 4, 0x00, 0x00, 0x11, 0x11,

    WRITE_C8_D16, 0xE4, 0x44, 0x44,

    WRITE_COMMAND_8, 0xE5,
    WRITE_BYTES, 16,
    0x0A, 0xE9, 0xD8, 0xA0,
    0x0C, 0xEB, 0xD8, 0xA0,
    0x0E, 0xED, 0xD8, 0xA0,
    0x10, 0xEF, 0xD8, 0xA0,

    WRITE_COMMAND_8, 0xE6,
    WRITE_BYTES, 4, 0x00, 0x00, 0x11, 0x11,

    WRITE_C8_D16, 0xE7, 0x44, 0x44,

    WRITE_COMMAND_8, 0xE8,
    WRITE_BYTES, 16,
    0x09, 0xE8, 0xD8, 0xA0,
    0x0B, 0xEA, 0xD8, 0xA0,
    0x0D, 0xEC, 0xD8, 0xA0,
    0x0F, 0xEE, 0xD8, 0xA0,

    WRITE_COMMAND_8, 0xEB,
    WRITE_BYTES, 7,
    0x02, 0x00, 0xE4, 0xE4,
    0x88, 0x00, 0x40,

    WRITE_C8_D16, 0xEC, 0x3C, 0x00,

    WRITE_COMMAND_8, 0xED,
    WRITE_BYTES, 16,
    0xAB, 0x89, 0x76, 0x54,
    0x02, 0xFF, 0xFF, 0xFF,
    0xFF, 0xFF, 0xFF, 0x20,
    0x45, 0x67, 0x98, 0xBA,

    //-----------VAP & VAN---------------
    WRITE_COMMAND_8, 0xFF,
    WRITE_BYTES, 5, 0x77, 0x01, 0x00, 0x00, 0x13,

    WRITE_C8_D8, 0xE5, 0xE4,

    WRITE_COMMAND_8, 0xFF,
    WRITE_BYTES, 5, 0x77, 0x01, 0x00, 0x00, 0x00,

    //WRITE_COMMAND_8, 0x21,   // 0x20 normal, 0x21 IPS /* Changed by manufacturer */
    WRITE_C8_D8, 0x3A, 0x60, // 0x70 RGB888, 0x60 RGB666, 0x50 RGB565

    WRITE_COMMAND_8, 0x11, // Sleep Out
    END_WRITE,

    DELAY, 120,

    BEGIN_WRITE,
    WRITE_COMMAND_8, 0x29, // Display On
    END_WRITE};
```

Is possible to change in LovyanGFX too?

*Manufacturer download links are no longer active. I will leave a link to the files sent by the manufacturer.

[ESP32 4.0inch.zip] https://file.io/JPyNFOs0t4in

FigueiredoStable, 2024-04-22:
Hy just to document, i solve this question change the instructions to RGP Panel like a manufacture did with Arduino_GFX, but on LovyanGFX, solution above:

## Add these lines to Panel_RGB.cpp
```cpp
//----------------------------------------------------------------------------

  const uint8_t* Panel_ST7701_gution_esp32_4848S040::getInitCommands(uint8_t listno) const
  {
    static constexpr const uint8_t list0[] =
    {
      // Command2 BK0 SEL
      0xFF,  5, 0x77, 0x01, 0x00, 0x00, 0x10,

      0xC0,  2,	0x3B, 0x00,  // added by manufacture
      0xC1,  2, 0x0D, 0x02,
      0xC2,  2, 0x31, 0x05,
      0xCD,  1, 0x00,  // changed by manufacture

      // Positive Voltage Gamma Control
      0xB0, 16, 0x00, 0x11, 0x18, 0x0E, 0x11, 0x06, 0x07, 0x08,
                0x07, 0x22, 0x04, 0x12, 0x0F, 0xAA, 0x31, 0x18,

      // Negative Voltage Gamma Control
      0xB1, 16, 0x00, 0x11, 0x19, 0x0E, 0x12, 0x07, 0x08, 0x08,
                0x08, 0x22, 0x04, 0x11, 0x11, 0xA9, 0x32, 0x18,

      // Command2 BK1 SEL
      0xFF,  5, 0x77, 0x01, 0x00, 0x00, 0x11,

      0xB0,  1, 0x60, // Vop=4.7375v
      0xB1,  1, 0x32, // VCOM=32
      0xB2,  1, 0x07, // VGH=15v
      0xB3,  1, 0x80,
      0xB5,  1, 0x49, // VGL=-10.17v
      0xB7,  1, 0x85,
      0xB8,  1, 0x21, // AVDD=6.6 & AVCL=-4.6
      0xC1,  1, 0x78,
      0xC2,  1, 0x78,

      0xE0,  3, 0x00, 0x1B, 0x02,

      0xE1, 11, 0x08, 0xA0, 0x00, 0x00, 0x07, 0xA0, 0x00, 0x00, 0x00, 0x44, 0x44,
      0xE2, 12, 0x11, 0x11, 0x44, 0x44, 0xED, 0xA0, 0x00, 0x00, 0xEC, 0xA0, 0x00, 0x00,

      0xE3,  4, 0x00, 0x00, 0x11, 0x11,
      0xE4,  2, 0x44, 0x44,

      0xE5, 16, 0x0A, 0xE9, 0xD8, 0xA0, 0x0C, 0xEB, 0xD8, 0xA0,
                0x0E, 0xED, 0xD8, 0xA0, 0x10, 0xEF, 0xD8, 0xA0,

      0xE6,  4, 0x00, 0x00, 0x11, 0x11,

      0xE7,  2, 0x44, 0x44,

      0xE8, 16, 0x09, 0xE8, 0xD8, 0xA0, 0x0B, 0xEA, 0xD8, 0xA0,
                0x0D, 0xEC, 0xD8, 0xA0, 0x0F, 0xEE, 0xD8, 0xA0,

      0xEB,  7, 0x02, 0x00, 0xE4, 0xE4, 0x88, 0x00, 0x40,
      0xEC,  2, 0x3C, 0x00,
      0xED, 16, 0xAB, 0x89, 0x76, 0x54, 0x02, 0xFF, 0xFF, 0xFF,
                0xFF, 0xFF, 0xFF, 0x20, 0x45, 0x67, 0x98, 0xBA,

      //-----------VAP & VAN---------------
      // Command2 BK3 SEL
      0xFF,  5, 0x77, 0x01, 0x00, 0x00, 0x13,

      0xE5,  1, 0xE4,

      // Command2 BK0 SEL
      0xFF,  5, 0x77, 0x01, 0x00, 0x00, 0x00,
[... 187 lines cut ...]

      0x11, CMD_INIT_DELAY, 200, // Sleep Out

      0x29,  0,
      0xFF, 0xFF
    };
    switch (listno)
    {
    case 0: return list0;
    default: return nullptr;
    }
  }
```

# Add the struct to header Panel_RGB.hpp

```cpp
struct Panel_ST7701_gution_esp32_4848S040: public Panel_RGB
  {
    bool init(bool) override;
  protected:
    const uint8_t* getInitCommands(uint8_t listno) const override;
  };
```
Now just use this version for this specifc board.

kchudoba, 2025-01-06:
Hi @FigueiredoStable 
I have the same board and even after adjusting the init commands etc I still get inverted colors. Do you have any ideas what exactly could be the reason for that?

FigueiredoStable, 2025-01-15:
> Hi @FigueiredoStable The colors started to work correctly after I disconnected the display from the laptop and reconnected again.. weird but I guess "have you tried to turn it off and on again" is still a valid strategy. Thanks for the link to the example, actually I saw it already before :) I couldn't try it directly as all my things are in LVGL 9.0 but I managed to get it working. However, the performance is still not so great - can you share what FPS are you getting in LVGL when doing a full page scrolling? When nothing is moving, I'm getting almost 200FPS but when I start scrolling a screen with many widgets, it goes down to 6 FPS. I'm not really sure if I am really using DMA for the screen updates. Did you get it to work any faster?

Hi, its really strange, but if your power delivery is no enough there a some of issue with this board, to me 5V 1A stays stable, less than this, not much, i am glad that works with the repository example, great job!.

Now about the FPS i really don't even used scrolling that on my projects at this moment, i need to upgrade to LVGL 9.0 to, in 8.1 i don't have any issue about FPS or performance, on my general project i use to generate a lot of QRCODES 2D, play some tiny videos and comunicattion with the backend over WiFi or GPIO external devices.

One tip to quickly check and monitoring your power delivery: [Usb Power Meter](https://pt.aliexpress.com/item/1005007170447680.html?spm=a2g0o.productlist.main.1.3a11kin6kin6mE&algo_pvid=5e495776-0ccc-424d-88a3-6fff8e43edd6&algo_exp_id=5e495776-0ccc-424d-88a3-6fff8e43edd6-0&pdp_npi=4%40dis%21BRL%2173.86%2136.19%21%21%2185.23%2141.76%21%40210337bc17369547377375008e1ea3%2112000039689287637%21sea%21BR%21168312606%21X&curPageLogUid=1RCzUmNZbdOT&utparam-url=scene%3Asearch%7Cquery_from%3A)