lovyan03/LovyanGFX #543: The screen is initialized correctly every 2 times

paxo-project, 2024-03-18:
**The screen is initialized correctly every 2 times**

Whether when starting the code, or when resetting the screen after a sleep mode, resetting communication with the screen only works once in 2, whether using ```lcd ->init();``` or ```lcd->init_without_reset()```
The project in question is still that of an open source phone and the fact that the screen only lights up once in 2 is quite problematic. The power supply is controlled by the esp32 and perfectly stable esp.
I specify that during a bad initialization, I also receive an error from the touch part in i2c:
```[332656][I][esp32-hal-i2c.c:75] i2cInit(): Initializing I2C Master: sda=21 scl=22 freq=100000
[333577][V][Touch_FT5x06.cpp:63] _check_init(): [FT5x06] CIPHER:0x64 / FIRMID:0x03 / VENDID:0x11```

## Environment ( 実行環境 )

- MCU or Board name:  esp32 wrover b
- Panel Driver IC:    ILI9488
- Bus type:           SPI (shared with SD card) + I2C (touchscreen)
- LovyanGFX version:  latest
- FrameWork version:   ArduinoESP32 v2.0.3
- Build Environment:   PlatformIO
- Operating System:   Linux

## Problem Description ( 問題の内容 )
Screen initialization fails 1 out of 2 times

### Expected Behavior ( 期待される動作 )
White sceen

### Steps to reproduce ( 再現のための前提条件 )
Just init the screen?

## Schematic
![image](https://github.com/lovyan03/LovyanGFX/assets/142974538/e2e29ca7-02aa-4ebb-b15c-094965fb489b)

### Code to reproduce this issue ( 再現させるためのコード )

**Please submit complete source code that can reproduce your problem.**
[Full Code](https://github.com/paxo-phone/PaxOS-9)

```cpp
// Config
#define LGFX_USE_V1

#include <LovyanGFX.hpp>

class LGFX : public lgfx::LGFX_Device
{
  lgfx::Panel_ILI9488     _panel_instance;
  lgfx::Bus_SPI       _bus_instance;   // SPIバスのインスタンス
  lgfx::Light_PWM     _light_instance;
  lgfx::Touch_FT5x06 _touch_instance;

public:
  LGFX(void)
  {
    {
      auto cfg = _bus_instance.config();

      cfg.spi_host = VSPI_HOST;
      cfg.spi_mode = 0;
      cfg.freq_write = 20000000;
      cfg.freq_read  = 16000000;
      cfg.spi_3wire  = true; // changed
      cfg.use_lock   = true;
      cfg.pin_sclk = 18;
      cfg.pin_mosi = 23;
      cfg.pin_dc   = 2;
     
     
     _bus_instance.config(cfg);
      _panel_instance.setBus(&_bus_instance);
    }

    {
      auto cfg = _panel_instance.config();

      cfg.pin_cs           =    13;
      cfg.pin_rst          =    -1;
      cfg.pin_busy         =    -1;

      cfg.panel_width      =   320; 
      cfg.panel_height     =   480; 
      cfg.offset_x         =     0; 
      cfg.offset_y         =     0; 
      cfg.offset_rotation  =     0; 
      cfg.dummy_read_pixel =     8; 
      cfg.dummy_read_bits  =     0;//1 
      cfg.readable         =  false; 
      cfg.invert           = false; 
      cfg.rgb_order        = false; 
      cfg.dlen_16bit       = false; 
      cfg.bus_shared       =  true; 

      cfg.memory_width     =   320; 
      cfg.memory_height    =   480; 

      _panel_instance.config(cfg);
    }
    {
      auto cfg = _light_instance.config();

      cfg.pin_bl = 5;
      cfg.invert = true;
      cfg.freq   = 44100;
      cfg.pwm_channel = 0;

      _light_instance.config(cfg);
      _panel_instance.setLight(&_light_instance);  // バックライトをパネルにセットします。
    }
    {
      auto cfg = _touch_instance.config();
      cfg.i2c_port = 1;
      cfg.i2c_addr = 0x38;
      cfg.pin_sda  = 21;
      cfg.pin_scl  = 22;
      
      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。
    }

    setPanel(&_panel_instance); // 使用するパネルをセットします。
  }
};
```

I hope it was me who made a simple mistake in my configuration
Thanks in advance!

tobozo, 2024-03-18:
> Bus type: SPI (shared with SD card)

is that behaviour consistent whether there is a sdcard inserted or not?

paxo-project, 2024-03-19:
It is the same even without SD card unfortunately

paxo-project, 2024-03-20:
I also add that the screen is connected via 3-wire spi, so no miso

closed 2024-04-26 by github-actions[bot]

paxo-rch, 2024-05-21:
Sorry to disturb you, but i still have the same problem, i can't figure how to fix it

tobozo, 2024-05-21:
had another look at the schema, but I can only theorize since I don't have the device: 

apparently the display reset pin isn't routed to the esp32, does it rely on power to operate the reset? 
what's pin 12 on the display socket?

[image]

speculation: if the lcd reset isn't controled by LovyanGFX at init, then the probablility that this problem is caused by something else is higher

it could be worth testing something like this at boot just to see if it changes anything, but with that reset pin attached to 3v3 on the lcd socket I'n not sure it'll have an effect:

```cpp
    hardware::setScreenPower(false);
    delay(200);
    hardware::setScreenPower(true);
```

paxo-rch, 2024-05-29:
I reordered a new pcb with a reset control. It now works 100% of the time!

Thank you for your help!