lovyan03/LovyanGFX #546: I need support for the CYD board (cheap yellow display) ESP32-2432S028R

SchneHa, 2024-04-03:
**Carefully written requests are more likely to be given priority.**
**丁寧に記述された要望は優先して対応される可能性が高くなります。**

### Device Name (デバイスの名称・型番等)
ESP32-2432S028R

### URL of Device Specifications document (仕様書等のURL)
https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/tree/main

### URL of the store where we can purchase (商品を購入できるURL)
https://de.aliexpress.com/item/1005006160645147.html?spm=a2g0o.order_list.order_list_main.5.21ef5c5f5ZKXsm&gatewayAdapt=glo2deu

lsdlsd88, 2024-04-05:
what makes u think it's not "supported"? 
usual chips, TFT ILI9341 + XPT2046 touch.

SchneHa, 2024-04-06:
I think the board has another screen that needs a variation of ILI9341 driver. Using TFT_eSPI library I need to take ILI9341_2 driver.

lsdlsd88, 2024-04-06:
try this alternative panel file. u can replace into yours just for testing

https://github.com/lsdlsd88/LovyanGFX/blob/7ba76d9fdf9f5bfe50a02e0394aedc7ec67dc125/src/lgfx/v1/panel/Panel_ILI9341_2.hpp

SchneHa, 2024-04-06:
After completing lines 79 and 80 with commas the driver works. 
I'd set the Gamma in lines 50 and 51 to 0 to get better colours:
    static constexpr uint8_t CMD_GMCTRP1 = 0; // Positive Gamma Correction (E0h)
    static constexpr uint8_t CMD_GMCTRN1 = 0; // Negative Gamma Correction (E1h)
Please see discussion at https://github.com/Bodmer/TFT_eSPI/issues/2985

SchneHa, 2024-04-06:
[image]

Smanar, 2024-04-19:
Strange on my side this one is working (Need to use ILI9341_2  too on TFT_eSPI  lib).
But it works fine with LovyanGFX with

```
class LGFX : public lgfx::LGFX_Device
{
    lgfx::Panel_ILI9341 _panel_instance;
    lgfx::Bus_SPI       _bus_instance;
    lgfx::Light_PWM     _light_instance;
    lgfx::Touch_XPT2046 _touch_instance;
  public:
    LGFX(void)
    {
      {
        auto cfg = _bus_instance.config();

        cfg.spi_host         = HSPI_HOST;// VSPI_HOST or HSPI_HOST (SPI2_HOST)
        cfg.spi_mode         = 0;
        cfg.freq_write       = SPI_FREQUENCY;
        cfg.freq_read        = SPI_READ_FREQUENCY;
        cfg.spi_3wire        = false;
        cfg.use_lock         = true;
        cfg.dma_channel      = 1; // Better to use 0 ?
        cfg.pin_sclk         = TFT_SCLK;
        cfg.pin_mosi         = TFT_MOSI;
        cfg.pin_miso         = TFT_MISO;
        cfg.pin_dc           = TFT_DC;

        _bus_instance.config(cfg);
        _panel_instance.setBus(&_bus_instance);
      }

      {
        auto cfg = _panel_instance.config();

        cfg.pin_cs           =    TFT_CS;
        cfg.pin_rst          =    TFT_RST;
        cfg.pin_busy         =    TFT_BUSY;
        cfg.memory_width     =    TFT_WIDTH;
        cfg.memory_height    =    TFT_HEIGHT;
        cfg.panel_width      =    TFT_WIDTH;
        cfg.panel_height     =    TFT_HEIGHT;
        cfg.offset_x         =     0;
        cfg.offset_y         =     0;
        cfg.offset_rotation  =     TFT_ROTATION;
        cfg.dummy_read_pixel =     8;
        cfg.dummy_read_bits  =     1;
        cfg.readable         = true;
        cfg.invert           = TFT_INVERSION_ON;
        cfg.rgb_order        = TFT_RGB_ORDER;
        cfg.dlen_16bit       = false;
        cfg.bus_shared       = false;

        _panel_instance.config(cfg);
      }

      {
        auto cfg = _light_instance.config();

        cfg.pin_bl = TFT_BL;
        cfg.invert = false;
        cfg.freq   = 44100;           
        cfg.pwm_channel = 7;

        _light_instance.config(cfg);
        _panel_instance.setLight(&_light_instance);
      }
      {
        auto cfg = _touch_instance.config();

        cfg.x_min      =  300;
        cfg.x_max      = 3900;
        cfg.y_min      =  200;
        cfg.y_max      = 3700;
        cfg.pin_int    = -1; //36 par defaut et -1 si marche pas
        cfg.bus_shared = false;
        cfg.offset_rotation = 1;

        cfg.spi_host = VSPI_HOST;
        cfg.freq = 1000000;
        cfg.pin_sclk = 25;
        cfg.pin_mosi = 32;
        cfg.pin_miso = 39;
        cfg.pin_cs   = 33;

        _touch_instance.config(cfg);
        _panel_instance.setTouch(&_touch_instance);
      }

      setPanel(&_panel_instance);
    }
};
```

closed 2024-05-26 by github-actions[bot]