lovyan03/LovyanGFX #547: Lilygo AMOLED series

Starbane, 2024-04-03:
**Carefully written requests are more likely to be given priority.**
**丁寧に記述された要望は優先して対応される可能性が高くなります。**
Lilygo have a fantastic lineup of beautiful AMOLED displayed ESP32-S3 with and without touch. They use display chip that is not supported by TFT_eSPI or AarduinoGFX, AdafruitGFX or LovyanGFX.

There is support for bit banging a TFT_eSprite to the display and that's it. It is slower than needed. All tree boards use QSPI and are fairly quick. With native support in LovyanGFX they would be super fast!.

Their software bundle is pretty good but restrictive since Lilygo more or less force you to use their display and touch library SensorLib and the implementations does not play nice with other implementations of SPI master and I2C/I2S.

Support for LovyanGFX on these boards would greatly increase the usability of the boards as well as be a good addition to the display chips supported by LovyanGFX

### Device Name (デバイスの名称・型番等)
Lilygo AMOLED 1.47 inch using display chip SH8501
Lilygo AMOLED 1.91 inch using display chip RM67162
Lilygo AMOLED 2.41 inch using display chip RM690B0
### URL of Device Specifications document (仕様書等のURL)
The documentation for the three boards can be found here on GitHUB:

https://github.com/Xinyuan-LilyGO/LilyGo-AMOLED-Series.git

### URL of the store where we can purchase (商品を購入できるURL)
https://www.lilygo.cc/products/t-display-s3-amoled (various models of 1.47 and 1.91 inch displays)
https://www.lilygo.cc/products/t4-s3 the 2.41 inch variant.

FeralAI, 2024-04-04:
I've done a bit of digging to see what it would take to support these modules, and the main piece missing right now is support for Quad SPI buses. All 3 of these displays use QSPI. If nobody else takes this on, it's likely something I'll be looking into very soon for one of my current projects.

I have the AMOLED Lite (1.47") and the T3 AMOLED Non-touch (1.91") versions in hand, and waiting on the the T3 AMOLED Touch (1.91") and T4 S3 (2.41") to ship from Lilygo.

Starbane, 2024-04-04:
I might be new to GitHub customs but this issue was a "Device support request" and #549 was a feature request so why close them? how would you differentiate between them?

Starbane, 2024-04-04:
> I've done a bit of digging to see what it would take to support these modules, and the main piece missing right now is support for Quad SPI buses. All 3 of these displays use QSPI. If nobody else [...]

I have a working embryo to QSPI driver. @nikthefix also has one very similar working over at thread https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED/issues/2#issue-1680748130

Starbane, 2024-04-05:
@FeralAI  you wanna collaborate on the QSPI integration into LovyanGFX?

nikthefix, 2024-04-05:
@Starbane 
Arduino_GFX (moononournation) recently added QSPI support in his library. His PDQ demo works well on the Lilygo T-Display-Amoled-1.91"

Not sure if he went for the 'frame buffer' approach or if there's tighter integration with the GFX components.
I'm still working on anti-tearing in my 'stand-alone' driver but almost finished now.
Give me a shout if you want to compare notes.

nik

moononournation, 2024-04-05:
yes, the canvas class is some sort of framebuffer

nikthefix, 2024-04-05:
@moononournation 
Yes indeed. I quite like the idea of detaching the GFX code from the display / bus drivers. I think it's simpler - like LVGL.
When the ESP-P4 arrives we'll have all the MIPI-DSI displays too! Modularity will help there I'm sure.

nik

Starbane, 2024-04-05:
I got my T4-S3 in the mail today and it broke my Bus/Driver but it should not since it is command compatible.

I got a set pixel function working on the 1.91 inch AMOLED but it is slow, 12 pixels/ms. It is mainly due to the lack of drawing capabilities of the chip. Not much more than memory write to use. and it is sequential over the address window :(.

Here is a code snippet from my lib for you to ponder, specially if it helps getting support on LovyanGFX :) I have a GitLab rep but will not publish it here for obvious reasons.

```C++
void RM67162::drawPixel(const int16_t x, const int16_t y, uint16_t color) {
        setAddrWindow(x,y,x,y);

        spi_transaction_ext_t t;
        memset(&t, 0, sizeof(t));
        t.base.flags = SPI_TRANS_MODE_QIO;
        t.base.cmd = 0x32 ;
        t.base.addr = 0x002C00;
        t.base.tx_buffer = &color;
        t.base.length = 16;
        _qspi.setCS();
        _qspi.pollingTransaction((spi_transaction_t *)&t);
        _qspi.clrCS();
    }
    ```
    _qspi is my bus class I am willing to collab and give access to my code on request only!

nikthefix, 2024-04-06:
@Starbane 
drawPixel() is discrete and slow. Try pushColors() for a continuous write to GRAM from a *buffer. 
ESP frame buffer needs to be in PSRAM for capacity.

nik

nikthefix, 2024-04-06:
@Starbane 
Here's an example for the 1.91" using just raw pushColors() - no GFX library:
[Lilygo_amoled_191_raw_push.zip](https://github.com/lovyan03/LovyanGFX/files/14893523/Lilygo_amoled_191_raw_push.zip)

Starbane, 2024-04-06:
> @Starbane Here's an example for the 1.91" using just raw pushColors() - no GFX library: [Lilygo_amoled_191_raw_push.zip](https://github.com/lovyan03/LovyanGFX/files/14893523/Lilygo_amoled_191_raw_push.zip)
> 
> Tested with ESP_Arduino Alpha 3.
> 
> Notes:
> 
> GPIO 38 needs to be OUTPUT HIGH for the touch version of the Lilygo 1.91" board (display_enable).
> 
> The drawing capabilities of the chip are never going to be a problem. The S3 is more than capable of rendering a full screen at 50fps in the case of the 1.91" display used here. It's just a matter of optimization. I'm working on it! :)
> 
> [Xinyuan-LilyGO/T-Display-S3-AMOLED#2 (comment)](https://github.com/Xinyuan-LilyGO/T-Display-S3-AMOLED/issues/2#issuecomment-2028337383)
> 
> The init codes for the T3-Amoled display are compatible with the T4 but are not optimal. Check the datasheet for the T4 display to see the differences.
> 
> nik

Well I can't get the T4 to even turn on the display I will paste in the output from my boot and perhaps if @nikthefix or @FeralAI  can spot something wrong?! Please remember that it is a T4 not the 1.91 inch. 1.91 actually works with it´s init sequence.

```
16:46:27.077 > --- lib/Devices/src/Bus/QSPI.cpp (75): configure D0 pin:   14
16:46:27.077 > --- lib/Devices/src/Bus/QSPI.cpp (76): configure D1 pin:   10
16:46:27.077 > --- lib/Devices/src/Bus/QSPI.cpp (77): configure D2 pin:   16
16:46:27.078 > --- lib/Devices/src/Bus/QSPI.cpp (78): configure D3 pin:   12
16:46:27.078 > --- lib/Devices/src/Bus/QSPI.cpp (79): configure D4 pin:   -1
16:46:27.078 > --- lib/Devices/src/Bus/QSPI.cpp (80): configure D5 pin:   -1
16:46:27.078 > --- lib/Devices/src/Bus/QSPI.cpp (81): configure D6 pin:   -1
16:46:27.079 > --- lib/Devices/src/Bus/QSPI.cpp (82): configure D7 pin:   -1
16:46:27.079 > --- lib/Devices/src/Bus/QSPI.cpp (83): configure SCK pin:  15
16:46:27.079 > --- lib/Devices/src/Bus/QSPI.cpp (84): configure RST pin:  -1
16:46:27.079 > --- lib/Devices/src/Bus/QSPI.cpp (85): configure CS pin:   11
16:46:27.080 > --- lib/Devices/src/Bus/QSPI.cpp (86): configure DC pin:   -1
16:46:27.080 > --- lib/Devices/src/Bus/QSPI.cpp (87): configure TE pin:   -1
16:46:27.080 > --- lib/Devices/src/Bus/QSPI.cpp (88): configure PMIC pin: 9
16:46:27.081 > --- lib/Devices/src/Bus/QSPI.cpp (89): configure Address bits:    24
16:46:27.081 > --- lib/Devices/src/Bus/QSPI.cpp (90): configure Command bits:    8
16:46:27.081 > --- lib/Devices/src/Bus/QSPI.cpp (91): configure Max Trans. Size: 262152
16:46:27.081 > --- lib/Devices/src/Bus/QSPI.cpp (92): configure Bus Speed:       36000000
16:46:27.082 > --- lib/Devices/src/Bus/QSPI.cpp (93): configure Bus Mode:        0
16:46:27.082 > --- lib/Devices/src/Bus/QSPI.cpp (94): configure Bus Queue Size:  17
16:46:27.082 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = fe00, data =  0x20, length = 1
16:46:27.083 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = 2600, data =  0xa, length = 1
16:46:27.083 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = 2400, data =  0x80, length = 1
16:46:27.084 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = 5a00, data =  0x51, length = 1
16:46:27.084 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = 5b00, data =  0x2e, length = 1
16:46:27.085 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = fe00, data =  0x0, length = 1
16:46:27.085 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = 3a00, data =  0x55, length = 1
16:46:27.086 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = c200, data =  0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x21 0x0 0x0 0x0 0x0 0x35 0x0 0x0 0x0 0x0 0x0 0x0 0x0, length = 21
16:46:27.097 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = 3500, data =  0x0, length = 1
16:46:27.097 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = 5100, data =  0x0, length = 1
16:46:27.098 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = 1100, data =  0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x80 0x0 0x0 0x0 0x0 0x29 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x20 0x0 0x0 0x0 0x0 0x51 0x0 0x0 0xff 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x0 0x0 0x3f 0xfe 0x6 0x3c 0x8 0x8a 0x0 0x42 0x0 0x0 0x0 0x0 0xf4 0x74 0xc9 0x3f 0xd 0x0 0x0 0x0 0x39 0x9d 0x0 0x82 0xe0 0xbd 0xce 0x3f 0xd0 0x74 0xc9 0x3f 0xea 0xbf 0xf1 0xfe 0xff 0xff 0xff 0xff 0x30 0xbc 0xce 0x3f 0x1c 0x0 0x0 0x0, length = 80
16:46:27.220 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = 2900, data =  0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x20 0x0 0x0 0x0 0x0 0x51 0x0 0x0 0xff 0x0 0x0 0x0, length = 20
16:46:27.231 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = 5100, data =  0xff, length = 1
16:46:27.231 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = 3600, data =  0x60, length = 1
16:46:27.528 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = 3600, data =  0x0, length = 1
16:46:27.560 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = 2a00, data =  0x0 0x0 0x2 0x57, length = 4
16:46:27.560 >     lib/Devices/src/Display/RM67162.cpp (124): sendCommand addr = 2b00, data =  0x0 0x0 0x1 0xc1, length = 4
```

[... cut ...]
#include <LilyGo/AMOLED_2.41_TOUCH.hpp>
#include <TFT_eSPI.h>

Display::Amoled display;
TFT_eSPI tft;
TFT_eSprite screen(&tft);

void setup() {
#ifdef USE_LOGGING
    Serial.begin(115200);

    logDebug("Waiting for Serial communication to establish...");
    delay(2000);
#endif

    display.begin();
    tft.begin();

    display.setRotation(1);

    screen.createSprite(display.width(), display.height());

    screen.fillSprite(TFT_BLUE);

    display.pushSprite(0,0,);
}

void loop(void) {
    delay(1000);
}
```

nikthefix, 2024-04-06:
@Starbane 
For the T4, GPIO pin 9 needs to be pulled HIGH for display enable.

nik

Starbane, 2024-04-06:
I do that as part of the initialization:

```
        if (_config.pmicen != -1) {
            if(digitalPinCanOutput(_config.pmicen)) {
                pinMode(_config.pmicen, OUTPUT);
                digitalWrite(_config.pmicen, HIGH);
            } else {
                logError("QSPI - Failed to initialize bus, PMICEn not a output pin");
                return DEVICE_BUS_CONFIG_ERROR;
            }
        }
```

So it has to be something different.

nikthefix, 2024-04-06:
@Starbane 
Does my test code work?

Starbane, 2024-04-06:
Yes it works. So something is off in my code. I suspect my QSPI bus driver is not working. but the thing is that it is working on the 1.91 inch board but not on the 2.41 inch board. That have me baffled!

Starbane, 2024-04-06:
If I include TFT_eSPI into you example it stops working just like mine code. So the line

```C++
#include <TFT_eSPI.h>
```

breaks your example for me

Starbane, 2024-04-06:
It seems that it is the order of the including that is the problem. As well as if I do tft.begin() on the TFT_eSPI.

I managed to get it to work but with a strange offset on the screen. According to my testing I have a screen that is 585 x 445 pixels since this code produces a one line box around the screen:

```C++
Display::Amoled display;
TFT_eSPI tft;
TFT_eSprite screen(&tft);

void setup() {
    display.begin();
//    tft.begin();

    screen.createSprite(display.width(), display.height());

    screen.fillSprite(TFT_WHITE);
    screen.fillRect(2,15,display.width()-4,display.height()-16, TFT_BLACK);
   
    display.pushSprite(0,0,screen);
}
```
When I try TFT_eSPI with your code I get the same results, the display is still 585x445 not as advertised 600x450
Anyone wanna try to educate me? (hint hint @lewisxhe)?

nikthefix, 2024-04-07:
@Starbane 
Check that your TFT_eSPI User_Setup.h display option is the default ILI9341_DRIVER. 
You'd think it shouldn't make a difference but I've found certain 'overrides' for each display profile which may explain why the #include order is critical in your experiments.

nik

jgauchia, 2024-12-31:
Hi, any news about it? 
I'm trying to use T4-S3 2.41 amoled with lovyangfx,  I search all issues, discussions, etc. but doesn't clear.

Thank you

jgauchia, 2024-12-31:
Thanks @tobozo ,

nikthefix, 2024-12-31:
@jgauchia Hi there. Happy New Year!
Lovyan doesn't have any examples supporting this display (last time I looked) but you can make it work by starting with a new Lovyan configuration using the QSPI bus and copying the pin assignments, display timings, init sequence and spi rate from either my dedicated driver or the Arduino_GFX library T4-S3 example. It's quite a bit of work but worth it if you want to stick with Lovyan. Don't forget to assert the display enable pin high and offset the y-axis (see above for details). I assume you have the T4 working OK with the supplied Lilygo files? If not then start there just to check the hardware is working properly. 

nik

jgauchia, 2024-12-31:
> @jgauchia Hi there. Happy New Year! Lovyan doesn't have any examples supporting this display (last time I looked) but you can make it work by starting with a new Lovyan configuration using the QSPI bus and copying the pin assignments, display timings, init sequence and spi rate from either my dedicated driver or the Arduino_GFX library T4-S3 example. It's quite a bit of work but worth it if you want to stick with Lovyan. Don't forget to assert the display enable pin high and offset the y-axis (see above for details). I assume you have the T4 working OK with the supplied Lilygo files? If not then start there just to check the hardware is working properly.
> 
> When I'm trying to port between graphics libraries I always start with the ubiquitous PDQ demo and worry about touch later.
> 
> nik

Hi @nikthefix , happy new year too!. (sorry for my english)
Yes Is that that I try to do .... add panel, display init, etc... and QSPI bus (from specific branch of lovyan). 
T4 works Ok, of course, 
I don't need full support of course, touch later, etc... 

Of course, I've limited in some programming skills, I've work with another displays (like SPI, RGB, etc...) but QSPI is new for me..

I'll continue working with this board and trying to understand how it's works...
Thanks to all , and sorry if this issue isn't appropiate.

nikthefix, 2024-12-31:
It's certainly appropriate. 

nik

tobozo, 2025-04-23:
@jgauchia @Starbane 

[partial support for RM690B0](https://github.com/lovyan03/LovyanGFX/blob/QSPI/src/lgfx/v1/panel/Panel_RM690B0.hpp) was added to the [QSPI branch](https://github.com/lovyan03/LovyanGFX/tree/QSPI) along with a [config example for Lilygo T4-S3](https://github.com/lovyan03/LovyanGFX/blob/QSPI/src/lgfx_user/Lilygo_T4_S3.hpp).

⚠ that display is cursed: it won't draw if x coordinates are odd, or if the width of the area is odd 😅

this limitation doesn't matter when LovyanGFX is used as a display driver only (e.g. LVGL + tft.pushImageDMA), or when the application repeatedly sends a buffer of the same size (e.g. camera stream)

for any other situation, it is recommended to enable the framebuffer to prevent most GFX functions from misbehaving.

next driver on the list is T-Display-S3 Amoled

jgauchia, 2025-04-23:
Cool! I'll try It soon! Thank you!

tobozo, 2025-04-24:
I've added a minimalistic touch driver to [Lilygo T4-S3 configuration example](https://github.com/lovyan03/LovyanGFX/blob/QSPI/src/lgfx_user/Lilygo_T4_S3.hpp)

jgauchia, 2025-04-24:
> I've added a minimalistic touch driver to [Lilygo T4-S3 configuration example](https://github.com/lovyan03/LovyanGFX/blob/QSPI/src/lgfx_user/Lilygo_T4_S3.hpp)

Ok, I've forked this branch, and try to test it this week, and give to you feedback about it.

nikthefix, 2025-04-24:
@tobozo 
Haha! The display is not cursed! (although it did feel like it at times to me).
The even / odd push requirements are well documented in the datasheet.
It's actually quite normal for these higher res displays.
Some of the common libs take it into account. Some do not.
LVGL allows you to implement a 'rounder' callback function which automatically quantizes your update region to meet these stipulations.
If using TFT_eSPI / Lovyan / Arduino_GFX with an unsupported display then you'll need to adapt your region update function to accommodate these requirements. It's easy to do and I suggest looking at the LVGL example rounder code which can be easily migrated to TFT_eSPI / Lovyan / Arduino_GFX. 

nik

jgauchia, 2025-04-24:
Hi @tobozo , sorry i'm trying to compile my project with QSPI branch and get this error (obviously i've limited knowledgment in coding :( )

`error: 'Bus_QSPI' in namespace 'lgfx' does not name a type`

jgauchia, 2025-04-24:
Hi @tobozo , sorry, I'm trying to compile my project with this panel, and raise this error:

`
lib/panel/LILYGO_T4_S3.hpp:18:9: error: 'Bus_QSPI' in namespace 'lgfx' does not name a type
   lgfx::Bus_QSPI      _bus_instance;
         ^~~~~~~~`
`lib/panel/LILYGO_T4_S3.hpp:19:9: error: 'Panel_RM690B0' in namespace 'lgfx' does not name a type
   lgfx::Panel_RM690B0 _panel_instance;
         ^~~~~~~~~~~~~
`
Obviously I've limited skills in coding, 🤕 , maybe it was a simple mistake

jgauchia, 2025-04-24:
Solved, added #define LGFX_USE_QSPI in panel settings.

tobozo, 2025-04-25:
@jgauchia thanks for reporting this, can you please share your platformio.ini, sketch and custom config if any?

jgauchia, 2025-04-25:
> [@jgauchia](https://github.com/jgauchia) thanks for reporting this, can you please share your platformio.ini, sketch and custom config if any?

yes, no problem.

 [platformio.ini](https://github.com/jgauchia/IceNav-v3/blob/a920ee5e440f9e51b115dec67bd1b6c8a30a0383/platformio.ini#L161)

[custom config panel](https://github.com/jgauchia/IceNav-v3/blob/devel_t4_s3/lib/panel/LILYGO_T4_S3.hpp)

[main project](https://github.com/jgauchia/IceNav-v3/tree/devel_t4_s3)

It's compile without problems with other panels, etc... 

Thank you.

tobozo, 2025-04-25:
Thanks for providing this information 👍 

The esp-idf version check has been [adjusted](https://github.com/jgauchia/LovyanGFX/compare/QSPI...lovyan03%3ALovyanGFX%3AQSPI), `LGFX_USE_QSPI` is now enabled when idf version is >= 4.3.x.

jgauchia, 2025-04-25:
> Thanks for providing this information 👍
> 
> The esp-idf version check has been [adjusted](https://github.com/jgauchia/LovyanGFX/compare/QSPI...lovyan03%3ALovyanGFX%3AQSPI), `LGFX_USE_QSPI` is now enabled when idf version is >= 4.4.x.
> 
> [edit] 4.4.x, not 4.3.x

No problem,Thank you for your effort.

You'll let me know when to sync the branch and test again.

tobozo, 2025-04-25:
you can sync anytime now

I just added support for [Lilygo T-Display-S3-AMOLED](https://github.com/lovyan03/LovyanGFX/blob/QSPI/src/lgfx_user/Lilygo_T_Display_S3_AMOLED.hpp)

jgauchia, 2025-04-25:
> you can sync anytime now
> 
> I just added support for [Lilygo T-Display-S3-AMOLED](https://github.com/lovyan03/LovyanGFX/blob/QSPI/src/lgfx_user/Lilygo_T_Display_S3_AMOLED.hpp)

Ok!, I'll try it  this afternoon and I'll give you feedback

jgauchia, 2025-04-25:
Firsts tests.... 

- I've changed **cfg.spi_host**  to **SPI3_HOST** now some graphics are displayed. 
- I've set to **HIGH** GPIO18 to enable T4-S3 Screen
- Touch doesn't works (I'm testing in my code)

tobozo, 2025-04-25:
oh right, are you using SPI2_HOST for the sdcard? works for me using either SPI3_HOST or SPI2_HOST so I guess it's project related

the touchscreen failing makes me wonder what other I2C device is wired on SDA/SCL, if so you may need to set `cfg.bus_shared` to `true` in touch configuration, otherwise an I2C scan should confirm that a device with id `0x5A` is present

jgauchia, 2025-04-25:
> oh right, are you using SPI2_HOST for the sdcard? works for me using either SPI3_HOST or SPI2_HOST so I guess it's project related
>
> the touchscreen failing makes me wonder what other I2C device is [...]

Yes I think that it's project related if you want delete my PR, sorry.
I'm configuring my project.
For the moment, all seems correct.
Thank you!

tobozo, 2025-04-26:
>  For the moment, all seems correct.

does it mean you got the touch working? 

I've tweaked the touch init for T4-S3, also implemented `tft.touch()->sleep()` and `tft.touch()->wakeup()`.

btw when reading touch data from LVGL you might prefer `getTouchRaw()` to `getTouch()`, it can populate up to 5 simultaneous touch points and their respective pressure.

jgauchia, 2025-04-26:
> does it mean you got the touch working?
>
> I've tweaked the touch init for T4-S3, also implemented `tft.touch()->sleep()` and `tft.touch()->wakeup()`.
>
> btw when reading touch data from LVGL you might [...]

Yes touch is working.
Interesting simultaneous touch points. It's available for other touch sensors?

On the other hand, it seems that the image is now correct with LVGL even when drawing a sprite directly with LovyanGFX without using the framebuffer. The only thing that is not quite right yet is drawPNG, but maybe I can find a solution by passing through a sprite and then drawing it on screen (just as I currently do with maps).

[image]

tobozo, 2025-04-26:
> It's available for other touch sensors?

It's available with CST226 (up to 5 points) so I've implemented it in the driver code, but I don't know for the other sensors.
 
`getTouchRaw()` will work wether the touch device supports multitouch or not (it'll only populate one point if multitouch isn't supported)

>  I can find a solution by passing through a sprite and then drawing it on screen

yup, an intermediate sprite is fine, but you'll still need to make sure the sprite-width and pushed-x coord are even.

jgauchia, 2025-04-26:
> It's available with CST226 (up to 5 points) so I've implemented it in the driver code, but I don't know for the other sensors.
> 
> `getTouchRaw()` will work wether the touch device supports multitouch or not (it'll only populate one point if multitouch isn't supported)

Ok! Perfect, I will look at how it works and test it.

> yup, an intermediate sprite is fine, but you'll still need to make sure the sprite-width and pushed-x coord are even.

Well it's strange, the framebuffer is disabled and it works perfectly, I'm only controlling the even coordinates in the LVGL round callback.

```
/**
 * @brief LVGL round callback for T4-S3
 *
 */
static void lv_rounder_cb(lv_event_t *event)
{
	lv_area_t *area = (lv_area_t*)lv_event_get_param(event);
	
  // make sure all coordinates are even
  if (area->x1 & 1)
    area->x1--;
  if (!(area->x2 & 1))
    area->x2++;
  if (area->y1 & 1)
    area->y1--;
  if (!(area->y2 & 1))
    area->y2++;
}

```
Thank you again. With this implementation of QSPI at the moment my project is working without problems. I just need to adjust the size of some assets.

brightproject, 2025-07-08:
> I just need to adjust the size of some assets.

Hello @jgauchia 🙂
You have been working with the `QSPI` bus for quite a long time, could you help me with connecting the `ST7716` display to the `LovyanGFX` library

> https://github.com/lovyan03/LovyanGFX/discussions/663#discussioncomment-13695004

jgauchia, 2025-07-08:
Oks no problem, I've a test branch of my project for QSPI. Remember my programming skills heheh

brightproject, 2025-07-08:
> Oks no problem, I've a test branch of my project for QSPI. Remember my programming skills heheh

These are my skills compared to yours ho-ho.
I didn't see the `QSPI` branch in your repositories.
Have you worked with the `ST77916` display yet?