lovyan03/LovyanGFX #552: Expose XPT2046 ability to measure VBAT

garethhcoleman, 2024-04-08:
I am the co-designer of the unphone ([Device](https://unphone.net/) [Shop](https://shop.pimoroni.com/products/unphone)) which is open source hardware and software.

We use the [XPT2046](https://datasheetspdf.com/pdf-file/746665/XPTEK/XPT2046/1) touchscreen already supported by this code, very nicely done thank you!

We would like to expose the XPT2046 battery measurement capability in the LovyanGFX drivers for ourselves and other users.

I have some candidate functions that I will introduce in a PR for your consideration. Please just let me know any improvements or changes I should make or if my approach should be modified.

Thank you again for all your work creating this library!

tobozo, 2024-04-08:
hi,

"battery capacity" sounds a bit out of scope for a touch driver, implementing a more generic method would make more sense e.g.

```cpp
int16_t Touch_XPT2046::readRegister( uint8_t register );
```  

then you can add you own dedicated temp/bat/aux reading functions to your custom driver implementation:

```cpp
class LGFX : public lgfx::LGFX_Device
{
public:

  lgfx::Bus_Whatever   _bus_instance;
  lgfx::Panel_Blah     _panel_instance;
  lgfx::Touch_XPT2046  _touch_instance;

  LGFX(void)
  {
    {
      auto cfg = _panel_instance.config();
      // ... panel config here
      _panel_instance.config(cfg);
    }
    {
      auto cfg = _bus_instance.config();
      cfg.panel = &_panel_instance;
      // ... panel bus config here
      _bus_instance.config(cfg);
    }
    _panel_instance.setBus(&_bus_instance);

    {
      auto cfg = _touch_instance.config();
      // ... touch config here
      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);
    }

    setPanel(&_panel_instance);
  }

  // custom functions here

  int16_t getTouchRegister( uint8_t register ) 
  {
    return _touch_instance.readRegister( register );
  }

  // specialized functions based on getTouchRegister()
  float getVBat() { }
  float getAuxIn() { }
  float getTemp() { }
  float getTempF() { }

};

```

garethhcoleman, 2024-04-08:
Yes I agree it is a bit odd but the xpt2046 supports it and we've used it...

I greatly appreciate your quick response and advice.

I will work on the approach you suggested and let you know how it goes

Many thanks again

garethhcoleman, 2024-05-08:
Thanks for the bump! I have been really struggling adapting the read register function from using esp spi functions to using the ones in this library. I'll carry on trying

vednildip, 2024-12-13:
error: 'HSPI_HOST' was not declared in this scope; did you mean 'SPI3_HOST'?
   73 |       cfg.spi_host = HSPI_HOST;// 使用するSPIを選択 (HSPI_HOST or VSPI_HOST