lovyan03/LovyanGFX #558: Failing to build after updating platformio

metehus, 2024-04-23:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  [ESP32-S3 N16R8]
- Panel Driver IC:    [GC9A01]
- Bus type:           [SPI]
- LovyanGFX version:  [latest/v1.1.12]
- FrameWork version:   [ESP-IDF v5.2.1]
- Build Environment:   [PlatformIO]
- Operating System:   [Windows]

## Problem Description ( 問題の内容 )
I am not able to build a simple sketch using LovyanGFX after updating the platformio tools

### Expected Behavior ( 期待される動作 )
The project to build

### Actual Behavior ( 実際の動作 )
The project fails to build with this message:
<details><summary>Click to show log output</summary>
<p>

```
PLATFORM: Espressif 32 (6.6.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50201.0 (5.2.1)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 13.2.0+20230928
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
Reading CMake configuration...
Generating assembly for certificate bundle...
Warning! Flash memory size mismatch detected. Expected 8MB, found 2MB!
Please select a proper value in your `sdkconfig.defaults` or via the `menuconfig` target!
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- LovyanGFX @ 1.1.12
Building in release mode
Compiling .pio\build\esp32-s3-devkitc-1\src\main.o
[ hided for better clarity ]
Compiling .pio\build\esp32-s3-devkitc-1\lib08e\LovyanGFX\lgfx\v1\platforms\rp2040\Light_PWM.o
.pio/libdeps/esp32-s3-devkitc-1/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_SPI.cpp:26: warning: "REG_SPI_BASE" redefined
   26 |   #define REG_SPI_BASE(i)   (DR_REG_SPI1_BASE + (((i)>1) ? (((i)* 0x1000) + 0x20000) : (((~(i)) & 1)* 0x1000 )))
      |
In file included from C:/Users/methi/.platformio/packages/framework-espidf/components/esp_hw_support/include/esp_memory_utils.h:12,
                 from C:/Users/methi/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:54,
                 from C:/Users/methi/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:57,
                 from C:/Users/methi/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:69,
                 from C:/Users/methi/.platformio/packages/framework-espidf/components/driver/include/esp_private/spi_common_internal.h:13,
                 from .pio/libdeps/esp32-s3-devkitc-1/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_SPI.hpp:36,
                 from .pio/libdeps/esp32-s3-devkitc-1/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_SPI.cpp:21:
C:/Users/methi/.platformio/packages/framework-espidf/components/soc/esp32s3/include/soc/soc.h:37: note: this is the location of the previous definition
   37 | #define REG_SPI_BASE(i)         (((i)>=2) ? (DR_REG_SPI2_BASE + (i-2) * 0x1000) : (0))    // GPSPI2 and GPSPI3
      |
Compiling .pio\build\esp32-s3-devkitc-1\lib08e\LovyanGFX\lgfx\v1\platforms\rp2040\common.o
Compiling .pio\build\esp32-s3-devkitc-1\lib08e\LovyanGFX\lgfx\v1\platforms\samd21\Bus_I2C.o
Compiling .pio\build\esp32-s3-devkitc-1\lib08e\LovyanGFX\lgfx\v1\platforms\samd21\Bus_SPI.o
[... 73 lines cut ...]
2. Create a new project and select ESP32-S3-DevkitC with espidf framework
3. Try to build the project (`pio run`). Should work (I had to rename the `void main` function to `void app_main` to make it work first)
4. Add the LovyanGFX library (`pio lib install LovyanGFX`)
5. Try to build again. Should give an error and fail

### Code to reproduce this issue ( 再現させるためのコード )

**Please submit complete source code that can reproduce your problem.**
**あなたの問題を再現できる完全なソースコードを提示してください。**

I haven't done anything on the main file:
```cpp
extern "C" void app_main() {

}
```

My platformio.ini file:
```ini
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = espidf
lib_deps = lovyan03/LovyanGFX@^1.1.12
```

metehus, 2024-04-23:
nvm just looked at #438 and tried the develop branch and now it works! Is there a date of when this version will be released on the platformio library?

tobozo, 2024-04-23:
hi,

the [partnership between Espressif and M5Stack](https://www.espressif.com/en/news/Espressif_Acquires_M5Stack) is going to influence how the maintenance is balanced between LovyanGFX and M5Unified, so there's no forecast yet for the next release but there will be an [announcement](https://twitter.com/lovyan03); until then your best option is to use the develop branch in your platformio.ini file

closing this as fixed; feel free to reopen for any platformio related issue