lovyan03/LovyanGFX #565: Color Inversion Not Working Correctly with ST7789 Display.

alpha815, 2024-05-13:
## Environment ( 実行環境 )
- MCU or Board name:  TTGO T-WATCH V2
- Panel Driver IC:    ST7789
- Bus type:           SPI
- LovyanGFX version:  LovyanGFX@1.1.12
- FrameWork version:   [ArduinoESP32 v2.0.3]
- Build Environment:   [PlatformIO]
- Operating System:   [Linux]

## Problem Description ( 問題の内容 )
I am encountering an issue with color inversion on an ST7789 TFT LCD display when using the this library on an ESP32 microcontroller. Despite trying to configure the cfg.rgb_order setting (false or true), the display's color inversion behavior does not change as expected.
### Expected Behavior ( 期待される動作 )
I expected that setting cfg.rgb_order to false or true would control the color inversion behavior of the display, but this does not appear to have any effect. The display should respond to this setting by correctly inverting colors as specified.
### Actual Behavior ( 実際の動作 )
The display's color inversion remains static and does not change when adjusting the cfg.rgb_order settings initialization. This behavior persists regardless of the value (false or true) assigned to cfg.rgb_order.
### Steps to reproduce ( 再現のための前提条件 )
Set up an ST7789 TFT display with ESP32 microcontroller.
Use the library to initialize the display.
Attempt to configure color inversion by setting cfg.rgb_order to false or true in the initialization code.
Observe that the color inversion behavior remains unchanged regardless of the cfg.rgb_order setting.
// If possible, attach a picture of your setup/wiring here.
[image]
[image]

### Code to reproduce this issue ( 再現させるためのコード )

**Please submit complete source code that can reproduce your problem.**
**あなたの問題を再現できる完全なソースコードを提示してください。**

```cpp

#define LGFX_USE_V1
#include "LovyanGFX.h"

class LGFX : public lgfx::LGFX_Device
{

    lgfx::Panel_ST7789 _panel_instance;
    lgfx::Bus_SPI _bus_instance;
    lgfx::Light_PWM _light_instance;
    lgfx::Touch_FT5x06 _touch_instance;

public:
    LGFX(void)
    {
        {
            auto cfg = _bus_instance.config();
            cfg.spi_host = VSPI_HOST;
            cfg.spi_mode = 0;
            cfg.freq_write = 40000000;
            cfg.freq_read = 16000000;
            cfg.spi_3wire = false;
            cfg.use_lock = true;
            cfg.dma_channel = SPI_DMA_CH_AUTO;
            cfg.pin_sclk = TWATCH_TFT_SCLK;
            cfg.pin_mosi = TWATCH_TFT_MOSI;
            cfg.pin_miso = TWATCH_TFT_MISO;
            cfg.pin_dc = TWATCH_TFT_DC;
            _bus_instance.config(cfg);
            _panel_instance.setBus(&_bus_instance);
        }

        {
            auto cfg = _panel_instance.config();
            cfg.pin_cs = TWATCH_TFT_CS;
            cfg.pin_rst = TWATCH_TFT_RST;
            cfg.pin_busy = -1;
            cfg.panel_width = 240;
            cfg.panel_height = 240;
            cfg.offset_x = 0;
            cfg.offset_y = 0;
            cfg.offset_rotation = 0;
            cfg.dummy_read_pixel = 8;
            cfg.dummy_read_bits = 1;
            cfg.readable = false;
            cfg.invert = false;
            cfg.rgb_order = false;
            cfg.dlen_16bit = false;
            cfg.bus_shared = false;
            _panel_instance.config(cfg);
        }

        {
            auto cfg = _light_instance.config();
            cfg.pin_bl = TWATCH_TFT_BL;
            cfg.invert = false;
            cfg.freq = 44100;
            cfg.pwm_channel = 7;
            _light_instance.config(cfg);
            _panel_instance.setLight(&_light_instance);
        }

        {
            auto cfg = _touch_instance.config();
            cfg.x_min = 0;
            cfg.x_max = 239;
            cfg.y_min = 0;
            cfg.y_max = 239;
            cfg.pin_int = TOUCH_INT;
            cfg.bus_shared = true;
            cfg.offset_rotation = 0;
            cfg.i2c_port = 1;
            cfg.i2c_addr = 0x38;
            cfg.pin_sda = TOUCH_SDA;
            cfg.pin_scl = TOUCH_SCL;
            cfg.freq = 400000;
            _touch_instance.config(cfg);
            _panel_instance.setTouch(&_touch_instance);
        }
        setPanel(&_panel_instance);
    }
};

LGFX tft;

setup(){

  tft.begin();
  tft.fillScreen(TFT_BLACK);
  delay(1000);
  tft.fillScreen(TFT_RED);
  delay(1000);
  tft.fillScreen(TFT_GREEN);
  delay(1000);
  tft.fillScreen(TFT_BLUE);
  delay(1000);
  tft.fillScreen(TFT_BLACK);
  delay(1000);
  tft.setCursor(0, 0);
  tft.print("hello, World");

}
void loop(){}
```

I would like to extend my sincere thanks to the developers of this library for their outstanding work in creating such a versatile and powerful library. Your efforts have greatly contributed to the success of many projects, including this one. Your dedication to open-source development is truly appreciated.
Any assistance in resolving this color inversion problem with the ST7789 display and this library would be greatly appreciated. Thank you!

tobozo, 2024-05-13:
hi,

`cfg.invert` is probably what you're looking for, `cfg.rgb_order` doesn't invert colors but toggles bgr / rgb

alpha815, 2024-05-13:
> hi,
> 
> `cfg.invert` is probably what you're looking for, `cfg.rgb_order` doesn't invert colors but toggles bgr / rgb

Damn, that worked very well. Thank you.

closed 2024-05-13 by alpha815