lovyan03/LovyanGFX #566: build  error with cross-compiler when using SDL2

nnn112358, 2024-05-20:
## Environment ( 実行環境 )
- MCU or Board name:  CoreMP135
- Panel Driver IC:   other
- Bus type:          other
- LovyanGFX version: latest develop
- FrameWork version:  other
- Build Environment:  gcc-arm-linux-gnueabihf
- Operating System:   Ubuntu-2204

## Problem Description ( 問題の内容 )
フレームバッファ機能を使ってLCDを描画するコードにて、Linuxでクロスコンパイラ(gcc-arm-linux-gnueabihf)を使ってビルドを行う場合に、libsdl2-devがインストールされている環境ではmake時にエラー発生する。

詳細；https://github.com/nnn112358/CoreMP135_LovyanGFX_test

### Expected Behavior ( 期待される動作 )
フレームバッファ機能を使ってLCDを描画するコードをビルドする場合に、SDLライブラリのファイルを参照しないこと。

### Steps to reproduce ( 再現のための前提条件 )

```
$ sudo apt install gcc-arm-linux-gnueabihf
$ git clone https://github.com/nnn112358/CoreMP135_LovyanGFX_test
$ cd CoreMP135_LovyanGFX_test
$ git clone https://github.com/lovyan03/LovyanGFX.git
$ sudo apt install libsdl2-dev
$ cd CoreMP135_LovyanGFX_test
$ mkdir build
$ cmake ..
$ make
```

```
$ make
[ 95%] Building CXX object CMakeFiles/coremp135_LovyanGFX.dir/coremp135_touchpanel.cpp.o
In file included from /usr/include/SDL2/SDL_stdinc.h:31,
                 from /usr/include/SDL2/SDL_main.h:25,
                 from /opt/LinuxHome/CoreMP135/CoreMP135_LovyanGFX_test/LovyanGFX/src/lgfx/v1/platforms/sdl/common.hpp:32,
                 from /opt/LinuxHome/CoreMP135/CoreMP135_LovyanGFX_test/LovyanGFX/src/lgfx/v1/platforms/sdl/Panel_sdl.hpp:23,
                 from /opt/LinuxHome/CoreMP135/CoreMP135_LovyanGFX_test/LovyanGFX/src/lgfx/v1/platforms/device.hpp:94,
                 from /opt/LinuxHome/CoreMP135/CoreMP135_LovyanGFX_test/LovyanGFX/src/lgfx/v1_init.hpp:22,
                 from /opt/LinuxHome/CoreMP135/CoreMP135_LovyanGFX_test/LovyanGFX/src/LovyanGFX.hpp:31,
                 from /opt/LinuxHome/CoreMP135/CoreMP135_LovyanGFX_test/coremp135_touchpanel.cpp:4:
/usr/include/SDL2/SDL_config.h:4:10: fatal error: SDL2/_real_SDL_config.h: No such file or directory
    4 | #include <SDL2/_real_SDL_config.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/coremp135_LovyanGFX.dir/build.make:370: CMakeFiles/coremp135_LovyanGFX.dir/coremp135_touchpanel.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/coremp135_LovyanGFX.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

### Code to reproduce this issue ( 再現させるためのコード )

詳細；https://github.com/nnn112358/CoreMP135_LovyanGFX_test

lovyan03, 2024-05-20:
ご連絡ありがとうございます！

確認なのですが、これは『Linux環境かつSDLのヘッダが存在する状況において、SDLは要らないのでFrameBufferを使わせて欲しい』
という認識で合っていますでしょうか？

nnn112358, 2024-05-20:
はい、そうです。
Linux環境かつSDLのヘッダが存在する状況において、SDLは要らないのでFrameBufferを使わせて欲しい、です。

lovyan03, 2024-05-20:
なるほど、理解しました。
そうしますと…LinuxかつSDLが使いたい場合との使い分けをどうするか考慮しておく必要がありますね。
例えばLGFX側で 『コンパイルオプションに -DLGFX_DISABLE_SDL を付けるとSDLを使わない動作になる』
という対策を追加する、という案ではご要望にかないますでしょうか？

nnn112358, 2024-05-20:
>>『コンパイルオプションに -DLGFX_DISABLE_SDL を付けるとSDLを使わない動作になる』
これで、期待される動作には十分だと思います。

closed 2024-06-26 by github-actions[bot]