lovyan03/LovyanGFX #575: Support ST7701s IPS TFT 5 inch 854x480 RGB+SPI With ESP32-S3

Neothai, 2024-06-07:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  ESP32-S3
- Panel Driver IC:   ST7701s Driver with GT911(Touch Driver IC)
- Bus type:           RGB+SPI
- LovyanGFX version:  Latest
- FrameWork version:   ArduinoESP32 v2.0.17
- Build Environment:   ArduinoIDE
- Operating System:   Windows 11

## Problem Description ( 問題の内容 )
Display details: ST7701s Driver IC 5inch 854*480 pixel IPS RGB+SPI Interface | Support RGB565, RGB666 and RGB888 Color
[2388602012亦亞徽EP5008S-DCT规格书.pdf](https://github.com/user-attachments/files/15710030/2388602012.EP5008S-DCT.pdf) <--Datasheet
https://item.taobao.com/item.htm?id=750263137747 <-- I bought it from here.
https://drive.google.com/file/d/1A29h3aU8qUe2CzuJNajd-pa8o6-KBBN1/view?usp=sharing <-- Sample code from seller
[image]
[image]

### Problem: The image on the display does not display properly. distorted and some parts missing

[image]

https://github.com/lovyan03/LovyanGFX/assets/123863242/a4b13c87-4761-4d22-89d0-dd6b247b7b24

### Expected Behavior ( 期待される動作 )
Want the display to display normally Not distorted and missing
### Actual Behavior ( 実際の動作 )
The image on the display does not display properly. distorted and some parts missing

### Code to reproduce this issue ( 再現させるためのコード )

**Please submit complete source code that can reproduce your problem.**
**あなたの問題を再現できる完全なソースコードを提示してください。**

```cpp
#include <LovyanGFX.hpp>
#include <lgfx/v1/platforms/esp32s3/Panel_RGB.hpp>
#include <lgfx/v1/platforms/esp32s3/Bus_RGB.hpp>
  
  class LGFX : public lgfx::LGFX_Device{
    
      lgfx::Touch_GT911 _touch_instance;

      lgfx::Bus_RGB      _bus_instance;
      lgfx::Panel_ST7701 _panel_instance;

    public:
      LGFX(void){
          {
            auto cfg = _panel_instance.config();
            cfg.memory_width  = 854;
            cfg.memory_height = 480;
            cfg.panel_width  = 854;
            cfg.panel_height = 480;

            cfg.offset_x = 0;
            cfg.offset_y = 0;

            _panel_instance.config(cfg);
          }
          {
            auto cfg = _panel_instance.config_detail();
            cfg.pin_cs = 42; 
            cfg.pin_mosi = 11;
            cfg.pin_sclk = 12;
            _panel_instance.config_detail(cfg);
          }
[... 758 lines cut ...]
  }

  return micros() - start;
}

uint32_t testFilledRoundRects()
{
  uint32_t start;
  int32_t i, i2;
  int32_t cx = tft.width() / 2 - 1;
  int32_t cy = tft.height() / 2 - 1;

  tft.fillScreen(TFT_BLACK);

  start = micros_start();

  for (i = std::min(tft.width(), tft.height()); i > 20; i -= 6)
  {
    i2 = i / 2;
    tft.fillRoundRect(cx-i2, cy-i2, i, i, i/8, tft.color565(0, i, 0));
  }

  return micros() - start;
}
```

tobozo, 2024-06-07:
hi,

given the square-ish shape it looks like `cfg.offset_rotation = 1` (or another value up to 3) is needed for the panel config

hscync/vsync values seem to need adjustment too, but first fix the rotation

Neothai, 2024-06-07:
Thank you for your advice, but setting cfg.offset_rotation from 0-3 still doesn't work.

```cpp
...
cfg.offset_x = 0;
cfg.offset_y = 0;
cfg.offset_rotation = 0; // <- I've tried entering values ​​from 0-3, but still doesn't work!
```
If you need information about this display and other information, It's already on top.

What should I do next?

tobozo, 2024-06-07:
> but still doesn't work!

this is an issue tracker, not a customer service, the quality of the support you get is as good as the feedback you give

Neothai, 2024-06-07:
Here is a picture of the results I got.

`cfg.offset_rotation = 0`

[image]
---
`cfg.offset_rotation = 1`

[image]
---
`cfg.offset_rotation = 2`

[image]
---
`cfg.offset_rotation = 3`

[image]

tobozo, 2024-06-07:
thanks,

offset rotation 1 and 3 don't show the cropped square, so there's progress, I suggest you keep `cfg.offset_rotation=1` for the meantime

I don't read Chinese and can't make much of the datasheet but it also looks like hsync*/vsync* values need to be adjusted with values provided by the vendor:

```cpp
            cfg.hsync_front_porch = 20;//10;
            cfg.hsync_pulse_width = 1;//8;
            cfg.hsync_back_porch  = 42;//50;
            
            cfg.vsync_front_porch = 4;//10;
            cfg.vsync_pulse_width = 10;//8;
            cfg.vsync_back_porch  = 12;//20;
```

> https://drive.google.com/file/d/1A29h3aU8qUe2CzuJNajd-pa8o6-KBBN1/view?usp=sharing <-- Sample code from seller

unfortunately the google drive zip file is invalid and/or infected and I can't compare your values with the recommended settings

```
error: invalid zip file with overlapped components (possible zip bomb)
 To unzip the file anyway, rerun the command with UNZIP_DISABLE_ZIPBOMB_DETECTION=TRUE environmnent variable
```

- did you manage to unzip the zip archive? 
- does the sample code work with your display?

if the answer is yes to both questions please share the code of that sample ( try https://gist.new ) so we can compare the configurations

Neothai, 2024-06-07:
> unfortunately the google drive zip file is invalid and/or infected and I can't compare your values with the recommended settings

I edited the file (on the original link) so I expect you should be able to open it normally.

> did you manage to unzip the zip archive?

Yes, I unzip the zip archive and I have access to all the files normally.

> does the sample code work with your display?

I have not yet tried out the code I received. Because the example the seller sent me has the project format used in ESP-IDF (the seller said there is no example code for Arduino IDE) and now I am developing my project on Arduino IDE(ArduinoESP32 V2 .0.17) only

tobozo, 2024-06-07:
esp-idf calls this variant LCD_5r0_480x854, so you can try to invert the height/width values

            cfg.memory_width  = 480
            cfg.memory_height = 854;
            cfg.panel_width  = 480;
            cfg.panel_height = 854;

you may have to adjust `cfg.offset_rotation` again during your tests though

Neothai, 2024-06-07:
> esp-idf calls this variant LCD_5r0_480x854, so you can try to invert the height/width values

The results look much better than before! And this is an illustration.

`cfg.offset_rotation = 0`

[image]

---
`cfg.offset_rotation = 1`

[image]

---
`cfg.offset_rotation = 2`

[image]

---
`cfg.offset_rotation = 3`

[image]

tobozo, 2024-06-07:
could it be that pin_vsync and  pin_hsync need to inverted too?

```
            cfg.pin_vsync   = 17;
            cfg.pin_hsync   = 47;
```

Neothai, 2024-06-07:
> could it be that pin_vsync and pin_hsync need to inverted too?

I tested it and the result was that nothing was displayed on the screen and everything was black.

---

Could the incorrect display of the display be related to the ST7701s init command?
I noticed that this display has SPI+RGB communication.

tobozo, 2024-06-07:
[init commands sequences are hardcoded](https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx/v1/platforms/esp32s3/Panel_RGB.cpp#L237) for this panel, but there is another variant of `Panel_ST7701` that you may want to try: `Panel_ST7701_guition_esp32_4848S040`

Neothai, 2024-06-08:
> that you may want to try: `Panel_ST7701_guition_esp32_4848S040`

This is the result
[image]

tobozo, 2024-06-10:
colors can be adjusted with `cfg.invert` and `cfg.rgb_order` (both booleans) options

that deinterlacing effect is similar to what you had with the `Panel_ST7701` with `cfg.offset_rotation = 2` so the init sequence is probably not the cause and I'm out of ideas

closed 2024-07-17 by github-actions[bot]