lovyan03/LovyanGFX #577: ST7789 cannot get it to work

phhoef, 2024-06-15:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  ESP32-S3-WROOM-1-N16R8
- Panel Driver IC:    ST7789
- Bus type:           SPI
- LovyanGFX version:  v1.1.16
- FrameWork version:   ArduinoESP32 v2.0.3
- Build Environment:   PlatformIO
- Operating System:   macOS

## Problem Description ( 問題の内容 )
I am using TFT_eSPI with LVGL currently. This is working, so I am pretty sure the wiring is correct.
Now, I would like to switch to LovyanGFX and I cannot get it work.
There is no touch, I just want to draw on the display.

The display is just black and nothing happens (though, I've enabled backlight).
ESP32 is not crashing.

### Code to reproduce this issue ( 再現させるためのコード )

**Please submit complete source code that can reproduce your problem.**
**あなたの問題を再現できる完全なソースコードを提示してください。**

```cpp

#include <LovyanGFX.hpp>

class LGVX : public lgfx::LGFX_Device
{
    lgfx::Panel_ST7789 _panel_instance;

    // パネルを接続するバスの種類にあったインスタンスを用意します。
    lgfx::Bus_SPI _bus_instance; // SPIバスのインスタンス
                                 // lgfx::Bus_I2C        _bus_instance;   // I2Cバスのインスタンス
    // lgfx::Bus_Parallel8  _bus_instance;   // 8ビットパラレルバスのインスタンス

public:
    // コンストラクタを作成し、ここで各種設定を行います。
    // クラス名を変更した場合はコンストラクタも同じ名前を指定してください。
    LGVX(void)
    {
        {                                      // バス制御の設定を行います。
            auto cfg = _bus_instance.config(); // バス設定用の構造体を取得します。

            // SPIバスの設定
            cfg.spi_host = SPI3_HOST; // 使用するSPIを選択  ESP32-S2,C3 : SPI2_HOST or SPI3_HOST / ESP32 : VSPI_HOST or HSPI_HOST
            // ※ ESP-IDFバージョンアップに伴い、VSPI_HOST , HSPI_HOSTの記述は非推奨になるため、エラーが出る場合は代わりにSPI2_HOST , SPI3_HOSTを使用してください。
            cfg.spi_mode = 0;                  // SPI通信モードを設定 (0 ~ 3)
            cfg.freq_write = 40000000;         // 送信時のSPIクロック (最大80MHz, 80MHzを整数で割った値に丸められます)
            cfg.freq_read = 16000000;          // 受信時のSPIクロック
            cfg.spi_3wire = true;              // 受信をMOSIピンで行う場合はtrueを設定
            cfg.use_lock = true;               // トランザクションロックを使用する場合はtrueを設定
            cfg.dma_channel = SPI_DMA_CH_AUTO; // 使用するDMAチャンネルを設定 (0=DMA不使用 / 1=1ch / 2=ch / SPI_DMA_CH_AUTO=自動設定)
            // ※ ESP-IDFバージョンアップに伴い、DMAチャンネルはSPI_DMA_CH_AUTO(自動設定)が推奨になりました。1ch,2chの指定は非推奨になります。
            cfg.pin_sclk = 36; // SPIのSCLKピン番号を設定
            cfg.pin_mosi = 35; // SPIのMOSIピン番号を設定
            cfg.pin_miso = -1; // SPIのMISOピン番号を設定 (-1 = disable)
            cfg.pin_dc = -1;   // SPIのD/Cピン番号を設定  (-1 = disable)
                               // SDカードと共通のSPIバスを使う場合、MISOは省略せず必ず設定してください。
                               //*/

            _bus_instance.config(cfg);              // 設定値をバスに反映します。
            _panel_instance.setBus(&_bus_instance); // バスをパネルにセットします。
        }

        {                                        // 表示パネル制御の設定を行います。
            auto cfg = _panel_instance.config(); // 表示パネル設定用の構造体を取得します。

            cfg.pin_cs = 19;   // CSが接続されているピン番号   (-1 = disable)
            cfg.pin_rst = 21;  // RSTが接続されているピン番号  (-1 = disable)
            cfg.pin_busy = -1; // BUSYが接続されているピン番号 (-1 = disable)

            // ※ 以下の設定値はパネル毎に一般的な初期値が設定されていますので、不明な項目はコメントアウトして試してみてください。

            cfg.panel_width = 240;    // 実際に表示可能な幅
            cfg.panel_height = 320;   // 実際に表示可能な高さ
            cfg.offset_x = 0;         // パネルのX方向オフセット量
            cfg.offset_y = 0;         // パネルのY方向オフセット量
            cfg.offset_rotation = 0;  // 回転方向の値のオフセット 0~7 (4~7は上下反転)
            cfg.dummy_read_pixel = 8; // ピクセル読出し前のダミーリードのビット数
            cfg.dummy_read_bits = 1;  // ピクセル以外のデータ読出し前のダミーリードのビット数
            cfg.readable = true;      // データ読出しが可能な場合 trueに設定
            cfg.invert = false;       // パネルの明暗が反転してしまう場合 trueに設定
            cfg.rgb_order = false;    // パネルの赤と青が入れ替わってしまう場合 trueに設定
            cfg.dlen_16bit = false;   // 16bitパラレルやSPIでデータ長を16bit単位で送信するパネルの場合 trueに設定
            cfg.bus_shared = true;    // SDカードとバスを共有している場合 trueに設定(drawJpgFile等でバス制御を行います)

            // 以下はST7735やILI9163のようにピクセル数が可変のドライバで表示がずれる場合にのみ設定してください。
            //    cfg.memory_width     =   240;  // ドライバICがサポートしている最大の幅
            //    cfg.memory_height    =   320;  // ドライバICがサポートしている最大の高さ

            _panel_instance.config(cfg);
        }

        setPanel(&_panel_instance); // 使用するパネルをセットします。
    }
};

void setup()
{
  LGVX *tft = new LGVX();
  tft->init();
  tft->setRotation(1);
  tft->initDMA();

  tft->clear();
  tft->fillRect(30, 30, 100, 100, tft->color565(255, 0, 0));
}
```

tobozo, 2024-06-15:
hi,

this configuration is incomplete (missing Light_PWM and Touch_* instances), please refer to the [user_settings](https://github.com/lovyan03/LovyanGFX/tree/master/examples/HowToUse/2_user_setting) example.

also if you need help converting an existing TFT_eSPI configuration to LGFX configuration, make sure to provide a working TFT_eSPI config file as a reference, because reading your LovyanGFX config block as it is isn't enough to find any pinout or bus problem.

phhoef, 2024-06-15:
thanks for your thoughts.
I've removed the `Light_PWM` and `Touch_` sections, as I understood, that if not needed I should remove it like described in the comments. Is this wrong? 
https://github.com/lovyan03/LovyanGFX/blob/50a1800eb771ff8c028334d055ad37e1f12ec863/examples/HowToUse/2_user_setting/2_user_setting.ino#L55
https://github.com/lovyan03/LovyanGFX/blob/50a1800eb771ff8c028334d055ad37e1f12ec863/examples/HowToUse/2_user_setting/2_user_setting.ino#L52

This is the working `User_Setup.h` file from TFT_eSPI.
```cpp
//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc.
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

// User defined information reported by "Read_User_Setup" test & diagnostics example
#define USER_SETUP_INFO "User_Setup"

// Define to disable all #warnings in library (can be put in User_Setup_Select.h)
// #define DISABLE_ALL_LIBRARY_WARNINGS

// ##################################################################################
//
// Section 1. Call up the right driver file and any options for it
//
// ##################################################################################

// Define STM32 to invoke optimised processor support (only for STM32)
// #define STM32

// Defining the STM32 board allows the library to optimise the performance
// for UNO compatible "MCUfriend" style shields
// #define NUCLEO_64_TFT
// #define NUCLEO_144_TFT

// STM32 8-bit parallel only:
// If STN32 Port A or B pins 0-7 are used for 8-bit parallel data bus bits 0-7
// then this will improve rendering performance by a factor of ~8x
// #define STM_PORTA_DATA_BUS
// #define STM_PORTB_DATA_BUS

// Tell the library to use parallel mode (otherwise SPI is assumed)
// #define TFT_PARALLEL_8_BIT
// #defined TFT_PARALLEL_16_BIT // **** 16-bit parallel ONLY for RP2040 processor ****

// Display type -  only define if RPi display
// #define RPI_DISPLAY_TYPE // 20MHz maximum SPI

// Only define one driver, the other ones must be commented out
// #define ILI9341_DRIVER       // Generic driver for common displays
// #define ILI9341_2_DRIVER     // Alternative ILI9341 driver, see https://github.com/Bodmer/TFT_eSPI/issues/1172
// #define ST7735_DRIVER      // Define additional parameters below for this display
// #define ILI9163_DRIVER     // Define additional parameters below for this display
// #define S6D02A1_DRIVER
// #define RPI_ILI9486_DRIVER // 20MHz maximum SPI
// #define HX8357D_DRIVER
// #define ILI9481_DRIVER
// #define ILI9486_DRIVER
// #define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)
#define ST7789_DRIVER // Full configuration option, define additional parameters below for this display
// #define ST7789_2_DRIVER    // Minimal configuration option, define additional parameters below for this display
// #define R61581_DRIVER
// #define RM68140_DRIVER
// #define ST7796_DRIVER
// #define SSD1351_DRIVER
// #define SSD1963_480_DRIVER
// #define SSD1963_800_DRIVER
// #define SSD1963_800ALT_DRIVER
[... 295 lines cut ...]

// Optional reduced SPI frequency for reading TFT
#define SPI_READ_FREQUENCY 20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY 2500000

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
// #define USE_HSPI_PORT

// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!

// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

// #define SUPPORT_TRANSACTIONS

```

tobozo, 2024-06-15:
> I've removed the Light_PWM and Touch_ sections, as I understood, that if not needed I should remove it

I'm not sure how you evaluated the "not needed" part, but you need Light_PWM since it's the backlight and without backlight nothing is visible, also if your display has a touch interface you need it too

I already found one difference between both configurations, there are probably more but I don't want to spoil your opportunity to get more experience at debugging your own code:

TFT_eSPI:

        #define TFT_MISO 37

LGFX:

        cfg.pin_miso = -1; // SPIのMISOピン番号を設定 (-1 = disable)

I'll close this issue as it is not a defect of the library, if still you need help with migrating TFT_eSPI => LGFX, use the discussions tab instead.

closed 2024-06-15 by tobozo