lovyan03/LovyanGFX #582: Support for 4.3" RGB Interface

stackedCG, 2024-06-19:
**Carefully written requests are more likely to be given priority.**
Just want to confirm if we already have a support for this type of device, and if we have can you help me out how to properly setup the display ready for use. I already make the lvgl 8.3 work on the device but I dont have any luck whenever I try running Lovyangfx to the screen. The reason why I want to use this library is because I am much more familiar in sprite and I found it much simpler to work with than the lvgl, thank you!

### Device Name ( ESP32 - 4827S043 | No touch panel)

###[ URL of Device Specifications document](http://pan.jczn1688.com/directlink/1/ESP32%20module/4.3inch_ESP32-4827S043.zip)

### URL of the store where we can purchase (I don't have the URL of the store anymore but it looks very similar to [this](https://www.aliexpress.us/item/3256804601832939.html?gatewayAdapt=glo2usa4itemAdapt))

tobozo, 2024-06-20:
see this [comment](https://github.com/lovyan03/LovyanGFX/issues/374#issuecomment-1494590560)

stackedCG, 2024-06-22:
@tobozo, it is now working on my end, I just tweaked some setting form here and there to accomodate my current project, thank you very much!!!

closed 2024-07-29 by github-actions[bot]