lovyan03/LovyanGFX #583: Incomplete support for SSD1963?

tylercamp, 2024-06-26:
I'm using this display with an ESP32 adapter board (available on the same page) which has an SSD1963Q9: https://www.aliexpress.us/item/3256805880921100.html

I was unable to get this display to work with LovyanGFX (original code below) but the vendor provided some sample code which does work. The sample code includes an extra `CD` pin, which I can't find any reference to in LovyanGFX (maybe a different name? don't see this in SSD1963 spec sheet either), and backlight control is handled through commands to the SSD1963 controller instead of directly modulating a pin for PWM.

[Working sample code.zip](https://github.com/user-attachments/files/15986590/src.zip) - In `LCDWIKI_KBV.cpp` see `Set_1963_PWM` at line 726.

When using the below config with 3v3 manually tied directly to the display's backlight, I only got a black screen despite filling with purple. (I confirmed that the pins are assigned correctly and tried a few different options for `_panel_instance`.)

Is there some other set of classes/functions I should call when configuring for SSD1963?

My `LGFX_Device` definition:

```cpp
#ifdef DISPLAY_SSD1963
#include <LovyanGFX.hpp>

/*
	-DDISPLAY_SSD1963
	-DDISPLAY_SSD1963_D0=12
	-DDISPLAY_SSD1963_D1=13
	-DDISPLAY_SSD1963_D2=26
	-DDISPLAY_SSD1963_D3=25
	-DDISPLAY_SSD1963_D4=17
	-DDISPLAY_SSD1963_D5=16
	-DDISPLAY_SSD1963_D6=27	
	-DDISPLAY_SSD1963_D7=14
	-DDISPLAY_SSD1963_CS=33
	-DDISPLAY_SSD1963_RST=32
	-DDISPLAY_SSD1963_RS=15
	-DDISPLAY_SSD1963_RD=2
	-DDISPLAY_SSD1963_WR=4
*/

// ESP32でLovyanGFXを独自設定で利用する場合の設定例

/// 独自の設定を行うクラスを、LGFX_Deviceから派生して作成します。
class LGFX : public lgfx::LGFX_Device
{
lgfx::Panel_SSD1963     _panel_instance;

// パネルを接続するバスの種類にあったインスタンスを用意します。
lgfx::Bus_Parallel8  _bus_instance;   // 8ビットパラレルバスのインスタンス

// バックライト制御が可能な場合はインスタンスを用意します。(必要なければ削除)
  lgfx::Light_PWM     _light_instance;

// タッチスクリーンの型にあったインスタンスを用意します。(必要なければ削除)
// TODO - Funkeydo display is FT5426
  // lgfx::Touch_FT5x06           _touch_instance; // FT5206, FT5306, FT5406, FT6206, FT6236, FT6336, FT6436
public:

  // コンストラクタを作成し、ここで各種設定を行います。
  // クラス名を変更した場合はコンストラクタも同じ名前を指定してください。
  LGFX(void)
  {
    { // バス制御の設定を行います。
      auto cfg = _bus_instance.config();    // バス設定用の構造体を取得します。

// 8ビットパラレルバスの設定
      // cfg.i2s_port = I2S_NUM_0;     // 使用するI2Sポートを選択 (I2S_NUM_0 or I2S_NUM_1) (ESP32のI2S LCDモードを使用します)
      cfg.freq_write = 200000;    // 送信クロック (最大20MHz, 80MHzを整数で割った値に丸められます)
      cfg.pin_wr = DISPLAY_SSD1963_WR;              // WR を接続しているピン番号
      cfg.pin_rd = DISPLAY_SSD1963_RD;              // RD を接続しているピン番号
      cfg.pin_rs = DISPLAY_SSD1963_RS;              // RS(D/C)を接続しているピン番号
      cfg.pin_d0 = DISPLAY_SSD1963_D0;              // D0を接続しているピン番号
      cfg.pin_d1 = DISPLAY_SSD1963_D1;              // D1を接続しているピン番号
      cfg.pin_d2 = DISPLAY_SSD1963_D2;              // D2を接続しているピン番号
      cfg.pin_d3 = DISPLAY_SSD1963_D3;              // D3を接続しているピン番号
      cfg.pin_d4 = DISPLAY_SSD1963_D4;              // D4を接続しているピン番号
      cfg.pin_d5 = DISPLAY_SSD1963_D5;              // D5を接続しているピン番号
      cfg.pin_d6 = DISPLAY_SSD1963_D6;              // D6を接続しているピン番号
      cfg.pin_d7 = DISPLAY_SSD1963_D7;              // D7を接続しているピン番号
//*/

      _bus_instance.config(cfg);    // 設定値をバスに反映します。
      _panel_instance.setBus(&_bus_instance);      // バスをパネルにセットします。
    }

    { // 表示パネル制御の設定を行います。
      auto cfg = _panel_instance.config();    // 表示パネル設定用の構造体を取得します。

      cfg.pin_cs           =    DISPLAY_SSD1963_CS;  // CSが接続されているピン番号   (-1 = disable)
      cfg.pin_rst          =    DISPLAY_SSD1963_RST;  // RSTが接続されているピン番号  (-1 = disable)
      cfg.pin_busy         =    -1;  // BUSYが接続されているピン番号 (-1 = disable)

      // ※ 以下の設定値はパネル毎に一般的な初期値が設定されていますので、不明な項目はコメントアウトして試してみてください。

      cfg.panel_width      =   480;  // 実際に表示可能な幅
      cfg.panel_height     =   800;  // 実際に表示可能な高さ
      cfg.offset_x         =     0;  // パネルのX方向オフセット量
      cfg.offset_y         =     0;  // パネルのY方向オフセット量
      cfg.offset_rotation  =     0;  // 回転方向の値のオフセット 0~7 (4~7は上下反転)
      cfg.dummy_read_pixel =     8;  // ピクセル読出し前のダミーリードのビット数
      cfg.dummy_read_bits  =     1;  // ピクセル以外のデータ読出し前のダミーリードのビット数
      cfg.readable         =  false;  // データ読出しが可能な場合 trueに設定
      cfg.invert           = false;  // パネルの明暗が反転してしまう場合 trueに設定
      cfg.rgb_order        = false;  // パネルの赤と青が入れ替わってしまう場合 trueに設定
      cfg.dlen_16bit       = true;  // 16bitパラレルやSPIでデータ長を16bit単位で送信するパネルの場合 trueに設定
      cfg.bus_shared       =  false;  // SDカードとバスを共有している場合 trueに設定(drawJpgFile等でバス制御を行います)

      _panel_instance.config(cfg);
    }

    setPanel(&_panel_instance); // 使用するパネルをセットします。
  }
};

LGFX *display;
#endif
```

tobozo, 2024-06-26:
hi,

    cfg.freq_write = 200000;

200 KHz seems rather slow for a parallel display, have you tried in the MHz range ?

tylercamp, 2024-06-26:
I set it to 20MHz to start with, and since I wasn't getting any output, I tried lowering it as one of the troubleshooting steps

The adapter board does not expose a backlight pin to the ESP32, and I didn't see a suitable alternative impl. for controlling backlight with SSD1963, so I left that commented out

The sample code I attached shows a command sent to SSD1963 for controlling backlight

tobozo, 2024-06-26:
sorry I didn't read your full post before replying :blush: 

you can probably take the `Set_1963_PWM()` function from the vendor and implement it inside your LGFX custom class, this is untested code, and the function should be called after `display.init()`: 

```cpp
class LGFX : public lgfx::LGFX_Device
{
  // ... ( your existing config )

    void Set_1963_PWM(uint8_t value)
    {
      uint8_t pwm=value*2.55;
      _bus_instance.writeCommand(0xBE, 8);
      _bus_instance.wait();
      _bus_instance.writeData( 0xBE, 8 );
      _bus_instance.writeData( 0x05, 8 );
      _bus_instance.writeData( pwm, 8 );
      _bus_instance.writeData( 0x01, 8 );
      _bus_instance.writeData( 0xFF, 8 )
      _bus_instance.writeData( 0x00, 8 )
      _bus_instance.writeData( 0x00, 8 );
      _bus_instance.wait();
    }
}
```

if this solves the issue for you then we'll think about a way to include this in the driver without affecting the other SSD1963 variants

tobozo, 2024-06-26:
maybe `_bus_instance.config().cfg.pin_ctrl[3]` can be used for pulling CD high ? e.g. `cfg.pin_ctrl = { DISPLAY_SSD1963_CD, -1, -1 };` ?

> (The pin is just used to digitalWrite(HIGH) in the LCDWIKI_KBV ctor, but it's also referenced with CD_COMMAND/CD_DATA macros from mcu_8bit_magic.h)

apparently `CD_COMMAND` uses that DC pin too in `LCDWIKI_KBV::reset()`, but I don't know `Panel_SSD1963` well enough to take a guess at how LGFX should handle it, maybe this needs a custom implementation of `Bus_Parallel8` too

tylercamp, 2024-06-26:
Oh whoops, looks like "CD" was a typo that made its way throughout the sample codebase:

```
#define CD_PIN   15  // Data Command control pin - must use a pin in the range 0-31
```

D/C i.e. RS pin is 15, which is mapped correctly

So it looks like all of the pins are assigned as expected, I'm just not sure why I'm getting shimmering

I tried comparing the init command bytes which are similar but different enough that I can't make a good comparison. I did notice differences in timing values, and changing `xtal_clock` affects the shimmering, so it's likely a timing issue

dekesone, 2024-07-03:
> sorry I didn't read your full post before replying 😊
> 
> you can probably take the `Set_1963_PWM()` function from the vendor and implement it inside your LGFX custom class, this is untested code, and the function should be called after `display.init()`:
> 
> ```c++
> class LGFX : public lgfx::LGFX_Device
> {
>   // ... ( your existing config )
> 
>     void Set_1963_PWM(uint8_t value)
>     {
>       uint8_t pwm=value*2.55;
>       _bus_instance.writeCommand(0xBE, 8);
>       _bus_instance.wait();
>       _bus_instance.writeData( 0xBE, 8 );
>       _bus_instance.writeData( 0x05, 8 );
>       _bus_instance.writeData( pwm, 8 );
>       _bus_instance.writeData( 0x01, 8 );
>       _bus_instance.writeData( 0xFF, 8 )
>       _bus_instance.writeData( 0x00, 8 )
>       _bus_instance.writeData( 0x00, 8 );
>       _bus_instance.wait();
>     }
> }
> ```
> 
> if this solves the issue for you then we'll think about a way to include this in the driver without affecting the other SSD1963 variants

I have LGFX working great on the same 5" display noted by @tylercamp but have spent the last 6 hours pulling my hair out trying to get the backlight to work properly.  I have the same sample code  (In LCDWIKI_KBV.cpp see Set_1963_PWM at line 726) that properly adjust the backlight PWM, but for the life of me can not get it to work within LGFX,

The example function above provided by @tobozo seems to be close but does not work.  I noticed a few things that I attempted to correct:
      This line is seems to be unnecessary:      _bus_instance.writeData( 0xBE, 8 );
      The writeCommand() should 16 bit length
      There's a few missing ;

But nonetheless still no luck.
I have subsequently dug deeper into the Dynamic Backlight Control and the SSD1963 datasheet.  It seems that after setting the PWM value, you also have to turn on DBC.  Here's the pseudo code from the datasheet:

The hardware pin, PWM is the output signal from SSD1963 to the system backlight driver.  So it should configure PWM module before enable DBC. 
WRITE COMMAND “0xBE” 
WRITE DATA “0x0E” (set PWM frequency) 
WRITE DATA “0xFF” (dummy value if DBC is used) 
WRITE DATA “0x09” (enable PWM controlled by DBC) 
WRITE DATA “0xFF” 
WRITE DATA “0x00” 
WRITE DATA “0x00” 
WRITE COMMAND “0xD4” (Define the threshold value)
 WRITE DATA ..... 
WRITE COMMAND “0xD0” 
WRITE DATA “0x0D”     (Enable DBC with Aggressive mode

Any suggestions on how to proceed?

tylercamp, 2024-07-03:
@dekesone Could you provide your LGFX config?

dekesone, 2024-07-03:
Here's what works well for me, minus the PWM backlight obviously.

[gfx_conf.h.txt](https://github.com/user-attachments/files/16085853/gfx_conf.h.txt)

tylercamp, 2024-07-04:
@dekesone thanks for the reference, I found that I was missing this:

```cpp
      cfg.panel_width = cfg.memory_width = 800;
      cfg.panel_height = cfg.memory_height = 480;
```

I thought I saw init of panel dimensions in the SSD1963 ctor so it seemed unnecessary, but this was the cause of the shimmering I saw.

For the PWM backlight:

> I have subsequently dug deeper into the Dynamic Backlight Control and the SSD1963 datasheet. It seems that after setting the PWM value, you also have to turn on DBC. Here's the pseudo code from the datasheet: ...

The last two lines for enabling DBC are present in the SSD1963 `fixed_cmds`:

```cpp
...
      0xD0, 1, 0x0D, // <-- here (L133)
...
```

... so whatever the underlying backlight issue is, it doesn't seem like it's due to missing DBC init.

tylercamp, 2024-07-04:
@dekesone Got it! The general commands were correct but there were underlying operations that were missing (e.g. toggling CS pin) since we were interacting with `_bus_instance` directly. I updated the commands to use the same functions as `Panel_SSD1963::init`:

```cpp
  void Set_1963_PWM(uint8_t value)
  {
    startWrite();
    writeCommand16(0xBE);
    _bus_instance.flush();

    writeData( 0x05 );
    writeData( value );
    writeData( 0x01 );
    writeData( 0xFF );
    writeData( 0x00 );
    writeData( 0x00 );

    endWrite();
  }
```

tylercamp, 2024-07-04:
To summarize:

The shimmering effect I saw was due to incomplete display init. Assigning `panel_width/height` and `memory_width/height` fixed this problem.

We've figured out how to manage backlight control with LGFX and some customizations, but there should be an update to LGFX to properly support this case. #586

closed 2024-07-04 by tylercamp

dekesone, 2024-07-04:
Thanks @tylercamp, great work !

tobozo, 2024-07-04:
@tylercamp thanks for breaking out the different suggested solutions :+1: , I have forwarded the new issue link to lovyan03 as I honestly don't know which one is better since this is beyond my skills

however he's been very busy lately so don't expect a lightning fast answer :wink:

tylercamp, 2024-07-25:
@dekesone An [update was pushed](https://github.com/lovyan03/LovyanGFX/issues/586#issuecomment-2249590911) to `develop` branch by @lovyan03, are you able to test the update? I'm away from home and don't have a device to test with right now

tylercamp, 2024-08-03:
@dekesone proper fix merged to 'develop' branch, call 'myLgfx->setBrightness' instead of 'myLgfx->light()->setBrightness' (which is apparently the preferred method in general but the examples get the light directly instead)