lovyan03/LovyanGFX #585: Define Panels with multiple Touch input device

jgauchia, 2024-07-04:
Hi!
I've a question. (Maybe I don't know how to do it)

It's possible to define panels (TFT) with different touch input devices?
This is. I need to define TFT but I need to define the touch controller separately.

Thank you

tobozo, 2024-07-04:
hi,

do you mean a second touch controller (if so, please specify the geometry) or just a single touch controller running from a different driver ?

jgauchia, 2024-07-04:
> hi,
> 
> do you mean a second touch controller (if so, please specify the geometry) or just a single touch controller running from a different driver ?

sorry, my english isn't good.

I mean one tft driver definition only and for the touch controller another one. that is... an example

Driver ILI9488 only definition of panel (SPI , Parallel ... etc..)
Touch: (for example)
  * XPT2046 another include for definition
  * FT6236 the same

The goal is define the TFT panel with one include and the input part (touch, etc,..) in other because there are many types of them

tobozo, 2024-07-04:
if you wish to support multiple touch devices in your project, but only need to enable one of them (no concurrency), then you can try this:

```cpp
#define USE_XPT2046 true
#define USE_FT5x06 false

class LGFX : public lgfx::LGFX_Device
{
  lgfx::Panel_ILI9499     _panel_instance6;

  lgfx::Touch_FT5x06           _touch_instance_FT5x06; 
  lgfx::Touch_XPT2046           _touch_instance_XPT2046; 

  // (... ) bus and panel config block

  // (...) FT6236 config block without  _panel_instance.setTouch()

  // (...) XPT2046 config block without  _panel_instance.setTouch()

  if( USE_XPT2046 ) 
    _panel_instance.setTouch(&_touch_instance_XPT2046 );  
  else if( USE_FT5x06 ) 
    _panel_instance.setTouch(&_touch_instance_FT5x06);

  setPanel(&_panel_instance);
};
```

jgauchia, 2024-07-04:
> if you wish to support multiple touch devices in your project, but only need to enable one of them (no concurrency), then you can try this:
> 
> ```cpp
> #define USE_XPT2046 true
> #define USE_FT5x06 false
> 
> class LGFX : public lgfx::LGFX_Device
> {
>   lgfx::Bus_SPI       _bus_instance;
>   lgfx::Panel_ILI9499 _panel_instance;
> 
>   lgfx::Touch_FT5x06  _touch_instance_FT5x06; 
>   lgfx::Touch_XPT2046 _touch_instance_XPT2046; 
> 
>   public:
>     LGFX(void)
>     {
> 
>       // (... ) bus and panel config block
> 
>       // (...) FT6236 config block without  _panel_instance.setTouch()
> 
>       // (...) XPT2046 config block without  _panel_instance.setTouch()
> 
>       if( USE_XPT2046 ) 
>         _panel_instance.setTouch(&_touch_instance_XPT2046 );  
>       else if( USE_FT5x06 ) 
>         _panel_instance.setTouch(&_touch_instance_FT5x06);
> 
>       setPanel(&_panel_instance);
>     }
> };
> ```
> 

Thank you!

closed 2024-08-10 by github-actions[bot]