lovyan03/LovyanGFX #587: Add smooth arc function

oldmud0, 2024-07-05:
**Carefully written requests are more likely to be given priority.**
**丁寧に記述された要望は優先して対応される可能性が高くなります。**

**Is your feature request related to a problem? Please describe.**
Compared to TFT_eSPI, LGFX lacks smooth arc functions. (In TFT_eSPI, this is the `TFT_eSPI::drawSmoothArc` function). This would also be a great feature to have, considering that LGFX already has smooth lines, smooth circles, and smooth round rects.

**Describe the solution you'd like**
Add the `smooth` parameter to `fill_arc_helper`, `fillEllipseArc`, and `drawEllipseArc` to support new `drawSmoothArc` and `fillSmoothArc` functions, and implement the antialiasing code.

**Describe alternatives you've considered**
I've looked at the TFT_eSPI code and considered copying it, but the arc functions don't line up exactly the same from what I can see.

**Additional context**
Screenshot showing a normal call to `LGFX_Base::fillArc` from the SDL2 runtime.
[image]

tobozo, 2024-07-08:
hi, thanks for your suggestion :+1: 

apparently `drawSmoothArc()` function exists in [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L3968-L4025), since LGFX already implements the necessary functions `drawSpot()` and `drawWedgeLine()`, it should not be a problem to import it and make it work with all lgfx color types.

also the [`DrawLineThickness.ino` example](https://github.com/lovyan03/LovyanGFX/blob/master/examples/Standard/DrawLineThickness/DrawLineThickness.ino) will need to be extended with that new function.

I'll update this thread with the progress.

tobozo, 2024-07-08:
I was a bit shocked by the [brute force used in `drawArc()` to find the edges](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L4162) :thinking:  

also I forgot `fill_arc_helper()` takes two radiuses and has a totally different approach from bodmer's `drawArc()` so it's not as simple as using the existing functions

@oldmud0 anything you can suggest to help implement smoothness in `fill_arc_helper()` will be welcome

tonbor, 2024-07-14:
Any progress?

tobozo, 2024-07-15:
yes: `fill_arc_helper()` can't have smoothness added without performance loss as the function is already optimized for fastest rendering, and adding any conditional block for antialiased edges will compromise that

LGFX's arc helper primitives are based on ellipse (e.g. a circle is an ellipse with two equal radiuses), so there are two paths to get this done:

1) Use `fill_arc_helper()` as a base to create `fill_smooth_arc_helper()`: the hard part is to implement alpha for the edges in a two-radiuses context while keeping the ellipse based logic
2) Integrate Bodmer's `drawSmoothArc()` and its primitives: very slow, and someone will eventually ask for an ellipse version of the function

I don't have the time and skills to do either points, but I can provide support and help for anything related to a proper integration in LovyanGFX.