lovyan03/LovyanGFX #591: drawArc lovyanGFX not the same as drawArc TFT_Espi

tonbor, 2024-07-18:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  [ESP-WROOM-32|ESP32-S3-BOX|M5Stack|WioTerminal|RPi2040|other]
- Panel Driver IC:    [ILI9341|SSD1963|ST7796|other]
- Bus type:           [SPI|I2C|Parallel8|Parallel16|other]
- LovyanGFX version:  [v0.4.17|latest develop|other]
- FrameWork version:   [ArduinoESP32 v2.0.3|ESP-IDF v4.4|other]
- Build Environment:   [ArduinoIDE|PlatformIO|Eclipse|other]
- Operating System:   [Windows|Linux|macOS]

## Problem Description ( 問題の内容 )
This is what the headers say: TFT_Espi:
 void     drawArc(int32_t x, int32_t y, int32_t r, int32_t ir, uint32_t startAngle, uint32_t endAngle, uint32_t fg_color, uint32_t bg_color, bool smoothArc = true);
Yours:
 void 	drawArc  ( int32_t x, int32_t y, int32_t r0, int32_t r1, float angle0, float angle1, const T& color) { setColor(color);
### Expected Behavior ( 期待される動作 )
drawing the same arc with equal sketch only omit background color
### Actual Behavior ( 実際の動作 )
arc starts at different place
### Steps to reproduce ( 再現のための前提条件 )

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue ( 再現させるためのコード )

**Please submit complete source code that can reproduce your problem.**
**あなたの問題を再現できる完全なソースコードを提示してください。**

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

tonbor, 2024-07-18:
[image]
Arcs are reversed with the same source code

tonbor, 2024-07-19:
[image]
Suppose this is the answer, add 90 to both the angles?

lovyan03, 2024-07-25:
私がdrawArc関数を作成した時点では、TFT_eSPIにはdrawArc関数がありませんでした。

closed 2024-07-25 by lovyan03

tonbor, 2024-07-25:
Ok but my solution add 90 works so for other users, this will help