lovyan03/LovyanGFX #592: Issues rewriting text with color & background = TFT_BLACK

aportela, 2024-07-20:
## Environment ( 実行環境 )

- MCU or Board name:  [ESP-WROOM-32|]
- Panel Driver IC:    [ST7796]
- Bus type:           [SPI]
- LovyanGFX version:  [v1.1.16]
- FrameWork version:   [ArduinoESP32 v2.0.13]
- Build Environment:   [ArduinoIDE]
- Operating System:   [Windows]

## Problem Description ( 問題の内容 )

When rewriting a string above another at the same coordinates and size with black color, the original string is not completely deleted. If I use a color other than black, the original string is deleted correctly and the new one is written over it.

Personally I use it in many cases to avoid having to use an intermediate buffer (sprite) or draw a filled box on top of the old text before drawing the new text.

### Expected Behavior ( 期待される動作 )

I know that this is not the AdafruitGFX library but since it shares many things and since if I use another color other than black it works correctly, here is the idea of ​​the procedure I use:

[Overwriting Text with the Built-In Font](https://learn.adafruit.com/adafruit-gfx-graphics-library/minimizing-redraw-flicker)

### Actual Behavior ( 実際の動作 )

### Steps to reproduce ( 再現のための前提条件 )

1. Use TFT_BLACK, TFT_BLACK on last setTextColor for FAIL (In theory, the specified number should be "drawn in black", meaning nothing would be seen since it would simulate a deletion of the text)
2. Use TFT_RED, TFT_BLACK on last setTextColor for WORK (The previous text is completely erased and the new text is displayed in red correctly)

### Code to reproduce this issue ( 再現させるためのコード )
```cpp
void setup()
{
    // ... LGFX init (custom pins) (my pointer LGFX is screen)
    screen->init();
    screen->fillScreen(TFT_BLACK);
    screen->setCursor(100, 100);
    screen->setTextSize(2);
    screen->setTextColor(TFT_WHITE, TFT_BLACK);
    screen->print("0123456789");
    screen->setCursor(100, 100);
    screen->setTextSize(2);
    screen->setTextColor(TFT_RED, TFT_BLACK); // THIS WORKS, CHANGE FIRST TFT_RED to TFT_BLACK FOR BUG
    screen->print("9876543210");
}

```

This is the picture reflecting the BUG:

[image]

This is the picture working FINE (using color != TFT_BLACK):
[image]

Maybe it's not a bug, but I assumed I could use the TFT_BLACK color to simulate text deletion in certain cases

lovyan03, 2024-07-25:
前景色と背景色を同じにした場合は背景を塗りつぶしません。
これは仕様通りの動作です。

closed 2024-07-25 by lovyan03

aportela, 2024-07-28:
Ah, okay, that's clear then, thanks and sorry for the confusion.