lovyan03/LovyanGFX #600: ILI9341 shows me mirrored?

YanDevDe, 2024-08-04:
## Environment ( 実行環境 )

- MCU or Board name:  board-esp32-devkit-c-v4
- Panel Driver IC:   ILI9341
- Bus type:  SPI
- LovyanGFX version:  1.1.16
- Build Environment:   PlatformIO
- Operating System:   macOS Apple Silicon

I've been converting my code from TFT_eSPI to LovyanGFX and I was able to draw contents on the display. But for some odd reason - the screen are inverted.

After carefully reading here I learned that you can actually flip the screen by using "setRotation", so I tried the value 0-7 and I got following weird result:

0 - 0deg mirrored (default)
1 - 90deg mirrored
2 - 0deg mirrored
3 - 90deg mirrored
4 - 180deg normal
5 - 270deg normal
6 - 180deg normal
7 - 270deg normal

As you can see, 0/2, 1/3, 4/6, 5/7 is the same, which is a odd behavior. Is that maybe due misconfigurations in LovyanGFX_Setup.h?

LovyanGFX_Setup.h:

```
#ifndef LGFX_SETUP_H
#define LGFX_SETUP_H

#include <LovyanGFX.hpp>

// Define your custom setup class inheriting from LGFX_Config
class LGFX : public lgfx::LGFX_Device
{
    lgfx::Panel_ILI9341 _panel_instance;
    lgfx::Bus_SPI _bus_instance;
    lgfx::Light_PWM _light_instance;

public:
    LGFX(void)
    {
        {
            auto cfg = _bus_instance.config(); // Gets the configuration structure for bus control.

            // SPI bus settings
            cfg.spi_host = VSPI_HOST;  // Select SPI host (HSPI_HOST or VSPI_HOST)
            cfg.spi_mode = 0;          // Set SPI communication mode (0 or 3)
            cfg.freq_write = 40000000; // SPI clock frequency for write transactions
            cfg.freq_read = 16000000;  // SPI clock frequency for read transactions
            cfg.spi_3wire = false;     // true = Use MOSI and MISO pins for 3-wire SPI communication
            cfg.use_lock = true;       // Set to true if SPI bus should be locked during transactions
            cfg.dma_channel = 1;       // DMA channel for SPI transactions (1 or 2 for ESP32)
            cfg.pin_sclk = 18;         // SPI SCK pin
            cfg.pin_mosi = 23;         // SPI MOSI pin
            cfg.pin_miso = 19;         // SPI MISO pin (optional)
            cfg.pin_dc = 2;            // Data/Command pin

            _bus_instance.config(cfg);              // Apply the bus configuration
            _panel_instance.setBus(&_bus_instance); // Set the bus to the panel instance
        }

        {
            auto cfg = _panel_instance.config(); // Gets the configuration structure for panel control.

            cfg.pin_cs = 15;   // Chip select pin
            cfg.pin_rst = 4;   // Reset pin
            cfg.pin_busy = -1; // Busy pin

            cfg.memory_width = 240;
            cfg.memory_height = 320;
            cfg.panel_width = 240;
            cfg.panel_height = 320;
            cfg.offset_x = 0;
            cfg.offset_y = 0;
            cfg.offset_rotation = 0;
            cfg.dummy_read_pixel = 8;
            cfg.dummy_read_bits = 1;
            cfg.readable = true;
            cfg.invert = false;    // Set this to true if your display is inverted
            cfg.rgb_order = false; // Set this to true if RGB color order is incorrect
            cfg.dlen_16bit = false;
            cfg.bus_shared = false;

            _panel_instance.config(cfg);
        }

        {
            auto cfg = _light_instance.config(); // Gets the configuration structure for backlight control.

            cfg.pin_bl = 33;     // LED backlight pin
            cfg.invert = false;  // true if you need to invert the backlight brightness
            cfg.freq = 44100;    // PWM frequency
            cfg.pwm_channel = 7; // PWM channel

            _light_instance.config(cfg);
            _panel_instance.setLight(&_light_instance); // Set backlight configuration to the panel instance
        }

        setPanel(&_panel_instance); // Apply the panel configuration
    }
};

#endif // LGFX_SETUP_H
```

I prefer to have a custom display setup script as it allows me to modify the value directly, hence "LovyanGFX_Setup" is there.

I'm using Wokwi Simulator for now (But I'm going to change display + different devkit of ESP32 soon), so the result looks like this:

[image]

As you can see, all the text and drawings are in mirrored mode. The position X starts on the right side instead of the left one.

The pins are connected:

CS 15
RST 4
DC 2
MOSI 23
SCLK 18
MISO 19
LED 33

Does anybody knows why I'm getting this weird result?

Thank you so much!

YanDevDe, 2024-08-04:
Here is the code which runs under Wokwi:

https://wokwi.com/projects/405326271216619521

tobozo, 2024-08-19:
hi,

please try to modify the value of `cfg.offset_rotation`, by assigning a number in the 0-7 range.

YanDevDe, 2024-08-19:
Hello @tobozo 

I forgot to mention, but yes, I have already played around and the result were the exactly the same as setRotation. So unfortunately it didn't fixed it.

tobozo, 2024-08-19:
btw the [wokwi project link](https://wokwi.com/projects/405326271216619521) only compiles if the source is unmodified.

as soon as I change one line of code I get this: 

[image]

sorry if I can't help you more

YanDevDe, 2024-09-21:
It's not really fixed, but I decided to go for another display.

tobozo, 2024-09-21:
I notified wokwi about the editing issue, but it doesn't seem they fixed it, clicking on your link and reproducing the error is fine, but if I edit the sketch and run it again it fails with that cryptic network error

closed 2024-10-28 by github-actions[bot]