lovyan03/LovyanGFX #608: spi_3wire doesn't seem to be working?

lbuque, 2024-08-24:
## Environment ( 実行環境 )

- MCU or Board name:  [ESP32-C3]
- Panel Driver IC:    [ST7789P3]
- Bus type:           [SPI]
- LovyanGFX version:  [v1.1.16]
- FrameWork version:   [ArduinoESP32 v2.0.14]
- Build Environment:   [PlatformIO]
- Operating System:   [macOS]

## Problem Description ( 問題の内容 )

I have a ST7789P3 TFT, and the connection method it uses is SPI3WIRE.

I have tried to drive it successfully using spi. I want to use lgfx to drive it because lgfx has rich functions to meet my needs.

### Expected Behavior ( 期待される動作 )

The screen can display red.

[image]

### Actual Behavior ( 実際の動作 )

There is no display on the screen, it seems that TFT initialization failed.

[image]

### Steps to reproduce ( 再現のための前提条件 )

1. Set the USE_LGFX macro to 1.

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue ( 再現させるためのコード )

```cpp
// select one of the following options
#define USE_LGFX 1
#define USE_SPI_DRIVER 0
```

You can see my code at the link below:
https://github.com/lbuque/HardWareTest

tobozo, 2024-08-24:
hi,

thanks for your feedback :+1: 

`cfg.freq_write = 10 * 1000 * 1000;` (=10MHz) seems rather low considering the parent model ST7789 can handle 40-80MHz

the HardwareTest example code uses `Panel_ILI9342` as the panel object but your display is ST7789P3, unless both display use the same registers, the init sequence from ILI9342 is just ignored and has no effect on the ST7789P3.

I took the 'working' [init sequence](https://github.com/lbuque/HardWareTest/blob/master/HardWareTest.ino#L144-L241) from the HardwareTest to create a `Panel_ST7789P3` class, please let me know if this works for you:

`src/lgfx/v1/panel/Panel_ST7789P3.hpp`

```cpp
/*----------------------------------------------------------------------------/
 * Lovyan GFX - Graphics library for embedded devices.
 *
 O *riginal Source:
 https://github.com/lovyan03/LovyanGFX/

 Licence:
 [FreeBSD](https://github.com/lovyan03/LovyanGFX/blob/master/license.txt)

 Author:
 [lovyan03](https://twitter.com/lovyan03)

 Contributors:
 [ciniml](https://github.com/ciniml)
 [mongonta0716](https://github.com/mongonta0716)
 [tobozo](https://github.com/tobozo)
 /----------------------------------------------------------------------------*/
#pragma once

#include "Panel_LCD.hpp"

namespace lgfx
{
    inline namespace v1
    {
        //----------------------------------------------------------------------------

        struct Panel_ST7789P3  : public Panel_LCD
        {
            Panel_ST7789P3(void)
            {
                _cfg.panel_height = _cfg.memory_height = 320;
                _cfg.dummy_read_pixel = 16;
            }

        protected:

            const uint8_t* getInitCommands(uint8_t listno) const override {

                static constexpr uint8_t list0[] = {
                    0x11, 0+CMD_INIT_DELAY, 120
                    0x36, 1, 0x00,
                    0x3A, 1, 0x05,
                    0xB2, 5, 0x0C, 0x0C, 0x00, 0x33, 0x33,
                    0xb7, 1, 0x56,
                    0xbb, 1, 0x1d,
                    0xc0, 1, 0x2c,
                    0xc2, 1, 0x01,
                    0xc3, 1, 0x0f,
                    0xc6, 1, 0x0f,
                    0xd0, 1, 0xa7,
                    0xd0, 2, 0xa4, 0xa1,
                    0xd6, 1, 0xa1,
                    0xe0, 14, 0xF0, 0x02, 0x07, 0x05, 0x06, 0x14, 0x2F, 0x54, 0x46, 0x38, 0x13, 0x11, 0x2E, 0x35,
                    0xe1, 14, 0xF0, 0x08, 0x0C, 0x0C, 0x09, 0x05, 0x2F, 0x43, 0x46, 0x36, 0x10, 0x12, 0x2C, 0x32,
                    0x21, 0
                    0x29, 0,
                    0x2c, 0+CMD_INIT_DELAY, 100,
                    0x2a, 4, 0x00, 0x00, 0x01, 0x3f,
                    0x2b, 4, 0x00, 0x00, 0x00, 0xef,
                    0x2c, 0,
                    0xFF,0xFF, // end
                };
                switch (listno) {
                    case 0: return list0;
                    default: return nullptr;
                }
            }
        };

        //----------------------------------------------------------------------------
    }
}

```

lbuque, 2024-08-24:
@tobozo Thanks for your reply and helping me find the bug.

I followed your suggestion and tried using Panel_ST7789 and Panel_ST7789P3, but found that tft still did not work properly.

I am about to connect a logic analyzer to the board to check the spi timings.

Finally, thank you again for your quick response.

lbuque, 2024-08-25:
I lowered the spi frequency to 4MHZ to prevent the logic from interfering when collecting data, however, I could only collect 0x00.

[image]

What am I doing wrong?

tobozo, 2024-08-25:
pin assignation seems wrong (same pin for miso and mosi?):

```cpp
#define MOSI_PIN 3
#define MISO_PIN 3 // should be -1 according to SPI_3Wire_Interface_Init() !!
```

anyway, given the contents of platformio.ini it's already a miracle the esp even starts:

- it selects esp32-c3, but should select esp32-s3
- it uses an old version of espressif core (2.1.4)
- it sets the cpu speed to 160MHz

please try with this platformio env instead, it'll use the latest espressif core:

```ini
[env:stamp-s3]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip
board = m5stack-stamps3
framework = arduino
lib_deps =
    mathertel/OneButton@^2.0.3
    bblanchon/ArduinoJson @ ^6.21.3
    lovyan03/LovyanGFX @ 1.1.16
upload_speed = 1500000
upload_protocol = esptool
monitor_speed = 115200
build_flags =
    -D CORE_DEBUG_LEVEL=3
board_build.filesystem = littlefs
```

and here's a fixed version of your sketch:

```cpp

#include <LovyanGFX.hpp>

namespace lgfx
{
  inline namespace v1
  {
    //----------------------------------------------------------------------------

    struct Panel_ST7789P3  : public Panel_LCD
    {
      Panel_ST7789P3(void)
      {
        _cfg.panel_height = _cfg.memory_height = 320;
        _cfg.dummy_read_pixel = 16;
      }

    protected:

      const uint8_t* getInitCommands(uint8_t listno) const override {

        static constexpr uint8_t list0[] = {
          0x11, 0+CMD_INIT_DELAY, 120,
          0x36, 1, 0x00,
          0x3A, 1, 0x05,
          0xB2, 5, 0x0C, 0x0C, 0x00, 0x33, 0x33,
          0xb7, 1, 0x56,
          0xbb, 1, 0x1d,
          0xc0, 1, 0x2c,
          0xc2, 1, 0x01,
          0xc3, 1, 0x0f,
          0xc6, 1, 0x0f,
          0xd0, 1, 0xa7,
          0xd0, 2, 0xa4, 0xa1,
          0xd6, 1, 0xa1,
          0xe0, 14, 0xF0, 0x02, 0x07, 0x05, 0x06, 0x14, 0x2F, 0x54, 0x46, 0x38, 0x13, 0x11, 0x2E, 0x35,
          0xe1, 14, 0xF0, 0x08, 0x0C, 0x0C, 0x09, 0x05, 0x2F, 0x43, 0x46, 0x36, 0x10, 0x12, 0x2C, 0x32,
          0x21, 0,
          0x29, 0,
          0x2c, 0+CMD_INIT_DELAY, 100,
          0x2a, 4, 0x00, 0x00, 0x01, 0x3f,
          0x2b, 4, 0x00, 0x00, 0x00, 0xef,
          0x2c, 0,
          0xFF,0xFF, // end
        };
        switch (listno) {
          case 0: return list0;
          default: return nullptr;
        }
      }
    };

    //----------------------------------------------------------------------------
  }
}

class C3GFX: public lgfx::LGFX_Device {
  lgfx::Panel_ST7789P3 _panel_instance;
  lgfx::Bus_SPI        _spi_bus_instance;
  lgfx::Light_PWM      _light_instance;

public:

  C3GFX() {

    {
      auto cfg = _light_instance.config();
      cfg.pin_bl = GPIO_NUM_5;
      cfg.invert = false;
      cfg.freq = 1000;
      cfg.pwm_channel = 5;
      _light_instance.config(cfg);
    }

    {
      auto cfg = _spi_bus_instance.config();
      //cfg.spi_host = LCD_HOST;
      cfg.spi_mode = 0;
      cfg.freq_write = 40000000;
      cfg.freq_read = 16000000;
      cfg.spi_3wire = true;
      cfg.pin_sclk = GPIO_NUM_1;
      cfg.pin_mosi = GPIO_NUM_3;
      cfg.pin_miso = -1;
      cfg.pin_dc = -1;
      _spi_bus_instance.config(cfg);
    }

    {
      auto cfg = _panel_instance.config();
      cfg.invert       = false;
      cfg.pin_cs       = GPIO_NUM_2;
      cfg.pin_rst      = GPIO_NUM_4;
      cfg.pin_busy     = -1;
      cfg.panel_width  = 320;
      cfg.panel_height = 240;
      cfg.offset_x     = 0;
      cfg.offset_y     = 0;
      cfg.rgb_order = false;
      _panel_instance.setBus(&_spi_bus_instance);
      _panel_instance.config(cfg);
    }

    _panel_instance.setLight(&_light_instance);
    setPanel(&_panel_instance);
  };

  // override the init()
  inline bool init(void)
  {
    // HW_Reset();
    lgfx::pinMode(GPIO_NUM_4, lgfx::pin_mode_t::output);
    lgfx::gpio_hi(GPIO_NUM_4);
    lgfx::delay(100);
    lgfx::gpio_lo(GPIO_NUM_4);
    lgfx::delay(120);
    lgfx::gpio_hi(GPIO_NUM_4);
    delay(120); // ms

    /* Lgfx */
    return lgfx::LGFX_Device::init();
  }

};

C3GFX lcd;

void setup()
{
  Serial.begin(115200);
  lcd.init();
  lcd.println("Hello World");
  delay(1000);
  lcd.clear(TFT_RED);
}

void loop()
{
}

```

lbuque, 2024-08-25:
@tobozo The chip I use is esp32-c3. I don't have stamps3 now.

lovyan03, 2024-08-25:
hello, @lbuque 
I think you're misunderstanding something.
3Wire mode is a mode that transmits and receives via the MOSI pin, not a mode that omits the D/C pin.

lbuque, 2024-08-25:
@lovyan03 Thank you for your answer!

Will lgfx support SPI3WIRE 9BIT?

[image]

If you don't consider it for the time being, I think this issue can be closed.

lbuque, 2024-08-27:
Thanks for your reply! I think SPI3WIRE 9BIT is unreasonable, this issue will be closed. I will use other TFT.

closed 2024-08-27 by lbuque