lovyan03/LovyanGFX #609: QSPI branch - E (1006) spi_master: spi_master_init_driver(236): invalid core id

Westcott1, 2024-08-29:
**Many thanks to Tobozo for setting up this new panel for me.**
Running the code below gives this error in `tft.init();` -
` E (1006) spi_master: spi_master_init_driver(236): invalid core id`
Using instead the SPI bus type (in this branch) runs without this error - but of course no graphics!
Assigning a negative number to `cfg.pin_io2` in the QSPI _bus_instance also runs without this error - but gives others.

## Environment ( 実行環境 )

- Board: ESP32-S3
- Panel Driver IC:  Panel_NV3041A
- Bus type:   QSPI
- LovyanGFX 1.1.16
- FrameWork version:   esp32 v3.1.0
- Build Environment:   ArduinoIDE
- Operating System:   Windows

## Problem Description ( 問題の内容 )
Running the code below gives the error -
` E (1006) spi_master: spi_master_init_driver(236): invalid core id`

### Expected Behaviour ( 期待される動作 )
Display to display something.

### Actual Behaviour ( 実際の動作 )
Running the code below gives the error -
` E (1006) spi_master: spi_master_init_driver(236): invalid core id`

### Steps to reproduce ( 再現のための前提条件 )

1. Run the code below

### Code to reproduce this issue ( 再現させるためのコード )

**Please submit complete source code that can reproduce your problem.**
**あなたの問題を再現できる完全なソースコードを提示してください。**

```
#include <LovyanGFX.hpp>
#include <lgfx_user/LGFX_JC4827W543_4.3inch_ESP32S3.hpp>

LGFX tft;

void setup()
{
  tft.init();
}

void loop()
{
  tft.clear();
  tft.println("Hello World!");
  delay(1000);
}

```

tobozo, 2024-08-30:
> ` E (1006) spi_master: spi_master_init_driver(236): invalid core id`

the [error message](https://github.com/espressif/esp-idf/blob/6e5414b6c4f265a0adfb56a15fbfbe6beb1f8373/components/esp_driver_spi/src/gpspi/spi_master.c#L297) is directly related to `isr_cpu_id` so I guess you can try to play with `cfg.isr_cpu_id` value in bus config:

```cpp
    auto cfg = _bus_instance.config();
    cfg.isr_cpu_id = 1; // or 0
```

Westcott1, 2024-09-02:
Hi Tobozo,

`Compilation error: 'struct lgfx::v1::Bus_QSPI::config_t' has no member named 'isr_cpu_id'`
So I added it to Bus_QSPI.hpp, and tried values of both 0 and 1.
No luck, it still hangs in init()
I also get this -
`pragma message "Quad SPI.begin() postponed to core 3.1.0, see ...`
I'm using core 3.1.0-RC1

tobozo, 2024-09-06:
Hi Hamish,

the message ""Quad SPI.begin() postponed to core 3.1.0" is related to [this issue](https://github.com/espressif/arduino-esp32/issues/7063), which will stay open until the SPI library implements QSPI. Unfortunately 3.1.0-RC1 does not implement it yet.

is this [product page](https://www.aliexpress.com/i/1005006729657546.html) featuring the same model?

aliexpress claims it can be delivered in two weeks so I can eventually order a unit

Westcott1, 2024-12-29:
Hi Tobozo,
I notice that Arduino ESP32 is now officially 3.1.0

tobozo, 2024-12-30:
Hi @Westcott1 

indeed 3.1.0 is out but [only 59% of the milestones](https://github.com/espressif/arduino-esp32/milestone/15) were completed and Quad SPI Mode is still awaiting triage.

happy end of year :partying_face:

tobozo, 2025-04-26:
hi @Westcott1 

[QSPI branch](https://github.com/lovyan03/LovyanGFX/tree/QSPI) has received a few updates, one of which is a fix to spi device configuration improperly assigning core 0 as the isr_cpu_id after an overzealous memset on `spi_bus_config_t`.

Some QSPI displays have been tested successfully, I hope yours will work too :)

Westcott1, 2025-04-27:
Hi,
I can display things now!
However, the NV3041A panel setup has an error - the colours are inverted.
Changing these has no effect -
```
cfg.invert     = true;
cfg.rgb_order  = true;
```

tobozo, 2025-04-27:
can you try to edit `Panel_NV3041A.hpp` and set `CMD_MADCTL_RGB` to 0x01 ?

```cpp
static constexpr uint8_t CMD_MADCTL_RGB   = 0x1;
```

if that works then it's just some logic missing in update_madctl() and I can provide a fix so that setting `cfg.rgb_order` gives the expected result

Westcott1, 2025-04-28:
Hi,
I've added that line, wasn't sure where to put it, so it's just after the #includes.
What changes (if any) should I see?

tobozo, 2025-04-28:
sorry I wasn't clear enough, nothing should be added, there is a line to modify in `src/lgfx/v1/panel/Panel_NV3041A.hpp`, where you should **temporarily** replace `0x0` by `0x1`:

https://github.com/lovyan03/LovyanGFX/blob/4ba414758dffee4cf517238296288414757a4b63/src/lgfx/v1/panel/Panel_NV3041A.hpp#L56

With NV3041A display, rgb order is set along with rotation information. 
Current implementaiton of Panel_NV3041A is using the hardcoded value of [`CMD_MADCTL_RGB`](https://github.com/lovyan03/LovyanGFX/blob/4ba414758dffee4cf517238296288414757a4b63/src/lgfx/v1/panel/Panel_NV3041A.cpp#L92-L105), but the technical sheet indicates that the value (bit) for rgb order can be either 1 or 0, so it's worth testing what happens when that value is set to 1.

By doing so you should observe at least a rgb -> bgr shift in colors.

Please let me know if that works.

Westcott1, 2025-04-28:
Sorry, it really was my fault!
I didn't read your filename properly, and only saw xxxNV3041Axxx.hpp, and instead opened LGFX_JC4827W543_4.3inch_ESP32S3.hpp instead.
Now having edited the correct file. sorry (again) it made no difference.

tobozo, 2025-04-28:
does `tft.setRotation()` work?

I just noticed [`update_madctl()`](https://github.com/lovyan03/LovyanGFX/blob/4ba414758dffee4cf517238296288414757a4b63/src/lgfx/v1/panel/Panel_NV3041A.hpp#L193C18-L193C31) is not marked as override in `src/lgfx/v1/panel/Panel_NV3041A.hpp`, it should be:

```cpp
            void update_madctl(void) override;
```

Westcott1, 2025-04-28:
Yes, setRotation does work.
Adding override made no difference.

mverch67, 2025-08-21:
Also, there a call to `setInvert(_cfg.invert);` missing in `Panel_NV3041A::init()`