lovyan03/LovyanGFX #613: Small Fix for latest RP2040 Core Version

savejeff, 2024-09-06:
おはよう！

I have noticed, that the latest version of the RP2040 (Version 4.X) has the latest pico-sdk. 
it changes two struct names slightly: 

gpio_function -> gpio_function_t 
iobank0_hw_t -> io_bank0_hw_t 

when these are replaced in `src\lgfx\v1\platforms\rp2040\common.cpp` it complies fine again
it, unfortunately, breaks code for the older Version 3.
maybe some can be done with checking defines and doing 
```

typedef gpio_function gpio_function_t;
typedef iobank0_hw_t io_bank0_hw_t;

```
for fallback.

tobozo, 2024-09-06:
hi, thanks for your feedback :+1: 

a slightly different fix was applied recently on the develop branch, I guess the effect is the same

https://github.com/lovyan03/LovyanGFX/blob/develop/src/lgfx/v1/platforms/rp2040/common.cpp#L39-L43

savejeff, 2024-09-06:
cool 👍
I'll use my fork until the changes are merged into the main branch