lovyan03/LovyanGFX #627: Lovyan will not compile with Arduino 2.3.3 all examples

tonbor, 2024-10-09:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  ESP32-S3
- Panel Driver IC:    [ILI9341
- Bus type:           [SPI|
- LovyanGFX version: latest develop
- FrameWork version:  ESP-IDF v4.4|other]
- Build Environment:   ArduinoIDE
- Operating System:   Windows

## Problem Description ( 問題の内容 )

### Expected Behavior ( 期待される動作 )

### Actual Behavior ( 実際の動作 )

### Steps to reproduce ( 再現のための前提条件 )

1. step1 load example
2. compile
3. error
E:\Arduino files\read_pixel\read_pixel.ino:6:8: error: 'LGFX' does not name a type
    6 | static LGFX lcd;
      |        ^~~~
E:\Arduino files\read_pixel\read_pixel.ino: In function 'void tileRect(int, int, int, int, int, uint16_t)':
E:\Arduino files\read_pixel\read_pixel.ino:23:28: error: 'lcd' was not declared in this scope
   23 |       if ( (px+py) & 1 ) { lcd.drawPixel(px, py, color); }
      |                            ^~~
E:\Arduino files\read_pixel\read_pixel.ino: In function 'void drawTile(int, int, uint16_t, uint16_t, uint16_t)':
E:\Arduino files\read_pixel\read_pixel.ino:30:20: error: 'lcd' was not declared in this scope
   30 |   uint16_t color = lcd.color565(r,g,b);
      |                    ^~~
E:\Arduino files\read_pixel\read_pixel.ino: In function 'void drawStep(int, int, int, uint32_t, uint32_t, uint32_t)':
E:\Arduino files\read_pixel\read_pixel.ino:48:18: error: 'lcd' was not declared in this scope
   48 |     x2 = (i+1) * lcd.width() / step;
      |                  ^~~
E:\Arduino files\read_pixel\read_pixel.ino: In function 'void drawGrad(int, int, uint16_t, uint16_t, uint16_t)':
E:\Arduino files\read_pixel\read_pixel.ino:65:14: error: 'lcd' was not declared in this scope
   65 |     x2 = i * lcd.width() / 256;
      |              ^~~
E:\Arduino files\read_pixel\read_pixel.ino: In function 'void drawTestPattern()':
E:\Arduino files\read_pixel\read_pixel.ino:75:11: error: 'lcd' was not declared in this scope
   75 |   int w = lcd.width() >> 5;
      |           ^~~
E:\Arduino files\read_pixel\read_pixel.ino: In function 'void setup()':
E:\Arduino files\read_pixel\read_pixel.ino:111:3: error: 'lcd' was not declared in this scope
  111 |   lcd.init();
      |   ^~~
E:\Arduino files\read_pixel\read_pixel.ino: In function 'void loop()':
E:\Arduino files\read_pixel\read_pixel.ino:120:3: error: 'lcd' was not declared in this scope
  120 |   lcd.startWrite();
      |   ^~~

exit status 1

Compilation error: 'LGFX' does not name a type

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue ( 再現させるためのコード )
All examples Lovyan
**Please submit complete source code that can reproduce your problem.**

**あなたの問題を再現できる完全なソースコードを提示してください。**

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

tobozo, 2024-10-09:
> step1 load example

what example?

`error: 'LGFX' does not name a type`

this error is typical when the project is missing a Panel / Bus configuration

a few config examples are available here:

https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx_user/

closed 2024-10-09 by tobozo

tonbor, 2024-10-09:
The not compiling examples I used are
test_readpixel.ino,TFT_graphicstest_PDQ.ino. Same error

tobozo, 2024-10-09:
I suggest you look at the [user settings example](https://github.com/lovyan03/LovyanGFX/blob/master/examples/HowToUse/2_user_setting/2_user_setting.ino) then read my first response again