lovyan03/LovyanGFX #628: M5Stack CoreS3 I2C conflict

knifter, 2024-10-15:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name: M5Stack CoreS3Lite
- Panel Driver IC:    AUTODETECT (ILI9341)
- Bus type:           I2C
- LovyanGFX version:  latets (define LGFX_USE_V1)
- FrameWork version:   ArduinoESP32 v2.0.3
- Build Environment:   PlatformIO
- Operating System:   Windows

## Problem Description ( 問題の内容 )
M5Stack defines Wire & Wire1, Wire is is the SYS_I2C on bus nr 0, and configured on  on G11/G12.
LGFX Autodetect configures bus nr 1 (Wire1) on pins G11/G12, breaking 'Wire' from that point on.
Configuring my other I2C devices on Wire1 (port 1), on pins 11/12 makes them and LGFX touch work together. But this breaks the second default configured Wire1, which was/should be configured on G19/G20.
Setting i2c_port = I2C_NUM_0 (LGFX_AutoDetect_ESP32_all.hpp:76) also fixes the problem as this makes LGFX use the same config as was already present.

### Expected Behavior ( 期待される動作 )
LGFX using port nr 0 (== SYS_I2C) for the touch controller.

### Actual Behavior ( 実際の動作 )
LGFX configures port nr 1 (in LGFX_AutoDetect_ESP32_all.hpp, line 76)

### Steps to reproduce ( 再現のための前提条件 )
1. LGFX.init()
2. Use Wire on any device on the internal I2C bus using Wire (also present on MBUS, and used by M5Stack modules)
3. <device does not work and touch readout very slow: conflict because two i2c controllers on same pins)
4. Configure i2c devices to use Wire1 (on G11/G12 now)
5. Devices work, touch works, as workaround

### Code to reproduce this issue ( 再現させるためのコード )
Nothing special:
lgfx.init()
Use Wire for devices on G11/G12, they no longer work after configuring touch callback getTouchRaw is called.
I do not use the M5 library/classes itself. But I guess they could break after LGFX reconfigures pins.

tobozo, 2024-10-15:
hi,

does it also happen with the [develop branch](https://github.com/lovyan03/LovyanGFX/tree/develop)?

knifter, 2024-10-15:
Could test it. Yes, same thing. It still configures I2C port 1 with the same pins as port 0 is already configured to.
The workaround (Use Wire1 on G11/G12) and the fix (76: NUM_1 -> NUM_0)
Is there a reason to use NUM_1 there? The touch is connected to G11/G12 and Wire is (by Arduino framework) configured on NUM_0 with these pins., configuring I2C_NUM_1 there will configure the second i2c peripheral on the same pins. Even with the workaround I now use.

btw: latest version also gives me a warning (which is an error because I use -Werror):
```
lib/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_SPI.cpp:26: error: "REG_SPI_BASE" redefined [-Werror]
   #define REG_SPI_BASE(i)   (DR_REG_SPI1_BASE + (((i)>1) ? (((i)* 0x1000) + 0x20000) : (((~(i)) & 1)* 0x1000 )))
...
C:/Users/tvroon1/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/soc/esp32s3/include/soc/soc.h:122: note: this is the location of the previous definition   
 #define REG_SPI_BASE(i)         (((i)>=2) ? (DR_REG_SPI2_BASE + (i-2) * 0x1000) : (0))    // GPSPI2 and GPSPI3

```
But let's not dwell on that too much. I can work around it. I can make a separate issue if you need it.

tobozo, 2024-10-15:
editing line 76 will affect all M5 devices

does changing the port at line 120 instead [`_cfg.i2c_port = I2C_NUM_0;`](https://github.com/lovyan03/LovyanGFX/blob/develop/src/lgfx/v1_autodetect/LGFX_AutoDetect_ESP32_all.hpp#L120) work for you?

knifter, 2024-10-18:
That's weird as on line 77+78, just below I2C_NUM_1, the I2C gpio pins are set to GPIO_NUM_12 and GPIO_NUM_11 while all other M5 (core) devices have their internal I2C bus on GPIO21 and GPIO22. So if this part is enabled then, those IO pins do not connect to the internal bus which has the touch controller on in.
I've switched platformio to the core and these lines get greyed out. Makes sense as CONFIG_IDF_TARGET_ESP32S3 on line 55 is then not defined. Then line 283 gets enabled, setting I2C to GPIO21/22.

Don't know about ESP32_S3_BOX version though, which this part seems to be made for as well?

It looks to me as some general S3 config (or at least S3_BOX) is applied to the CoreS3 as well which is not compatible with what M5 configures per default. For now autodetection does work in the sense that touch works, but not in the sense that the other I2C devices no longer do..

Thanks btw!