lovyan03/LovyanGFX #635: error: 'lcd_periph_signals' was not declared in this scope; did you mean 'lcd_periph_rgb_signals'?

Westcott1, 2024-11-03:
## Environment ( 実行環境 )

- MCU or Board name:  ESP32-S3
- Panel Driver IC:   LGFX_Sunton_ESP32-8048S070.h
- Bus type:           LGFX_Sunton_ESP32-8048S070.h
- LovyanGFX version:  1.1.16
- FrameWork version:   ArduinoESP32 v3.1.0-RC2
- Build Environment:   ArduinoIDE 2.3.3
- Operating System:   Windows
## Problem Description ( 問題の内容 )
LovyanGFX\src\lgfx\v1\platforms\esp32s3\Bus_RGB.cpp:159:19: error: 'lcd_periph_signals' was not declared in this scope; did you mean 'lcd_periph_rgb_signals'?

lovyan03, 2024-11-05:
Hello, @Westcott1 
please try develop branch.

Westcott1, 2024-11-05:
Hi Lovyan03,
Thanks - sorry but it gives the same error.
The dev version of `LovyanGFX\src\lgfx\v1\platforms\esp32s3\Bus_RGB.cpp` is several months old.

lovyan03, 2024-11-06:
Sorry,, I just update now.
Please try again ...

Westcott1, 2024-11-06:
Yes, all OK now.
Thanks!

closed 2024-11-06 by Westcott1