lovyan03/LovyanGFX #641: Arduino Compilation Issue

brodendan, 2024-11-12:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  ESP32-S3 (Makerlabs MATouch 7" TFT)
- Panel Driver IC:    GT911
- Bus type:           Parallel16
- LovyanGFX version:  v1.1.16
- FrameWork version:   ArduinoESP32 v2.0.3
- Build Environment:   ArduinoIDE2 (V2.3.3)
- Operating System:   macOS Sequoia 15.0.1 (ARM64)

## Problem Description ( 問題の内容 )
Unable to compile Arduino sketch using the LovyanGFX library. Output throws an error related to 'lcd_periph_signals' not being declared in scope (please refer below).

### Expected Behavior ( 期待される動作 )
Successful compilation.

### Actual Behavior ( 実際の動作 )
Error message relating to 'lcd_periph_signals' declaration as below:

```
/Users/user/Arduino/libraries/LovyanGFX/src/lgfx/v1/platforms/esp32s3/Bus_RGB.cpp:159:19: error: 'lcd_periph_signals' was not declared in this scope; did you mean 'lcd_periph_rgb_signals'?
  159 |       auto sigs = lcd_periph_signals.panels[_cfg.port];
      |                   ^~~~~~~~~~~~~~~~~~
      |                   lcd_periph_rgb_signals
/Users/user/Arduino/libraries/LovyanGFX/src/lgfx/v1/platforms/esp32s3/Bus_RGB.cpp:307:31: error: 'lcd_periph_signals' was not declared in this scope; did you mean 'lcd_periph_rgb_signals'?
  307 |     esp_intr_alloc_intrstatus(lcd_periph_signals.panels[_cfg.port].irq_id, isr_flags,
      |                               ^~~~~~~~~~~~~~~~~~
      |                               lcd_periph_rgb_signals
```
The complete output is found here: https://gist.github.com/brodendan/b29762f0bc1aab73dc1f10cf6782509a#file-error-output

### Steps to reproduce ( 再現のための前提条件 )

1. Create Arduino sketch with below source (or copy from https://github.com/Makerfabs/ESP32-S3-Parallel-TFT-with-Touch-7inch/tree/main/example/lovyanGFX/7inch800)
2. Compile sketch.

// If possible, attach a picture of your setup/wiring here.
This is unmodified unit from Makerlabs: https://www.makerfabs.com/esp32-s3-parallel-tft-with-touch-7-inch.html

### Code to reproduce this issue ( 再現させるためのコード )

Please refer to: https://gist.github.com/brodendan/b29762f0bc1aab73dc1f10cf6782509a

tobozo, 2024-11-14:
hi,

Please group all feedback in this issue and close the redundant #623, as it is likely the same solution for both arduino IDE and esp-idf.

LovyanGFX neither supports release versions of arduino-core (this excludes beta/alpha/RC) nor [esp-idf untagged versions](https://github.com/espressif/esp-idf/tags), however it does support `lcd_periph_rgb_signals()` function, si this is likely a bug that will need to be solved when idf 5.4 becomes officially supported.

Possible cause of the problem: LGFX uses the `SOC_LCDCAM_RGB_LCD_SUPPORTED` macro (see [SoC table](https://tanakamasayuki.github.io/ESP-IDF-SoC-Check/soc.html)) to confirm the presence of `lcd_periph_rgb_signals()`, but that macro is disabled or out of scope and the test fails when esp-idf version >= 5.1.4.

```cpp
#if SOC_LCDCAM_RGB_LCD_SUPPORTED // <<<< this test fails
      auto sigs = &lcd_periph_rgb_signals.panels[_cfg.port];
#else
      auto sigs = &lcd_periph_signals.panels[_cfg.port];
#endif
```

a quick solve for you only (this may break compilation for other devices/versions) is to add this after the includes in `esp32s3/Bus_RGB.cpp`

```C
#if (ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 4, 0))
  #define lcd_periph_signals lcd_periph_rgb_signals
#endif
```

please confirm the effectiveness of that temporary workaround so I can think of a cleaner way to address the problem :wink:

tobozo, 2024-11-15:
> did you mean ESP_IDF_VERSION <= 5.4.0 and not >=?

I meant >= 5.4.0 based on [esp-idf commit](https://github.com/espressif/esp-idf/commit/23c41fc77211610d4ca5d26021335f824167a693#diff-9daed3815872509e962b3f254a8de82c47129445c2422e75f495b0c0e4510084L199) where the struct changed happened.

That logic was [applied to a temporary branch](https://github.com/lovyan03/LovyanGFX/blob/bus-s3-rgb/src/lgfx/v1/platforms/esp32s3/Bus_RGB.cpp#L47-L52) and the [CI test didn't fail](https://github.com/lovyan03/LovyanGFX/actions/runs/11839053207/job/32989629276) so I guess that test is valid at least with esp-idf.

I couldn't test with the 3.1.0-RC2 from the arduino repo though, the idf package has an [invalid sha sum and arduino-cli rejects it](https://github.com/lovyan03/LovyanGFX/actions/runs/11839421680/job/32991388986). However as you can see in the error message the arduino-esp32 core is using idf 5.3 so in that situation it should still be using the old `lcd_periph_signals` struct name.

brodendan, 2024-11-16:
Thanks for your help! Let me know if you want any further information about my setup and I'd be happy to provide.

tobozo, 2024-11-16:
happy to help!

can you confirm that you don't get any error when using this branch?

https://github.com/lovyan03/LovyanGFX/blob/bus-s3-rgb/

if so I'll patch the develop branch so that it'll be ready when esp-idf 5.4 is tagged by espressif

brodendan, 2024-11-22:
Hi @tobozo, confirming this branch worked for me!

I removed the old Lovyan library from my machine, downloaded ZIP of your linked repo and installed it as an Arduino library. Successfully compiled the sketch using this branch with no hassle.

Thank you once again!

closed 2024-12-29 by github-actions[bot]