lovyan03/LovyanGFX #649: Need function to determine size in pixels of a string with a specified font

craiglindley, 2024-11-19:
I have a need to display various size text strings on my display. To position them I need to have a function that can give me the size in pixels of a specified string taking into consideration the current font and the current text size. 

I've looked through all the documentation I have found about your library and cannot find a function like this. This new function could be like the 

"getTextBounds" function in the Adafruit GFX library which returns the width and height of the string passed to it. The Adafruit function takes into consideration the current font and the current text size.

tobozo, 2024-11-19:
hi,

text height is usually fixed, only width is variable when the font is not monotype:

```cpp
    tft.setFont(&myFont);
    auto textWidth = tft.getTextWidth("blah");
    auto textHeight = tft.getTextSizeY();
```

craiglindley, 2024-11-19:
Thanks but I didn't find getTextWidth("String") but I did find
textWidth("String")

But thanks, you gave me the answer I needed. I guess the issue could be
closed.

closed 2024-11-20 by tobozo