lovyan03/LovyanGFX #651: ESPboy new batch display stays white

ESPboy-edu, 2024-11-23:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  ESPboy
- Panel Driver IC:    ST7735 (probably new revision)
- Bus type:           SPI
- LovyanGFX version:  latest
- FrameWork version:   Arduino LOLIN WeMos 3.1.2 
- Build Environment:  ArduinoIDE
- Operating System:   macOS

## Problem Description ( 問題の内容 )

A new batch of ESPboy has been released with no change in circuitry or components. Maybe the ST7735 128x128 display is some new version, but visually looks the same as before.
1. Applications created using LovyanGFX on new batch ESPboy consoles do not run the display properly. It remains all white. However, if we take the ESPboy from the old batch, everything works fine.
2. Applications that use TFTeSPI or Adafruit_ST7735 libs on both the old version of the displays and the new version work fine.

I have assumed that the display pin CS is not being set LOW (GND)  so that the SPI works as it should. However, there is no problem with this. Probably some kind of ST7735 initialisation problem.

I am willing to help in troubleshooting the problem, e.g. send an ESPboy from the new batch for testing or I can do experiments according to the instructions.

### Expected Behavior ( 期待される動作 )
Display should work fine.

### Actual Behavior ( 実際の動作 )
Remains white.

### Steps to reproduce ( 再現のための前提条件 )
Install Dino, Pong, 2048, Thermometer, WiFi Scanner from the WebAppStore
https://espboy.m1cr0lab.com/demo/appstore/

// If possible, attach a picture of your setup/wiring here.
https://github.com/m1cr0lab-espboy/Dino?ysclid=m3ub0c3bna30772337

### Code to reproduce this issue ( 再現させるためのコード )
https://github.com/ESPboy-edu/ESPboy_FirstTest/tree/master/ESPboy_FirstTest_ESPboyLibs_LovyanGFX

tobozo, 2024-11-23:
Hi,

please confirm that the latest version of LovyanGFX works with the older batch of ESPBoy, this should help pinpoint the source of the issue.

ESPboy-edu, 2024-11-23:
Yes.  The latest version of LovyanGFX works with the older batch of ESPBoy.
Thanks.

tobozo, 2024-11-23:
thanks, 

maybe the panel id changed, and it's ignored by autodetect?

maybe you can try this after init:

```cpp
uint32_t id = tft.getPanel()->readCommand(0x04, 0, 4);
// expected by autodetect:  (id & 0xFF) == 0x7C
```

tobozo, 2024-11-23:
or you can uncomment [this line](https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx/v1_autodetect/LGFX_AutoDetect_ESP8266.hpp#L148) so that it'll print the id to the console

ESPboy-edu, 2024-11-23:
old batch readed 0x7c
new batch readed 0x00
:(

tobozo, 2024-11-24:
it sounds like the newer display version is not readable, can you try to test `tft.copyRect()` function on both the older and newer display version to see if the results are consistent?

if it is confirmed that the newer display cannot be read, maybe playing with `cfg.freq_read` value will improve things?

providing the specs sheet for that newer display would help a lot I guess

ESPboy-edu, 2024-11-24:
the old 1.44'' 128x128 ST7735 display is also not readable as far as i know.
[image]

here is the LCD
https://aliexpress.ru/item/1005008067134801.html?sku_id=12000043520310231&spm=a2g2w.productlist.search_results.18.50a02d12ArdepM

datasheet
https://github.com/lanrat/CHVBadge_16/blob/master/Defcon%20SDK/doc/Native%20SDK/Hardware/Datasheets/SPEC-Z144SN005.pdf

and the same is used in blue modules like this
https://aliexpress.ru/item/1005005236327965.html?sku_id=12000032314391316&spm=a2g2w.productlist.search_results.8.50a02d12ArdepM

and probably in red modules also
https://aliexpress.ru/item/32948169577.html?sku_id=12000042920524825&spm=a2g2w.productlist.search_results.6.50a02d12ArdepM

ESPboy-edu, 2024-12-25:
Maybe there is no need to read panel id? That would solve the problem for future versions of the library.

tobozo, 2024-12-26:
> I am willing to help in troubleshooting the problem, e.g. send an ESPboy from the new batch for testing or I can do experiments according to the instructions.

can you eventually send one unit of each version to be supported by autodetect? 

if you're okay with that you can pm me via mastodon, twitter or bsky

tobozo, 2025-01-19:
hi and happy nouillere :D 

closing this issue as solved, please submit a PR on the develop branch if you want this added to the autodetect.

closed 2025-01-19 by tobozo