lovyan03/LovyanGFX #662: LiLyGO T4-S3

jgauchia, 2024-12-31:
**Carefully written requests are more likely to be given priority.**
**丁寧に記述された要望は優先して対応される可能性が高くなります。**

### Device Name (デバイスの名称・型番等)
LilyGO T4-S3

Hi, I search a lot of issues, PR, etc.. but nothing is clear.
This device use QSPI interface, that, I know isn't supported by lovyangfx.
I try to add QSPI branch and other PR's about it, but nothing is clear.

Really , I'll use Lovyangfx in order to give support for my firmware, but, if isn't possible or is so difficult I'll need to recode all my project.

Maybe someone can help me , guide me or similar...

Thank you.

tobozo, 2024-12-31:
Hi Jordi,

I've ordered one unit of this gizmo, however 3 weeks before delivery isn't fast enough so I also ordered a T-Display-S3-Amoled that will be delivered on Jan 6th.

meanwhile I'll create a post to ping all the QSPI relevant issues an persons of interest from there

jgauchia, 2024-12-31:
Ok, thank you.. my programming skills are limited, but i don't have any problem to test It, etc

closed 2025-02-06 by github-actions[bot]