lovyan03/LovyanGFX #666: Display problem when WiFi is connected

JPMJPM, 2025-01-09:
- MCU or Board name:  ESP32-S3 H4R2 custom board
- Panel Driver IC: FY07024DI26A67-FT
- Bus type: 16 bits RGB565
- LovyanGFX version:  1.2.0
- FrameWork version: Arduino 2.3.4
- Build Environment: Arduino IDE
- Operating System: Windows 10

Hello, I am attaching an Arduino sketch that I am testing and the text "SCRIPT STARTING" appears correctly on the screen, but when I try to connect to WiFi "WiFi.begin(ssid, password);", the text continually moves from left to right and vice versa.

Any help ?, thank you.

`
// Board : "ESP32S3 Dev Module" 2.0.17 (idf v4.4.7-dirty)
#define LGFX_USE_V1
#include <LovyanGFX.hpp>
#include <lgfx/v1/platforms/esp32s3/Panel_RGB.hpp>
#include <lgfx/v1/platforms/esp32s3/Bus_RGB.hpp>
#include "WiFi.h"

const char* ssid = "ssid";
const char* password = "password";
IPAddress staticIP(192, 168, 1, 101);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 255, 0);
IPAddress dns1(8,8,8,8);
IPAddress dns2(8,8,4,4);

class LGFX : public lgfx::LGFX_Device {
public:
  lgfx::Bus_RGB     _bus_instance;
  lgfx::Panel_RGB   _panel_instance;
  lgfx::Light_PWM   _light_instance;

  LGFX(void) {
    {
      auto cfg = _panel_instance.config();
      cfg.memory_width  = 1024;
      cfg.memory_height = 600;
      cfg.panel_width   = 1024;
      cfg.panel_height  = 600;
      cfg.offset_x      = 0;
      cfg.offset_y      = 0;
      _panel_instance.config(cfg);
    }
    {
      auto cfg = _panel_instance.config_detail();
      cfg.use_psram = 1;
      _panel_instance.config_detail(cfg);
    }
    {
      auto cfg = _bus_instance.config();
      cfg.panel = &_panel_instance;    
      cfg.pin_d0  = GPIO_NUM_37; // B0
      cfg.pin_d1  = GPIO_NUM_36; // B1
      cfg.pin_d2  = GPIO_NUM_35; // B2
      cfg.pin_d3  = GPIO_NUM_18; // B3
      cfg.pin_d4  = GPIO_NUM_15; // B4
      cfg.pin_d5  = GPIO_NUM_17; // G0
      cfg.pin_d6  = GPIO_NUM_16; // G1
      cfg.pin_d7  = GPIO_NUM_10; // G2
      cfg.pin_d8  = GPIO_NUM_14; // G3
      cfg.pin_d9  = GPIO_NUM_5;  // G4
      cfg.pin_d10 = GPIO_NUM_4;  // G5    
      cfg.pin_d11 = GPIO_NUM_6;  // R0
      cfg.pin_d12 = GPIO_NUM_7;  // R1
      cfg.pin_d13 = GPIO_NUM_11; // R2
      cfg.pin_d14 = GPIO_NUM_12; // R3
      cfg.pin_d15 = GPIO_NUM_13; // R4

      cfg.pin_henable = GPIO_NUM_9;
      cfg.pin_vsync   = GPIO_NUM_1;
      cfg.pin_hsync   = GPIO_NUM_3;
      cfg.pin_pclk    = GPIO_NUM_8;
      cfg.freq_write  = 15000000;
      
      cfg.hsync_polarity    = 0;
      cfg.hsync_front_porch = 40;
      cfg.hsync_pulse_width = 48;
      cfg.hsync_back_porch  = 40;     
      cfg.vsync_polarity    = 0;
      cfg.vsync_front_porch = 1;
      cfg.vsync_pulse_width = 31;
      cfg.vsync_back_porch  = 13;
      
      cfg.pclk_active_neg   = 1;
      cfg.de_idle_high      = 0;
      cfg.pclk_idle_high    = 0;
      _bus_instance.config(cfg);
    }
    {
      auto cfg = _light_instance.config();
      cfg.pin_bl = GPIO_NUM_2;
      _light_instance.config(cfg);
    }
    _panel_instance.light(&_light_instance);
    _panel_instance.setBus(&_bus_instance);
    setPanel(&_panel_instance);
  }
};

LGFX lcd;

void setup() {
  Serial.begin(115200);
  Serial.printf("\nESP-IDF version = %s\n", esp_get_idf_version());
 
  Serial.printf("psram_size: %d\n", ESP.getPsramSize());
  lcd.init();
  Serial.printf("free_psram_size: %d\n", ESP.getFreePsram());
  lcd.setRotation(2);
  lcd.setColorDepth(16);
  lcd.setTextSize(1);
  lcd.setTextDatum(middle_center);
  lcd.fillScreen(lcd.color565(0,0,0)); // Black
  lcd.setTextColor(lcd.color565(0,255,255)); // Cyan
  lcd.setBrightness(255);
  lcd.drawString("SCRIPT STARTING", 512, 50);
  
  WiFi.mode(WIFI_STA);
  WiFi.config(staticIP, gateway, subnet, dns1, dns2);
  WiFi.begin(ssid, password);
  
  ////////////// more code lines ......
  
}

void loop() {
  delay(100);
}
`

tobozo, 2025-01-09:
hi,

can you post a pic of your build, just to confirm this isn't caused by overly long wires?

JPMJPM, 2025-01-09:
> hi,
> 
> can you post a pic of your build, just to confirm this isn't caused by overly long wires?

Hello, from a HW point of view my card replaces a Raspberry Pi.

[image]
[image]
[image]
[image]
[image]

tobozo, 2025-01-09:
that looks suspicious

[image]

JPMJPM, 2025-01-10:
Thanks for your help. I will try your suggestion.

closed 2025-02-16 by github-actions[bot]