lovyan03/LovyanGFX #670: ESP32-P4(JC1060P470C)

tanakamasayuki, 2025-01-22:
**Carefully written requests are more likely to be given priority.**
**丁寧に記述された要望は優先して対応される可能性が高くなります。**

[image]

ESP32-P4 + 1024 * 600 MIPI Display(JD9165) + Touch Screen(GT911)

### Device Name (デバイスの名称・型番等)
JC1060P470

### URL of Device Specifications document (仕様書等のURL)

#### Board
http://pan.jczn1688.com/directlink/1/HMI%20display/JC1060P470C_I_W.zip

#### Display driver
https://github.com/espressif/esp-iot-solution/tree/master/components/display/lcd/esp_lcd_jd9165

#### Touch
https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/device/usb_extend_screen/components/espressif__esp_lcd_touch_gt911

### URL of the store where we can purchase (商品を購入できるURL)
https://www.aliexpress.com/item/1005008328088576.html

nubcore, 2025-02-12:
I just ordered some of these. I am working on a similar project and eager to bring this up as a dev board. Hopefully they arrive quickly to Denmark.

Are you affiliated with the board creators or looking to use this for a project yourself?

tanakamasayuki, 2025-02-13:
https://github.com/moononournation/Arduino_GFX

The libraries mentioned above may have affiliations with their creators.

closed 2025-03-22 by github-actions[bot]