lovyan03/LovyanGFX #678: How to use with an ESP-IDF project ?

lepirlouit, 2025-02-13:
Hello,
I tried to use this component with the iot-reference-esp32 project but I'm not able to build my project
I went in the component folder and wrote `git submodule add  git@github.com:lovyan03/LovyanGFX.git`
added `#include <LovyanGFX.hpp>` in a custom task
but still got this error : 
```
fatal error: LovyanGFX.hpp: No such file or directory
   11 | #include <LovyanGFX.hpp>
      |          ^~~~~~~~~~~~~~~
```

tobozo, 2025-02-13:
hi Johan :)

did you reference the component in your/project/src/CMakeLists.txt ?

```cmake
idf_component_register(
  SRCS "main.cpp"
  INCLUDE_DIRS  "."
  REQUIRES LovyanGFX)
```

tobozo, 2025-02-13:
is it `main/main.c` or `main/main.cpp`? it should be `main.cpp`

closed 2025-02-13 by lepirlouit