lovyan03/LovyanGFX #680: ESP-IDF with arduino as component - LGFX crash

hydrafi, 2025-02-23:
esp-idf with arduino as component in windows

- WSC01-PLUS ESP32S3
- LovyanGFX version:  1.2.0
- FrameWork version:   ESP-IDF v5.3
- Build Environment:   VScode
- Operating System:   Windows

The pure ESP32-LVGL8x-SDSPI esp-idf build example runs successfully if the environment is configured correctly.

## Problem Description 
if you add even just the dependency of the arduino component, it compiles but crashes

### Steps to reproduce 
just add in the main dir this yml:
dependencies:
    espressif/arduino-esp32:
      pre_release: true
      version: '*'
      override_path: ${IDF_PATH}/components/arduino-esp32

Compile ok, upload ok, then crash with:
Backtrace: 0x4200a83e:0x3fca8840 0x4200abf4:0x3fca8890 0x420547ff:0x3fca88e0 0x4037c5d1:0x3fca8910

that is:
0x4037c5d1: vPortTaskWrapper at esp-idf_example\ESP32-LVGL8x-SDSPI\build\ esp/v5.3.2/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa\port.c:139

fsender, 2025-03-07:
## I am using LovyanGFX with Arduino-as-a-component-of-ESP-IDF too.
Below is my code.
```C++
#include <Arduino.h>
#define LGFX_M5STICK_C                     // M5Stack M5Stick C / CPlus
#include "LovyanGFX.hpp"
#include <LGFX_AUTODETECT.hpp>

static LGFX ips;

void setup(void)
{
  Serial.begin(115200);
  Serial.println("你好, 测试程序");
//ips.setPanel(ips.getPanelInstance());
  bool initres = ips.init();
  Serial.printf("你好, 初始化完毕, 初始化数值: %s\n", initres?"True":"False");
  Serial.printf("平面地址: 0x%p\n", (void *)ips.getPanel());
  while((void *)ips.getPanel() == nullptr){
    Serial.println("Fail!!!!!");
    ips.setPanel(ips.getPanelInstance());
    delay(100);
  }
    delay(1000);
  int wh = (int)ips.width();
  delay(5000);
  Serial.println("正在获取屏幕信息");
  int ht = (int)ips.height();
  Serial.printf("屏幕信息: 宽度%d, 高度%d.\n", wh, ht);
}
void loop(void)
{}
```
The output is below:
```
--- Warning: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM5 instead...
--- esp-idf-monitor 1.5.0 on \\.\COM5 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6176
load:0x40078000,len:16112
load:0x40080400,len:4
--- 0x40080400: _init at ??:?

load:0x40080404,len:3820
entry 0x400805e4
W (119) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
你好, 测试程序
你好, 初始化完毕, 初始化数值: False
平面地址: 0x0
Fail!!!!!
Fail!!!!!
Fail!!!!!
Fail!!!!!
```
I even edited the library code to show the private variable `_panel`, to watch its value and address.
But I see a strange effect:

THE SAME MEMBER VARIABLE IN A CLASS `LGFX` HAS A DIFFERENT ADDRESS!

My edit:
File LGFX_Base.h Line 1429 --->
```c++
inline Panel_Device* panel(void) const { return reinterpret_cast<Panel_Device*>(_panel); }
    inline const void* getPanelAddr(void) { 
      ESP_LOGW("LGFX", "THIS: %p, PANEL val: %p, PANEL addr: %p", this, _panel, &_panel);
      return reinterpret_cast<const void*>(&_panel);
    }
    inline Panel_Device* getPanel(void) const { 
      ESP_LOGW("LGFX", "getPanel fx - THIS: %p, val: %p, PANEL addr: %p", this, _panel, &_panel);
      if(_panel == nullptr) {
        ESP_LOGE("LGFX", "private ptr _panel is NULL!");
      }
      return reinterpret_cast<Panel_Device*>(_panel); }
```
My output:
```
load:0x40080404,len:3820
entry 0x400805e4
W (119) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
你好, 测试程序
W (20) LGFX: getPanel fx - THIS: 0x3ffb2cd8, val: 0, PANEL addr: 0x3ffb2ce0
E (20) LGFX: private ptr _panel is NULL!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d8fba  PS      : 0x00060c30  A0      : 0x800d80f9  A1      : 0x3ffb7e70  
--- 0x400d8fba: lgfx::v1::LGFX_Device::init_impl(bool, bool) at components/LovyanGFX/src/lgfx/v1/LGFXBase.cpp:3522 (discriminator 1)

Backtrace: 0x400d8fb7:0x3ffb7e70 0x400d80f6:0x3ffb7e90 0x400e0192:0x3ffb7ec0
.--- 0x400d8fb7: lgfx::v1::LGFX_Device::init_impl(bool, bool) at D:/espidf5_prj/readguy_esp32/components/LovyanGFX/src/lgfx/v1/LGFXBase.cpp:3522
0x400d80f6: lgfx::v1::LGFX_Device::init() at D:/espidf5_prj/readguy_esp32/components/LovyanGFX/src/lgfx/v1/LGFXBase.hpp:1410
 (inlined by) setup() at D:/espidf5_prj/readguy_esp32/main/main.cpp:16
0x400e0192: loopTask(void*) at D:/espidf5_prj/readguy_esp32/components/arduino/cores/esp32/main.cpp:59
```

If you try to analyse (via serial. I do not have any debug tools) the addr before `setPanel()` and `init()` the addr will have a 4-byte difference. (Sometimes output like below)
```
load:0x40080404,len:3820
entry 0x400805e4
W (118) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
你好, 测试程序
W (20) LGFX: THIS: 0x3ffb2cd8, PANEL val: 0x3ffb2ddc, PANEL addr: 0x3ffb2cdc
W (233) LGFX: getPanel fx - THIS: 0x3ffb2cd8, val: 0, PANEL addr: 0x3ffb2ce0
你好, 初始化完毕, 初始化数值: True
W (243) LGFX: getPanel fx - THIS: 0x3ffb2cd8, val: 0, PANEL addr: 0x3ffb2ce0
平面地址: 0x0x3ffb2ddc
W (253) LGFX: getPanel fx - THIS: 0x3ffb2cd8, val: 0, PANEL addr: 0x3ffb2ce0
```
Anyway, the program never lights up any pixels on the screen properly.

fsender, 2025-03-08:
Open file `boards.cmake/esp-idf.cmake` and uncomment line
```cmake
list(APPEND COMPONENT_REQUIRES arduino-esp32)
```
then clear and build.

closed 2025-03-09 by hydrafi