lovyan03/LovyanGFX #693: Updating arduino-esp32 to v3.2.0 (ESP-IDF v5.4) causes runtime errors when using LGFX_AUTODETECT

embedded-kiddie, 2025-03-29:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  Both **ESP32 Dev Module** and **ESP32-2432S028R CYD**
- Panel Driver IC:    Both **ILI9341** and **ST7789**
- Bus type:           SPI
- LovyanGFX version:  v1.2.0
- FrameWork version:   ESP32 by Espressif v3.2.0 (ESP-IDF v5.4)
- Build Environment:   ArduinoIDE 2.3.4
- Operating System:   macOS

## Problem Description ( 問題の内容 )

When I compile and upload a sketch with `LGFX_AUTODETECT` for ESP32 2432S028R, nothing is displayed.
This issue does not occur with ESP32 by Espressif v3.1.3 (ESP-IDF v5.3).

### Expected Behavior ( 期待される動作 )

Just display images on LCD as specified by the code.

### Actual Behavior ( 実際の動作 )

Nothing is displayed, but the following messages will be out in serial monitor.

```
E (981) spi: spi_bus_initialize(815): no support changing io default level 
E (982) spi_master: spi_master_init_driver(274): host_id not initialized
E (984) spi: spi_bus_initialize(815): no support changing io default level 
E (990) spi_master: spi_master_init_driver(274): host_id not initialized
E (997) spi: spi_bus_initialize(815): no support changing io default level 
E (1003) spi_master: spi_master_init_driver(274): host_id not initialized
E (1010) spi: spi_bus_initialize(815): no support changing io default level 
E (1017) spi_master: spi_master_init_driver(274): host_id not initialized
E (1023) spi: spi_bus_initialize(815): no support changing io default level 
E (1030) spi_master: spi_master_init_driver(274): host_id not initialized
E (1036) spi: spi_bus_initialize(815): no support changing io default level 
E (1043) spi_master: spi_master_init_driver(274): host_id not initialized
E (1049) spi: spi_bus_initialize(815): no support changing io default level 
E (1056) spi_master: spi_master_init_driver(274): host_id not initialized
E (1063) spi: spi_bus_initialize(815): no support changing io default level 
E (1069) spi_master: spi_master_init_driver(274): host_id not initialized
E (1076) spi: spi_bus_initialize(815): no support changing io default level 
E (1082) spi_master: spi_master_init_driver(274): host_id not initialized
E (1089) spi: spi_bus_initialize(815): no support changing io default level 
E (1096) spi_master: spi_master_init_driver(274): host_id not initialized
```

### Steps to reproduce ( 再現のための前提条件 )

Target board: ESP32 2432S028R (aka Cheap Yellow Display)
Board type: "[ESP32 Dev Module](https://github.com/espressif/arduino-esp32/tree/master/variants/esp32)" or "[ESP32-2432S028R CYD](https://github.com/espressif/arduino-esp32/tree/master/variants/jczn_2432s028r)"
Sketch: [1_simple_use.ino](https://github.com/lovyan03/LovyanGFX/tree/master/examples/HowToUse/1_simple_use)
Define: `LGFX_AUTODETECT` or `LGFX_SUNTON_ESP32_2432S028`

### Code to reproduce this issue ( 再現させるためのコード )

**Please submit complete source code that can reproduce your problem.**
**あなたの問題を再現できる完全なソースコードを提示してください。**

```cpp
#define LGFX_SUNTON_ESP32_2432S028 // Sunton ESP32 2432S028
// #define LGFX_AUTODETECT

#include <LovyanGFX.hpp>
#include <LGFX_AUTODETECT.hpp>

static LGFX lcd;
static LGFX_Sprite sprite(&lcd);

void setup(void)
{
  Serial.begin(115200);
  while (millis() < 1000);

  lcd.init();
  lcd.setRotation(1);
  lcd.setBrightness(128);
  lcd.setColorDepth(16);

  lcd.drawFastVLine(2, 0, 100, lcd.color888(255,   0,   0));
  lcd.drawFastVLine(4, 0, 100, lcd.color565(  0, 255,   0));
  lcd.drawFastVLine(6, 0, 100, lcd.color332(  0,   0, 255));
}
void loop(void) {}
```

## Note

One of the message `E (989) spi_master: spi_master_init_driver(274): host_id not initialized` seems to come from [spi_master.c line 274](https://github.com/espressif/esp-idf/blob/release/v5.4/components/esp_driver_spi/src/gpspi/spi_master.c#L274).

I picked up some related changes from [ESP-IDF v5.4 release note](https://github.com/espressif/esp-idf/releases/tag/v5.4).

- [Commit bde735f](https://github.com/espressif/esp-idf/commit/bde735fe3fc56a7fcf0efa9bbb96f1099a86bb3f)
- [Commit fb52ad1](https://github.com/espressif/esp-idf/commit/fb52ad1d260583c73065cdbdac25300dba170eca)

I don't think this is the issue of LovyanGFX, but I would be happy if LovyanGFX could keep up with the changes in ESP-IDF specifications.

Thank you for the awesome library!

embedded-kiddie, 2025-03-29:
This is the configuration file for board type "**ESP32-2432S028R CYD**". These settings should be the same as when `LGFX_AUTODETECT ` is defined.

```c++
#pragma once

#include <LovyanGFX.hpp>

// false: Panel driver: ILI9341 (micro-USB x 1 type)
// true : Panel driver: ST7789  (micro-USB x 1 + USB-C x 1 type)
#ifndef DISPLAY_CYD_2USB
#error DISPLAY_CYD_2USB should be defined.
#endif

// The pin configuration (CYD_*) is based on the following:
// https://github.com/espressif/arduino-esp32/blob/master/variants/jczn_2432s028r/pins_arduino.h

/// Create a class for your own settings by deriving from LGFX_Device.
class LGFX : public lgfx::LGFX_Device
{
// Select an instance that matches the type of panel.
#if DISPLAY_CYD_2USB
  lgfx::Panel_ST7789      _panel_instance;
#else
  lgfx::Panel_ILI9341     _panel_instance;
#endif

  lgfx::Bus_SPI       _bus_instance;
  lgfx::Light_PWM     _light_instance;
  lgfx::Touch_XPT2046 _touch_instance;

public:
  LGFX(void)
  {
    { //Sets the bus control.
      auto cfg = _bus_instance.config();    // Get the bus configuration structure.

// SPI bus settings
      cfg.spi_host = HSPI_HOST;     // Select the SPI (ESP32-S2,C3: SPI2_HOST or SPI3_HOST / ESP32: VSPI_HOST or HSPI_HOST)
      // Due to the ESP-IDF version upgrade, the VSPI_HOST and HSPI_HOST are deprecated, so if an error occurs, use SPI2_HOST and SPI3_HOST instead.
      cfg.spi_mode = 0;             // SPI communication mode (0 to 3)
#if DISPLAY_CYD_2USB
      cfg.freq_write = 80000000;    // SPI clock for transmit (Maximum 80MHz, rounded to an integer value of 80MHz)
#else
      cfg.freq_write = 40000000;    // SPI clock for transmit (Maximum 80MHz, rounded to an integer value of 80MHz)
#endif
      cfg.freq_read  = 16000000;    // SPI clock for receive
      cfg.spi_3wire  = false;       // Set to true if receive on the MOSI pin
      cfg.use_lock   = true;        // Set to true if transaction lock is used
      cfg.dma_channel = SPI_DMA_CH_AUTO; // Set the DMA channel (0=DMA not used / 1=1ch / 2=2ch / SPI_DMA_CH_AUTO=automatic)
      // Due to the ESP-IDF version upgrade, SPI_DMA_CH_AUTO is recommended. Specifying 1ch or 2ch is no longer recommended.
      cfg.pin_sclk = CYD_TFT_SCK;   // Set the SPI SCLK pin
      cfg.pin_mosi = CYD_TFT_MOSI;  // Set the SPI MOSI pin
      cfg.pin_miso = CYD_TFT_MISO;  // Set the SPI MISO pin (-1 = disable)
      cfg.pin_dc   = CYD_TFT_DC;    // Set the SPI D/C  pin (-1 = disable)
     // When you use a common SPI bus with the SD card, be sure to set MISO and do not omit it.

      _bus_instance.config(cfg);    // Configure setting values in the bus.
      _panel_instance.setBus(&_bus_instance); // Set the bus on the panel.
    }

    { // Configure the display panel control settings.
      auto cfg = _panel_instance.config(); // Get the structure for display panel settings.

      cfg.pin_cs           = CYD_TFT_CS;  // CS   pin number (-1 = disable)
      cfg.pin_rst          = -1;          // RST  pin number (-1 = disable) CYD_TFT_RS = CYD_TFT_CS, RESET is connected to board RST
      cfg.pin_busy         = -1;          // BUSY pin number (-1 = disable)

      // The following are set to general values, so try commenting out if you are unsure of.

      cfg.panel_width      =   240;  // Panel width
[... 42 lines cut ...]
      cfg.y_max =  200;         // Maximum Y value (raw value) from touch screen
      cfg.pin_int = CYD_TP_IRQ; // Interrupt pin number
      cfg.bus_shared = false;   // Set to true if the bus shared with the screen
#if DISPLAY_CYD_2USB
      cfg.offset_rotation = 2;  // Adjust when display and touch orientation do not match (0~7)
#else
      cfg.offset_rotation = 0;  // Adjust when display and touch orientation do not match (0~7)
#endif

// For SPI connection
      cfg.spi_host = -1;            // Select the SPI (HSPI_HOST or VSPI_HOST) or XPT2046 can be set to -1 for Bit Banging
      cfg.freq = 1000000;           // Set the SPI clock
      cfg.pin_sclk = CYD_TP_CLK;    // SCLK pin number
      cfg.pin_mosi = CYD_TP_MOSI;   // MOSI pin number
      cfg.pin_miso = CYD_TP_MISO;   // MISO pin number
      cfg.pin_cs   = CYD_TP_CS;     // CS   pin number

      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);  // Set the touch screen on the panel.
    }

    setPanel(&_panel_instance); // Set the panel to be used.
  }
};
```

dazzor, 2025-04-05:
For the Arduino IDE, in the Boards Manager, I reverted to an older version of the esp32. From 3.2.0 to 3.1.3 and it solved the issue for me. Hope this helps others with this problem. (made me crazy for several days!)

embedded-kiddie, 2025-05-02:
Solved by [1.2.7](https://github.com/lovyan03/LovyanGFX/releases/tag/1.2.7).

closed 2025-05-02 by embedded-kiddie