lovyan03/LovyanGFX #694: SPI bus mind bit order

dizcza, 2025-03-29:
**Carefully written requests are more likely to be given priority.**
Trying to add support for LS027B7DH01 and similar sharp displays using ESP32, I came across a different bit order: these SPI devices are LSB first, while it seems that your code implicitly supports only MSB first devices.

**Is your feature request related to a problem? Please describe.**
I cannot issue a simple "clear display" command available on these sharp displays. The following code does nothing:

```cpp

    static constexpr uint8_t CMD_UPDATE             = 0x01;  // 0x80 in LSB format
    static constexpr uint8_t BIT_VCOM               = 0x02;  // 0x40 in LSB format
    static constexpr uint8_t CMD_CLEAR              = 0x04;  // 0x20 in LSB format
    static constexpr uint8_t VAL_TRAILER            = 0x00;

  struct Panel_SharpLCD : public Panel_HasBuffer
...

  void Panel_SharpLCD::cs_control(bool level)
  {
    auto pin = _cfg.pin_cs;
    if (pin < 0) return;
    ESP_LOGI(TAG, "CS %d level %d", pin, level);
    if (level)   // CS inversed
    {
      lgfx::gpio_lo(pin);
    }
    else
    {
      lgfx::gpio_hi(pin);
    }
  }

  void Panel_SharpLCD::clearDisplay() {
    beginTransaction();  // same as in Panel_HasBuffer

    _bus->writeData(BIT_VCOM | CMD_CLEAR, 8);
    _bus->writeData(VAL_TRAILER, 8);

    endTransaction();
  }
```

The code is ported from https://github.com/adafruit/Adafruit_SHARP_Memory_Display/blob/master/Adafruit_SharpMem.cpp#L216, which works correctly. Because it constructs SPISettings with the LSB first data order inside.

**Describe the solution you'd like**
Add to your `Bus_SPI::config_t` struct something similar to what Adafruit_SPIDevice.h defines:
```c
typedef enum _BitOrder {
  SPI_BITORDER_MSBFIRST = SPI_MSBFIRST,
  SPI_BITORDER_LSBFIRST = SPI_LSBFIRST,
} BusIOBitOrder;
```
or at least add a boolean field (MSB true, LSB otherwise) and, of course, implement the logic in the `Bus_SPI::beginTransaction()`. I personally don't know how to implement the logic there. If it was as simple as
```cpp
  _spiSetting = new SPISettings(freq, dataOrder, dataMode);
  _spi->beginTransaction(*_spiSetting);
```
I'd have added it myself. But I don't see anything like that in your esp32/Bus_SPI.cpp implementation.

dizcza, 2025-03-29:
On second thought, adding `uint8_t bitOrder = SPI_MSBFIRST;` to your config_t seems simpler than defining enum. You already have `freq` and `dataMode` fields from the SPISettings class. The only missing is `bitOrder` to have one-to-one correspondance.

dizcza, 2025-05-08:
Hello,

the quick fix did not work for me. Maybe because I did something wrong elsewhere: it's hard to check all the logic across many connected files in your framework.

The main file is https://github.com/dizcza/LovyanGFX/blob/Sharp_Memory_LCD/src/lgfx/v1/panel/Panel_SharpLCD.cpp. And I'm trying to get the `clearDisplay()` function working. So far no luck.

dizcza, 2025-05-10:
The `refresh` function doesn't work either. It doesn't make sense to test it if a simple clear display command doesn't work.

Adding `_bus->wait()` after `_bus->writeBytes(clear_data, 2, false, false);` in the clearDisplay didn't help.

I don't know how SPI_SWAP_DATA_TX can be helpful for clearDisplay. It might be handy for other functions.

I also tried copying your generic arduino bus instead of custom esp32 SPI bus implementation. There I added SPISettings before every transaction but it didn't help either.

tobozo, 2025-05-19:
Okay this is what I have so far

[image]

Can I start with just this or do I need some extra capacitors/resistors as seen on this [breakout kit](https://kuzyatech.com/wp-content/uploads/2012/08/SharpLCD_BreakOutA1.pdf)? The Nokia 3210 LCD I have doesn't need any of that, so I'm wondering 🤔 

[image]

dizcza, 2025-05-19:
You don't need more than what's in your schematics. In fact, I have no caps at all, and it's working.

I've wired the LCD as straight as they suggest:

[image]

And I pulled DISP high.

[image]
[image]

tobozo, 2025-05-19:
thanks! 

looks like the pins numbering is inverted between the FPC shim and the Sharp tech sheet

dizcza, 2025-05-20:
> looks like the pins numbering is inverted between the FPC shim and the Sharp tech sheet

Yes, it is :) I understood this after I wired it for the first time.

dizcza, 2025-05-21:
Here is the code I'm using. I think I borrowed it directly from Adafruit_SharpMem. I just tested, and it works for me.

```cpp
#include <Adafruit_GFX.h>
#include <Adafruit_SharpMem.h>

#include <SPI.h>

// any pins can be used
#define SHARP_SCK  14
#define SHARP_MOSI 15
#define SHARP_SS   13

Adafruit_SharpMem display(&SPI, SHARP_SS, 400, 240);

#define SQUARE_SIZE 20
#define BLACK 0
#define WHITE 1

int minorHalfSize; // 1/2 of lesser of display width or height

void testdrawline() {
  for (int i=0; i<display.width(); i+=4) {
    display.drawLine(0, 0, i, display.height()-1, WHITE);
    display.refresh();
  }
  for (int i=0; i<display.height(); i+=4) {
    display.drawLine(0, 0, display.width()-1, i, BLACK);
    display.refresh();
  }
  delay(250);

  display.clearDisplay();
  for (int i=0; i<display.width(); i+=4) {
    display.drawLine(0, display.height()-1, i, 0, BLACK);
    display.refresh();
  }
  for (int i=display.height()-1; i>=0; i-=4) {
    display.drawLine(0, display.height()-1, display.width()-1, i, BLACK);
    display.refresh();
  }
  delay(250);

  display.clearDisplay();
  for (int i=display.width()-1; i>=0; i-=4) {
    display.drawLine(display.width()-1, display.height()-1, i, 0, BLACK);
    display.refresh();
  }
  for (int i=display.height()-1; i>=0; i-=4) {
    display.drawLine(display.width()-1, display.height()-1, 0, i, BLACK);
    display.refresh();
  }
  delay(250);

  display.clearDisplay();
  for (int i=0; i<display.height(); i+=4) {
    display.drawLine(display.width()-1, 0, 0, i, BLACK);
    display.refresh();
  }
  for (int i=0; i<display.width(); i+=4) {
    display.drawLine(display.width()-1, 0, i, display.height()-1, BLACK);
    display.refresh();
  }
  delay(250);
}

void testdrawrect(void) {
  for (int i=0; i<minorHalfSize; i+=2) {
    display.drawRect(i, i, display.width()-2*i, display.height()-2*i, BLACK);
    display.refresh();
  }
}

void testfillrect(void) {
  uint8_t color = BLACK;
  for (int i=0; i<minorHalfSize; i+=3) {
    // alternate colors
    display.fillRect(i, i, display.width()-i*2, display.height()-i*2, color&1);
    display.refresh();
    color++;
  }
}

void testdrawroundrect(void) {
  for (int i=0; i<minorHalfSize/2; i+=2) {
    display.drawRoundRect(i, i, display.width()-2*i, display.height()-2*i, minorHalfSize/2, BLACK);
    display.refresh();
  }
}

void testfillroundrect(void) {
  uint8_t color = BLACK;
  for (int i=0; i<minorHalfSize/2; i+=2) {
    display.fillRoundRect(i, i, display.width()-2*i, display.height()-2*i, minorHalfSize/2, color&1);
    display.refresh();
    color++;
  }
}

void testdrawtriangle(void) {
  for (int i=0; i<minorHalfSize; i+=5) {
    display.drawTriangle(display.width()/2, display.height()/2-i,
                     display.width()/2-i, display.height()/2+i,
                     display.width()/2+i, display.height()/2+i, BLACK);
    display.refresh();
  }
}

void testfilltriangle(void) {
  uint8_t color = BLACK;
  for (int i=minorHalfSize; i>0; i-=5) {
    display.fillTriangle(display.width()/2  , display.height()/2-i,
                         display.width()/2-i, display.height()/2+i,
                         display.width()/2+i, display.height()/2+i, color & 1);
    display.refresh();
    color++;
  }
}

void testdrawchar(void) {
  display.setTextSize(1);
  display.setTextColor(BLACK);
  display.setCursor(0,0);
  display.cp437(true);

  for (int i=0; i < 256; i++) {
    if (i == '\n') continue;
    display.write(i);
  }
  display.refresh();
}

void setup()
{
  Serial.begin(115200);
  delay(200);
  log_d("Starting...");

  // start & clear the display
  if (!display.begin()) {
    log_e("Display begin failed!");
    while(1);
  }
  display.clearDisplay();
  log_i("Display begin OK");

  // Several shapes are drawn centered on the screen.  Calculate 1/2 of
  // lesser of display width or height, this is used repeatedly later.
  minorHalfSize = min(display.width(), display.height()) / 2;

  // draw a single pixel
  display.drawPixel(10, 10, BLACK);
  display.refresh();
  delay(500);
  display.clearDisplay();

  // draw many lines
  //testdrawline();
  delay(500);
  display.clearDisplay();

  log_d("testdrawline finished");

  // draw rectangles
  testdrawrect();
  delay(500);
  display.clearDisplay();

  // draw multiple rectangles
  testfillrect();
  display.refresh();
  delay(500);
  display.clearDisplay();

  // draw a circle, 10 pixel radius
  display.fillCircle(display.width()/2, display.height()/2, 10, BLACK);
  display.refresh();
  delay(500);
  display.clearDisplay();

  testdrawroundrect();
  display.refresh();
  delay(500);
  display.clearDisplay();

  testfillroundrect();
  display.refresh();
  delay(500);
  display.clearDisplay();

  testdrawtriangle();
  display.refresh();
  delay(500);
  display.clearDisplay();

  testfilltriangle();
  display.refresh();
  delay(500);
  display.clearDisplay();

  testdrawchar();
  display.refresh();
  for(int i=0; i<4; i++) {
    display.refresh();
    delay(500);
  }
}

void loop(void) 
{
  for(int i=0; i<1; i++) {
    display.setRotation(i);
    display.clearDisplay();
    // text display tests
    display.setTextSize(1);
    display.setTextColor(BLACK);
    display.setCursor(0,0);
    display.println("Hello, world!");
    display.setTextColor(WHITE, BLACK); // inverted text
    display.println(3.141592);
    display.setTextSize(2);
    display.setTextColor(BLACK);
    display.print("0x"); display.println(0xDEADBEEF, HEX);
    // Screen must be refreshed at least once per second
    for(int j=0; j<4; j++) {
      display.refresh();
      delay(500); // 1/2 sec delay
    } // x4 = 2 second pause between rotations
  }
}
```

dizcza, 2025-05-21:
Thanks for looking into this. I sent you a gift that should cover the expenses for the display.

tobozo, 2025-05-23:
wow thanks!

switching to 5v gave better results indeed

so far I've managed to fill and clear the display after adding a `cfg.bit_order` config option and modifying Bus_SPI to handle that option, but I haven't succeeded at sending the panel buffer, probably because I messed up with inheritance (using `struct Panel_SharpLCD : public Panel_1bitOLED` instead of `struct Panel_SharpLCD : public Panel_HasBuffer`), or because I broke Bus_SPI.

I have also looked at other driver implementations, found [this one](https://github.com/AsahiLinux/linux/blob/a5fa95a8dc44442918ed613c7f63e4937eeb27ba/drivers/gpu/drm/tiny/sharp-memory.c#L87-L94) where SPI if set as MSBFIRST and bit order is handled by `bitrev8()`, it may be worth exploring this since it doesn't depend on Bus_SPI to be modified.

dizcza, 2025-05-23:
> I have also looked at other driver implementations, found [this one](https://github.com/AsahiLinux/linux/blob/a5fa95a8dc44442918ed613c7f63e4937eeb27ba/drivers/gpu/drm/tiny/sharp-memory.c#L87-L94) [...]

Yes, this could be an option. I abandoned this approach because I wasn't sure what I should and what should not reverse. For example, whether the data/command address should be reversed as well or not. If you manage with this approach, I think the time needed to reverse the bytes is negligible unless you stream large images.

Do you have a public branch where you're working with the sharp display?

tobozo, 2025-05-24:
indeed everything is affected by bit order including commands

I'm going with this so that I can try both:

```cpp
if( msbfirst ) {
  for(int i=0;i<;bytes_per_line;i++) 
    line[i+1] = bitrev8(_linebuf[i]);
} else {
  memcpy(line+1, _linebuf, bytes_per_line);
}
```

btw clock speed seems to be [hardcoded at 10MHz during init](https://github.com/lovyan03/LovyanGFX/blob/develop/src/lgfx/v1/platforms/esp32/common.cpp#L599), I'm not sure why (SD card maybe?) but manually changing this to 4MHz brought some relief.

what I haven't figured out yet is how to properly use Panel_HasBuffer, I can fillScreen, clearScreen, but `_buf` remains empty when I draw circles or boxes in black or white

I'll push on a branch as soon as I get that buffer thing working and removed the mess I've made in the code :)

dizcza, 2025-05-27:
Nice results! thanks for posting. I'll try your code today.

dizcza, 2025-05-27:
Wow it works thanks!

Seems that you need to set the color to any color (0 and 1 both work) with the `lcd.setColor()` function. Otherwise, it draws nothing. That looks like strange behavior. Maybe we need to set it by default to any value? It's binary either way.

Invert works too.

```cpp
#include "Arduino.h"
#include "lgfx_user/SharpLCD.hpp"

#define SHARP_SCK   14
#define SHARP_MOSI  15
#define SHARP_SS    13

SharpDisplay lcd(SHARP_SCK, SHARP_MOSI, SHARP_SS, 400, 240);

void setup()
{
    Serial.begin(115200);
    delay(200);
    lcd.begin();
    lcd.clearDisplay();
}

void loop()
{
    static int r = 10;
    lcd.setColor(0);
    for (int i = 10; i < 200; i++) {
        lcd.drawPixel(100, i);
    }
    lcd.drawCircle(100, 100, r);
    r += 10;
    if (r > 150) {
        r = 1;
    }
    delay(50);
    lcd.display();
}
```

This loop takes a while (small FPS) perhaps due to slow `drawPixel`. Also, if I increase the loop to
```
    for (int i = 10; i < 290; i++) {
        lcd.drawPixel(100, i);
    }
```
it crashes. I think due to `bitrev8` and going beyond the `byte_rev_table` dimensions.

Initially, the idea was to update the buffer via drawing functions like fillRect that all refer to basic drawPixel and then send buffer to the LCD with the `display` function. If I comment `display`, it still draws. Is it intended?

tobozo, 2025-05-27:
> Seems that you need to set the color to any color (0 and 1 both work) with the lcd.setColor() function. Otherwise, it draws nothing.

This sounds like a standard behaviour (e.g. no color defined = nothing to draw).

you can pass any color when calling the draw function to prevent this:

```cpp
    for (int i = 10; i < 200; i++) {
        lcd.drawPixel(100, i, TFT_BLACK);
    }
    lcd.drawCircle(100, 100, r, TFT_ORANGE);
```

> This loop takes a while (small FPS)

Is that compared to Arduino_GFX/Adafruit_GFX or is that just an expectation? 
It feels slow indeed, and the dithering does affect the performances : the `to_gray()` function is applied to every pixel, then it's dithered even when the source color type is 1bit. 

> bitrev8 and going beyond the byte_rev_table

I'll see why it crashes, but according to the function signature, going beyond the byte_rev_table is impossible, the index argument is an uint8_t so it can't be higher than 0xff and the table size is 256

```cpp
  inline static uint8_t bitrev8(uint8_t byte)
  {
    return byte_rev_table[byte];
  }
```

Panel_HasBuffer has `_autodisplay = true`, when that setting is enabled `lcd.display(x,y,w,h)` is called automatically after every draw.

If you want to prevent this from happening, don't set `_autodisplay=false` in the class constructor, but instead use `lcd.setAutoDisplay(false)` after `lcd.init()`, then you can call `lcd.display()` when necessary.

Meanwhile I'll add a config option and a setter to enable/disable the dithering, implement the missing writeImage/writePixels functions, and investigate the crash with drawPixel.

dizcza, 2025-05-28:
> BTW I couldn't reproduce the crash you had with that code so I guess whatever happened will be fixed with the next update on the experimental branch, which I'll push later in the evening.

This is because you didn't increase the loop limit as I said, and used the code that works for me. Try the following:

```
    for (int i = 10; i < 290; i++) {
        lcd.drawPixel(100, i);
    }
```

Note the upper limit `290`.

Feel free to close this issue and push the changes to the develop branch.

dizcza, 2025-05-29:
It's still triggered:

```
10:08:37.571 > Backtrace: 0x401016b3:0x3ffb4110 0x400d55fd:0x3ffb4130 0x400d4585:0x3ffb4160 0x400d45b3:0x3ffb41f0 0x4010081f:0x3ffb4210 0x400d41ea:0x3ffb4230 0x400d5f84:0x3ffb4250 0x400dbcf4:0x3ffb4270 0x4008bb62:0x3ffb4290
10:08:37.711 >   #0  0x401016b3 in lgfx::v1::Bus_SPI::dc_control(bool) at src/lgfx/v1/platforms/esp32/Bus_SPI.hpp:152
10:08:37.711 >   #1  0x400d55fd in lgfx::v1::Bus_SPI::writeBytes(unsigned char const*, unsigned long, bool, bool) at src/lgfx/v1/platforms/esp32/Bus_SPI.cpp:642
10:08:37.711 >   #2  0x400d4585 in lgfx::v1::Panel_SharpLCD::display(unsigned char*, unsigned int, unsigned int, unsigned int, unsigned int) at src/lgfx/v1/panel/Panel_SharpLCD.cpp:234
10:08:37.711 >   #3  0x400d45b3 in lgfx::v1::Panel_SharpLCD::display(unsigned int, unsigned int, unsigned int, unsigned int) at src/lgfx/v1/panel/Panel_SharpLCD.cpp:185
10:08:37.711 >   #4  0x4010081f in lgfx::v1::IPanel::endWrite() at src/lgfx/v1/panel/../Panel.hpp:88 (discriminator 4)
10:08:37.711 >   #5  0x400d41ea in lgfx::v1::Panel_HasBuffer::drawPixelPreclipped(unsigned int, unsigned int, unsigned long) at src/lgfx/v1/panel/Panel_HasBuffer.cpp:105
10:08:37.711 >   #6  0x400d5f84 in lgfx::v1::LGFXBase::drawPixel(long, long) at src/lgfx/v1/LGFXBase.hpp:162
10:08:37.711 >       (inlined by) loop() at src/main.cpp:36
10:08:37.711 >   #7  0x400dbcf4 in loopTask(void*) at /home/dizcza/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:74
10:08:37.711 >   #8  0x4008bb62 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

tobozo, 2025-05-29:
I've had some Exceptions pointing out dc_control too but it was always the consequence of another problem e.g. when the ESP was very low on memory after I tried to allocate a double buffer on my esp32-wroom. 

it could also mean something is wrong with how Panel_SharpLCD handles transations so I'll investigate this

tobozo, 2025-05-29:
well I didn't find any probable cause but fixed an unrelated bug that had the last line of the buffer not being sent. 

I'm very grateful for the gift, thanks for motivating this, it was a lot of fun 😍 

I'll push the Panel to the develop branch now, and later create a gist with the high-fps demo because none of the videos I shot managed to reproduce the real experience :D

dizcza, 2025-05-29:
> I've had some Exceptions pointing out dc_control too but it was always the consequence of another problem e.g. when the ESP was very low on memory after I tried to allocate a double buffer on my [...]

There is no need to check this function as it happened due to the timer watchdog not being fed (`delay` wasn't called) for several seconds in my loop function.

Sure, pleasure to work with you.

I'm looking at the `_draw_pixel` function in the commit you referenced and found

```
swap565_t color;
color.raw = rawcolor;
uint32_t value = to_gray(color.R8(), color.G8(), color.B8());
```

why do you convert the color to gray and not to a binary value? The LS027B7DH01 is a 1-bit (monochrome) display - it can only show black and white pixels, not grayscale values. It'll be for sure faster something like

```
uint32_t value = rawcolor > 0;
```