lovyan03/LovyanGFX #695: LovyanGFX does not work with the Wire.h library for the I2C bus

2serg2, 2025-03-29:
- MCU or Board name:  ESP32-S3-DevKit N16R8
- Panel Driver IC:   ILI9486
- Bus type:           Parallel8
- LovyanGFX version:  1.2.0
- FrameWork version:   ArduinoESP32 v3.2.0
- Build Environment:   ArduinoIDE v2.3.4
- Operating System:   Windows 11

After updating to ESP32 by Espressif v3.2.0, the ESP32 S3 microcontroller starts rebooting with the following error upon power-up:

```
E (106) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver
abort() was called at PC 0x420143af on core 0
```
Test sketch:
```cpp
#include <Wire.h>  // Library for I2C bus operation

#define LGFX_USE_V1
#include <LovyanGFX.hpp>
class LGFX : public lgfx::LGFX_Device {
  lgfx::Panel_ILI9486 _panel_instance; 
  lgfx::Bus_Parallel8 _bus_instance; 
public:
  LGFX(void) {
    {                                     //设置显示面板控制。
      auto cfg = _bus_instance.config();  // 获取用于总线设置的结构。
      // 8位并行总线设置
      //cfg.i2s_port = I2S_NUM_0;   // 选择要使用的I2S端口（I2S_NUM_0 or I2S_NUM_1）（使用ESP32的I2S LCD模式）
      cfg.freq_write = 30000000;  // 发送时钟（最大可将20MHz、80MHz舍入为整数）
      cfg.pin_wr = 16;            // 连接WR的针脚编号
      cfg.pin_rd = 15;            // 连接RD的针脚编号
      cfg.pin_rs = 6;             // 连接RS（D/C）的针脚编号  1 2 3  12

      cfg.pin_d0 = 2;   // 连接D0的管脚编号
      cfg.pin_d1 = 1;   // 连接D1的管脚编号
      cfg.pin_d2 = 47;  // 连接D2的管脚编号
      cfg.pin_d3 = 38;  // 连接D3的管脚编号
      cfg.pin_d4 = 39;  // 连接D4的管脚编号
      cfg.pin_d5 = 40;  // 连接D5的管脚编号
      cfg.pin_d6 = 41;  // 连接D6的管脚编号
      cfg.pin_d7 = 42;  // 连接D7的管脚编号

      _bus_instance.config(cfg);               // 在总线上反映设置值。
      _panel_instance.setBus(&_bus_instance);  // 把总线设置在面板上。
    }

    {                                       //设置显示面板控制。
      auto cfg = _panel_instance.config();  // 获取用于设置显示面板的结构。
      cfg.pin_rst = 3;                     // 连接RST的管脚编号（-1=disable）
      cfg.pin_cs = 5;                      // 连接CS的管脚编号（-1=disable）
      cfg.pin_busy = -1;                    // 连接BUSY的管脚编号（-1=disable）
      cfg.panel_width = 320;                // 実際に表示可能な幅
      cfg.panel_height = 480;               // 実際に表示可能な高さ
      cfg.offset_x = 0;                     // パネルのX方向オフセット量
      cfg.offset_y = 0;                     // パネルのY方向オフセット量
      cfg.offset_rotation = 0;              // 回転方向の値のオフセット 0~7 (4~7は上下反転)
      cfg.dummy_read_pixel = 8;             // ピクセル読出し前のダミーリードのビット数
      cfg.dummy_read_bits = 1;              // ピクセル以外のデータ読出し前のダミーリードのビット数
      cfg.readable = true;                  // データ読出しが可能な場合 trueに設定
      cfg.invert = false;                   // パネルの明暗が反転してしまう場合 trueに設定
      cfg.rgb_order = false;                // パネルの赤と青が入れ替わってしまう場合 trueに設定
      cfg.dlen_16bit = false;               // 16bitパラレルやSPIでデータ長を16bit単位で送信するパネルの場合 trueに設定
      cfg.bus_shared = false;               // SDカードとバスを共有している場合 trueに設定(drawJpgFile等でバス制御を行います)

      _panel_instance.config(cfg);
    }
    setPanel(&_panel_instance);  // 设置要使用的面板。
  }
};
static LGFX tft; 

void setup() {
}

void loop() {
}
```

2serg2, 2025-04-01:
To disable the I2C bus, you need to comment out the following lines in the "common.cpp" file:

```cpp
//#include <driver/i2c.h>
...
//#include <soc/i2c_reg.h>
//#include <soc/i2c_struct.h>
```
File path:
C:\Users\User\Documents\Arduino\libraries\LovyanGFX\src\lgfx\v1\platforms\esp32\common.cpp
[GitHub link](https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx/v1/platforms/esp32/common.cpp)
Additionally, remove most of the code from line 683 to line 1547.

[image]

tobozo, 2025-04-01:
hi,

did you try with the [develop branch](https://github.com/lovyan03/LovyanGFX/tree/develop)? 

it implements the [new I2C master driver](https://github.com/lovyan03/LovyanGFX/commits/develop/)

2serg2, 2025-04-01:
I didn't see this branch. I installed the library from the Arduino IDE. Now I've replaced it with [this version](https://github.com/lovyan03/LovyanGFX/tree/develop), and everything works by default.
Where were you before? I almost hanged myself when my project stopped working! 😂

tobozo, 2025-04-01:
[the issue template](https://github.com/lovyan03/LovyanGFX/blob/master/.github/ISSUE_TEMPLATE/bug-report--------.md) should probably be updated given how often this solution is suggested, or maybe an automated reply: "have you tried to turn it on an off again on the develop branch?"

sorry I intended to delete a duplicate comment from me and ended up deleting one of yours, can't say if I misclicked or if it's just github being github 🤭

2serg2, 2025-04-01:
Good idea! An automatic reply with this suggestion could really help reduce the number of repeated questions. Maybe it's worth discussing with the team. Not all users know that GitHub has branches I forgot about that myself.
I deleted my comment in another thread and moved it here.

Westcott1, 2025-04-10:
Glad I found this post, I was just ready to submit an identical error report.
Using the dev branch has fixed the problem, thanks Tobozo!
now all I have to do is find out how to fix these -

`E (29983) i2c.master: I2C transaction unexpected nack detected`

2serg2, 2025-04-10:
I have a display with an 8-bit parallel interface, so I don't use the I2C bus or SPI interface for the display. However, during compilation, these protocols are still included and take up space in RAM and FLASH memory. Therefore, I disabled these protocols as described above. For my connected devices, I use the following libraries:
```cpp
#include <SPI.h>
#include <Wire.h>
```

closed 2025-05-17 by github-actions[bot]