lovyan03/LovyanGFX #696: Issue with tft.readRect() after updating to ESP32 from Espressif v3.2.0 on ESP32-S3

2serg2, 2025-04-06:
- MCU or Board name:  ESP32-S3-DevKit N16R8
- Panel Driver IC:   ILI9486
- Bus type:           Parallel8
- LovyanGFX version:  1.2.0
- FrameWork version:   ArduinoESP32 v3.2.0
- Build Environment:   ArduinoIDE v2.3.5
- Operating System:   Windows 11

After updating the ESP32 from Espressif to version v3.2.0, the tft.readRect() function stopped working.  It was working fine before the update.

Test sketch:
```cpp #define LGFX_USE_V1
#include <LovyanGFX.hpp>
class LGFX : public lgfx::LGFX_Device {
  lgfx::Panel_ILI9486 _panel_instance; 
  lgfx::Bus_Parallel8 _bus_instance; 

public:
  LGFX(void) {
    {                                     //设置显示面板控制。
      auto cfg = _bus_instance.config();  // 获取用于总线设置的结构。

      // 8位并行总线设置
      //cfg.i2s_port = I2S_NUM_0;   // 选择要使用的I2S端口（I2S_NUM_0 or I2S_NUM_1）（使用ESP32的I2S LCD模式）
      cfg.freq_write = 20000000;  // 发送时钟（最大可将20MHz、80MHz舍入为整数）
      cfg.pin_wr = 16;            // 连接WR的针脚编号
      cfg.pin_rd = 17;            // 连接RD的针脚编号
      cfg.pin_rs = 18;             // 连接RS（D/C）的针脚编号  1 2 3  12

      cfg.pin_d0 = 1;   // 连接D0的管脚编号
      cfg.pin_d1 = 2;   // 连接D1的管脚编号
      cfg.pin_d2 = 3;  // 连接D2的管脚编号
      cfg.pin_d3 = 4;  // 连接D3的管脚编号
      cfg.pin_d4 = 5;  // 连接D4的管脚编号
      cfg.pin_d5 = 6;  // 连接D5的管脚编号
      cfg.pin_d6 = 7;  // 连接D6的管脚编号
      cfg.pin_d7 = 14;  // 连接D7的管脚编号

      _bus_instance.config(cfg);               // 在总线上反映设置值。
      _panel_instance.setBus(&_bus_instance);  // 把总线设置在面板上。
    }

    {                                       //设置显示面板控制。
      auto cfg = _panel_instance.config();  // 获取用于设置显示面板的结构。
      cfg.pin_rst = 21;                     // 连接RST的管脚编号（-1=disable）
      cfg.pin_cs = 15;                      // 连接CS的管脚编号（-1=disable）
      cfg.pin_busy = -1;                    // 连接BUSY的管脚编号（-1=disable）
      cfg.panel_width = 320;                // 実際に表示可能な幅
      cfg.panel_height = 480;               // 実際に表示可能な高さ
      cfg.offset_x = 0;                     // パネルのX方向オフセット量
      cfg.offset_y = 0;                     // パネルのY方向オフセット量
      cfg.offset_rotation = 0;              // 回転方向の値のオフセット 0~7 (4~7は上下反転)
      cfg.dummy_read_pixel = 8;             // ピクセル読出し前のダミーリードのビット数
      cfg.dummy_read_bits = 1;              // ピクセル以外のデータ読出し前のダミーリードのビット数
      cfg.readable = true;                  // データ読出しが可能な場合 trueに設定
      cfg.invert = false;                   // パネルの明暗が反転してしまう場合 trueに設定
      cfg.rgb_order = false;                // パネルの赤と青が入れ替わってしまう場合 trueに設定
      cfg.dlen_16bit = false;               // 16bitパラレルやSPIでデータ長を16bit単位で送信するパネルの場合 trueに設定
      cfg.bus_shared = false;               // SDカードとバスを共有している場合 trueに設定(drawJpgFile等でバス制御を行います)

      _panel_instance.config(cfg);
    }
    setPanel(&_panel_instance);  // 设置要使用的面板。
  }
};
static LGFX tft; 

void setup() {
 tft.init();
 tft.setRotation(1);
 tft.fillScreen(TFT_BLACK);

 tft.drawRect(0, 0, 50, 50, TFT_WHITE);
 uint16_t *imageBuffer = (uint16_t *)malloc(2 * 50 * 50);
 tft.readRect(0, 0, 50, 50, imageBuffer);
 tft.pushRect(0, 0, 50, 50, imageBuffer); 
 free(imageBuffer);
}

void loop() {
}
```
Has anyone else encountered this issue or knows how to fix it? I help would be appreciated!

closed 2025-05-13 by github-actions[bot]

2serg2, 2025-05-29:
When version 1.2.7 was first released, I tried it, but the tft.readRect() function didn’t work — it didn’t read any data from the display. I can test it again if needed.

tobozo, 2025-05-29:
> After updating the ESP32 from Espressif to version v3.2.0

it could be caused by recent changes in esp-idf e.g. using deprecated function in i2s or new config settings popped and need to be initialized

can you enable verbose output in Arduino prefs and set the compiler warnings to `All` then look for warning/notices in the console when compiling?

[image]

tobozo, 2025-06-02:
I see a couple of warnings but they're unrelated.

If you feel adventurous I'm trying to backport some recent changes from M5GFX on [this branch](https://github.com/lovyan03/LovyanGFX/tree/M5GFX_Backport). Some of the changes are bus related, other are color-type related, it's worth trying to see if the situation improves for tft.readRect()

btw there's also tft.readRectRGB()

tobozo, 2025-06-03:
looks like it manages to copy a few pixels before losing it, let's see if this is related to Bus_QSPI and make a couple of experiments: 

- temporarily set `cfg.freq_write` to 4MHz
- tft.readRect()->pushRect() each pixel separately with `delay(1)` between each draw
- confirm that tft.readRectRGB() has the same behaviour

2serg2, 2025-06-04:
cfg.freq_write = 4000000; has no effect — it doesn't change the behavior.
This code works correctly:
```cpp
tft.drawRect(0, 0, 50, 50, TFT_WHITE);
uint16_t *imageBuffer = (uint16_t *)malloc(2 * 50 * 50);
uint16_t color;
for (int y = 0; y < 50; y++) {
  for (int x = 0; x < 50; x++) {
    tft.readRect(x, y, 1, 1, &color); 
    tft.drawPixel(x, y, color);
    delay(1);
  }
}
// tft.readRect(0, 0, 50, 50, imageBuffer);
// tft.pushRect(0, 0, 50, 50, imageBuffer);
free(imageBuffer);
```
[image]
If I remove the delay(1); line, small gaps appear in the image.
[image]
I don't know how to display the buffer read using tft.readRectRGB(0, 0, 50, 50, imageBuffer);

tobozo, 2025-06-04:
> This code works correctly:
> ...
>     delay(1);
> ...
>  If I remove the delay(1); line, small gaps appear in the image.

so reading slower (and less) makes the symptom go away, but a slower bus clock doesn't 🤔 

auditing the bug: it would be interesting to know exactly how many pixels in a single readRect() call can be read before the bug occurs

idea: did you try to poke those values? 

```
      cfg.dummy_read_pixel = 8;             // ピクセル読出し前のダミーリードのビット数
      cfg.dummy_read_bits = 1;              // ピクセル以外のデータ読出し前のダミーリードのビット数
```

> I don't know how to display the buffer read using tft.readRectRGB(0, 0, 50, 50, imageBuffer);

use pushImage and send the buffer as uint32_t array

```cpp
size_t pixelsCount = 50*50;
size_t bufSize = pixelsCount * sizeof(uint32_t);
uint32_t *imageBuffer = (uint32_t *)malloc(bufSize);
for(int i=0;i<pixelsCount;i++) {
   imageBuffer[i] = 0xFF0000; // red
}

tft.pushImage(0, 0, 50, 50, imageBuffer);
tft.drawRect(0, 0, 50, 50, TFT_WHITE);
tft.readRectRGB(0, 0, 50, 50, imageBuffer);
tft.pushImage(0, 0, 50, 50, imageBuffer);
```

2serg2, 2025-06-05:
I tried setting cfg.dummy_read_pixel = 8, as well as 16, 4 and 1 — but nothing changes.
```cpp
  tft.fillRect(0, 0, 50, 50, TFT_RED);
  uint16_t *imageBuffer = (uint16_t *)malloc(2 * 50 * 50);
  tft.readRect(0, 0, 50, 50, imageBuffer);
  tft.pushRect(0, 0, 50, 50, imageBuffer);
  free(imageBuffer);
```
When I draw a filled white square, I get this result:

[image]

When I draw a filled red square, the color is distorted and some pixels are missing:

[image]

Also, it seems `tft.readRectRGB(0, 0, 50, 50, imageBuffer);` does not accept a uint32_t *imageBuffer.

2serg2, 2025-06-07:
"I tried adding CMD_RCOLMOD, but it had no effect.

[image]

When using tft.readRectRGB(), a red square is displayed as white, and some pixels are missing or skipped."

```cpp
  size_t width = 50, height = 50;
  size_t pixelsCount = width * height;
  size_t bufSize = pixelsCount * 3;  // 3 байта на пиксель для RGB888

  uint8_t *imageBuffer = (uint8_t *)malloc(bufSize);

  tft.fillRect(0, 0, width, height, TFT_RED);
  delay(2000);
  tft.readRectRGB(0, 0, width, height, imageBuffer);
  tft.pushImage(0, 0, width, height, imageBuffer);

  free(imageBuffer);
```

[image]

When using tft.redirect(), a red square is displayed as purple, and some pixels are missing or skipped.

[image]