lovyan03/LovyanGFX #699: Support for very common touch screen JC3248W535 (3.5", ESP32-S3 N16R8)

knopserl, 2025-04-21:
**Carefully written requests are more likely to be given priority.**
It would be great to get also support for this very common 3.5" touch screen with an ESP32-S3 N16R8 chip.
It is a good alternative to the WT32-SC01 plus (ZX3D50CE02S) and it's around 50% cheaper

### Device Name (デバイスの名称・型番等)
JC3248W535

### URL of Device Specifications document (仕様書等のURL)
some URL's
https://github.com/NorthernMan54/JC3248W535EN
https://spotpear.com/shop/ESP32-S3-3.5-inch-LCD-Captive-TouchScreen-Display-480x320-Tablet-MP3-Video-Weather-Clock.html
[http://pan.jczn1688.com/directlink/1/HMI%20display/JC3248W535EN.zip](http://pan.jczn1688.com/directlink/1/HMI%20display/JC3248W535EN.zip?spm=a2g0o.detail.1000023.1.5ed4zm7zzm7zSN&file=JC3248W535EN.zip)
This library supports it already, but it's slower. has no touch support and difficult to support both display types
https://github.com/moononournation/Arduino_GFX 

### URL of the store where we can purchase (商品を購入できるURL)
https://s.click.aliexpress.com/e/_DFO5uIV
https://spotpear.com/shop/ESP32-S3-3.5-inch-LCD-Captive-TouchScreen-Display-480x320-Tablet-MP3-Video-Weather-Clock.html

thanks a lot

knopserl, 2025-05-25:
github marked this as stale and will close it!?
Is there any reason, why there is no interest to support this very popular touch display!
Any chance to get support for it?

knopserl, 2025-05-27:
> I'll give it a go tomorrow and see how things are. I haven't messed with the new changes yet. The display part at least, no idea how touch will work as it is one controller for both. So probably still needs another touch library like bb_captouch which it works with or touch implemented here based on that or something which is beyond my skill level.

here would be a working "driver" for the touch feature of the  AXS15231B 

[Touch_AXS15231B.zip](https://github.com/user-attachments/files/20457975/Touch_AXS15231B.zip)

Shadowtrance, 2025-05-27:
I've tried that library with this display and it doesn't work correctly (at least with lvgl) as it is kinda special as mentioned [here](https://github.com/esp-arduino-libs/ESP32_Display_Panel/issues/164#issuecomment-2668006767)
I've tested various graphics libraries with it including pure esp-idf drivers and the only one it seems to work correctly with is Arduino_GFX.
It is a pity it's such a pain to get working correctly because it is a nice display and yeah, cheaper than similar displays like the WT32-SC01 / SC01 plus. :(

As for it working with lovyanGFX, i haven't gotten it working correctly yet and probably won't have time to figure it out for a while (if i do figure it out at all) because i have to move house soon after 16 years in the same house so LOTS of packing and cleaning to do. :(

knopserl, 2025-08-07:
I think this request will match to a certain extent with the open QSPI display support, which should be also meanwhile be supported by the latest ESP32 SDK/API. There needs probably also the proper init sequence for the  AXS15231B controller which also handles the touch screen. The Arduino GFX library supports the display already (could act as a template), but I rather would like to see it supported by this library.