lovyan03/LovyanGFX #707: GT911 Touch coordinate range is not as large as the display resolution

zalexzperez, 2025-05-02:
## Environment ( 実行環境 )

- MCU or Board name: ESP32-S3 WROOM FREENOVE
- Display module [datasheet](https://28392816.s21i.faiusr.com/61/ABUIABA9GAAg2eelmQYo_qLZ1gc.pdf)
- Panel Driver IC: ST7796
- Touch Panel Controller IC: GT911
- Bus type: Parallel8 (LCD) & I2C (touch)
- LovyanGFX version:  v1.2.7
- FrameWork version: ArduinoESP32 v2.0.17
- Build Environment: PlatformIO
- Operating System: Windows

## Problem Description ( 問題の内容 )
Touch panel is not reporting the full range of expected coordinates regardless of the set rotation and configured x_max / y_max values.

### Expected Behavior ( 期待される動作 )
Returned touch coordinates should span the entire display resolution (320x480)

### Actual Behavior ( 実際の動作 )
The touch values on the shorter side of the display only go up to half of one of the axis max value (e.g. if  cfg.y_max = 319, the reported touch values go from 0 to 159 instead of 319).

### Steps to reproduce ( 再現のための前提条件 )

1. Upload code to the board
2. Touch on display corners

With the following configuration:

```
{ // Display panel configuration
      auto cfg = _panel_instance.config();
      cfg.pin_cs          = 13;
      cfg.pin_rst         = 7;
      cfg.pin_busy        = -1;
      cfg.panel_width     = 320;
      cfg.panel_height    = 480;
      cfg.offset_x        = 0;
      cfg.offset_y        = 0;
      cfg.memory_width    = 320;
      cfg.memory_height   = 480;
      cfg.offset_rotation = 0;
      cfg.readable        = true;
      cfg.invert          = true; // IPS panel
      cfg.rgb_order       = false;
      cfg.dlen_16bit      = false;
      cfg.bus_shared      = false;
      
      _panel_instance.config(cfg);
    }

    { // Touch panel configuration
      auto cfg = _touch_instance.config();
      cfg.x_min = 0;
      cfg.x_max = 319;
      cfg.y_min = 0;
      cfg.y_max = 479;

      cfg.bus_shared = false; 
      cfg.offset_rotation = 0; // Adjustment when the display and touch orientation do not match Set with a value from 0 to 7

      cfg.pin_int = 38;       // Replace with your actual INT pin
      cfg.pin_rst = 1;       // Replace with your actual RST pin
      cfg.i2c_port = 1;      // Usually 0 or 1 for ESP32
      cfg.i2c_addr = 0x14;   // GT911 default I2C address is 0x5D
      cfg.freq = 400000;     // I2C frequency
      cfg.pin_sda = 41;       // Replace with your SDA pin
      cfg.pin_scl = 42;       // Replace with your SCL pin
      
      
      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);
    }
```

This is what I see: 

[image]

If I swap the touch panel x_max and y_max values like this:

```
  // Touch panel configuration
  auto cfg = _touch_instance.config();
  cfg.x_min = 0;
  cfg.x_max = 479;
  cfg.y_min = 0;
  cfg.y_max = 319;
    
```

these are the returned touch coordinates:

[image]

As you can see, with the initial config, the shorter side goes up to 239 (half of y_max = 479). On the second config, the shorter side goes up to 159 (half of x_max = 319), but in this case, the longer side goes from 0 to 479 as it should.

I already tried cfg.offset_rotation  friom 0 to 7 on the touch panel configuration with no luck.

### Code to reproduce this issue ( 再現させるためのコード )
https://gist.github.com/zalexzperez/9bc1e323d0f92ecf2535597905acc963

tobozo, 2025-05-04:
hi,

`cfg.x_max` and `cfg.y_max` should match the display resolution+orientation, **or not be set at all**.

What happens when you use your initial working configuration and call `tft.setRotation(1)`?

Expected results: touch coordinates should be rotated too

zalexzperez, 2025-05-04:
> hi,
> 
> `cfg.x_max` and `cfg.y_max` should match the display resolution+orientation, **or not be set at all**.
> 
> What happens when you use your initial working configuration and call `tft.setRotation(1)`?
> 
> Expected results: touch coordinates should be rotated too

Hello, thanks for your help.

If the initial touch configuration is used:

```
cfg.x_max = 319;
cfg.y_max = 479;
```
Then this is what I get:

[image]

It seems like the touch panel grid has a lower resolution of 240x320. I didn't know using lower resolution touch panels was normal.

Now, if we configure it like this, keeping tft.setRotation(1):
```
cfg.x_max = 239;
cfg.y_max = 319;
```
we finally get the right coordinate ranges:

[image]

closed 2025-05-04 by zalexzperez