lovyan03/LovyanGFX #708: No backlight after updating to esp-idf 5.4.1 + Arduino core 3.2.0

plystdr, 2025-05-05:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  Custom esp32-s3 with 8mb flash 2mb psram
- Panel Driver IC:    [ILI9341|SSD1963|ST7796|other]
- Bus type:           [SPI|I2C|Parallel8|Parallel16|other]
- LovyanGFX version:  [v0.4.17|latest develop|other]
- FrameWork version:   [ArduinoESP32 v2.0.3|ESP-IDF v4.4|other]
- Build Environment:   [ArduinoIDE|PlatformIO|Eclipse|other]
- Operating System:   [Windows|Linux|macOS]

## Problem Description ( 問題の内容 )
After updating esp-idf to latest version with Arduino core, I don't get any backlight. and receive ledc error on serial output. Before updating I was on LovyanGFX 1.1.16.

Updating esp-idf forced me to update to 1.2.7.

```
/////////////////////////////////////////////////////////////////
/*
	ESP32 | LVGL8 | Ep 0. GFX Setup (ft. LovyanGFX)
	Video Tutorial: https://youtu.be/IPCvQ4o_WP8
	Created by Eric N. (ThatProject)
*/
/////////////////////////////////////////////////////////////////

#include <lvgl.h>
// #define LGFX_USE_V1
#include <LovyanGFX.hpp>
#include "ui.h"

class LGFX : public lgfx::LGFX_Device
{
	lgfx::Panel_ST7796 _panel_instance;
	lgfx::Bus_SPI _bus_instance; // SPI
	lgfx::Light_PWM _light_instance;
	lgfx::Touch_XPT2046 _touch_instance;

public:
	LGFX(void)
	{
		{
			auto cfg = _bus_instance.config();
			cfg.spi_host = SPI3_HOST;					 	// (VSPI_HOST or HSPI_HOST)
			cfg.spi_mode = 0;								// SPI (0 ~ 3)
			cfg.freq_write = 40000000;				 		// SPI (80MHz)
			cfg.freq_read = 16000000;					 	//
			cfg.spi_3wire = false;						 	//
			cfg.use_lock = true;							//
			cfg.dma_channel = SPI_DMA_CH_AUTO; 				// Set the DMA channel (1 or 2. 0=disable)
			cfg.pin_sclk = 16;								// LCD_SCLK
			cfg.pin_mosi = 17;								// LCD_MOSI
			cfg.pin_miso = 18;								// LCD_MISO     (-1 = disable)
			cfg.pin_dc = 15;								// LCD_DC       SPI (-1 = disable) Data Command control pin
			_bus_instance.config(cfg);				 		//
			_panel_instance.setBus(&_bus_instance);
		}

		{
			auto cfg = _panel_instance.config();
			cfg.pin_cs = 11;	 // LCD_CS      (-1 = disable)
			cfg.pin_rst = 13;	 // LCD_RST     (-1 = disable)
			cfg.pin_busy = -1; // BUSY        (-1 = disable)
			cfg.memory_width = 320;
			cfg.memory_height = 480;
			cfg.panel_width = 320;
			cfg.panel_height = 480;
			cfg.offset_x = 0;
			cfg.offset_y = 0;
			cfg.offset_rotation = 0;
			cfg.dummy_read_pixel = 8;
			cfg.dummy_read_bits = 1;
			cfg.readable = true;
			cfg.invert = false;
			cfg.rgb_order = false;
			cfg.dlen_16bit = false;
			cfg.bus_shared = true;
			_panel_instance.config(cfg);
		}

		{
			auto cfg = _light_instance.config();
			cfg.pin_bl = 14; // LCD_BL
			cfg.invert = false;
			cfg.freq = 44100;
			cfg.pwm_channel = 7;
			_light_instance.config(cfg);
			_panel_instance.setLight(&_light_instance);
		}

		{
			auto cfg = _touch_instance.config();
			cfg.x_min = 0;
			cfg.x_max = 319;
			cfg.y_min = 0;
			cfg.y_max = 479;
			cfg.pin_int = -1;				 // INT pin number to which is connected
			cfg.bus_shared = true;	 // Set to true when using a common bus with the screen
			cfg.offset_rotation = 0; // Adjustment when the display and touch direction do not match Set with a value of 0 to 7
			cfg.spi_host = SPI3_HOST;
			cfg.freq = 2500000; // SPI set clock
			cfg.pin_sclk = 16;	// LCD_SCLK
			cfg.pin_mosi = 17;	// LCD_MOSI
			cfg.pin_miso = 18;	// LCD_MISO
			cfg.pin_cs = 12;		// LCD_TOUCH_CS
			_touch_instance.config(cfg);
			_panel_instance.setTouch(&_touch_instance);
		}

		setPanel(&_panel_instance);
	}
};

LGFX tft;

// Change to your screen resolution
static const uint32_t screenWidth = 480;
static const uint32_t screenHeight = 320;
static lv_disp_draw_buf_t draw_buf;

/*Static or global buffer(s). The second buffer is optional*/
static lv_color_t buf_1[screenWidth * 10];
// static lv_color_t buf_2[screenWidth * 10];

// Display flushing
void disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
	uint32_t w = (area->x2 - area->x1 + 1);
	uint32_t h = (area->y2 - area->y1 + 1);

	tft.startWrite();
	tft.setAddrWindow(area->x1, area->y1, w, h);
	// tft.pushColors( ( uint16_t * )&color_p->full, w * h, true );
	tft.writePixels((lgfx::rgb565_t *)&color_p->full, w * h);
	tft.endWrite();

	lv_disp_flush_ready(disp);
}

void passcodeAutoLock()
{
	passcodeAutoLockEnabled = true;
}

// Read the touchpad
void touchpad_read(lv_indev_drv_t *indev_driver, lv_indev_data_t *data)
{
	uint16_t touchX, touchY;
	bool touched = tft.getTouch(&touchX, &touchY);
	if (!touched)
	{
		data->state = LV_INDEV_STATE_REL;

		if (passcodeAutoLockTimerActive)
		{
			if ((lv_scr_act() == ui_homeScreen || lv_scr_act() == ui_modesScreen || lv_scr_act() == ui_stgsScreen || lv_scr_act() == ui_infoScreen) && !sessionStarted && !doorActivatedShreddingMode && passcodeSet)
			{
				if (!passcodeAutoLockTicker.active())
				{
					passcodeAutoLockTicker.once_ms(passcodeAutoLockInterval, passcodeAutoLock);
				}
			}
			else // if (lv_scr_act() == ui_bootScreen || lv_scr_act() == ui_passcodeConfirmScreen || sessionStarted || doorActivatedShreddingMode || !passcodeSet)
			{
				if (passcodeAutoLockTicker.active())
				{
					passcodeAutoLockTicker.detach();
				}
				passcodeAutoLockEnabled = false;
			}
		}
	}
	else
	{
		data->state = LV_INDEV_STATE_PR;

		// Set the coordinates
		data->point.x = touchX;
		data->point.y = touchY;

		if (passcodeAutoLockTicker.active())
		{
			passcodeAutoLockTicker.detach();
		}
	}
}

// Give feedback to the user
void feedback_audio(_lv_indev_drv_t *, uint8_t event)
{

	if (event == LV_EVENT_PRESSED)
	{
	}

	if (event == LV_EVENT_RELEASED)
	{
	}
}

void display_setup_init()
{
	//tft.setBrightness(254);
	tft.begin();
	tft.setRotation(1);

	lv_init();
	uint16_t calData[] = {3787, 3868, 3835, 200, 205, 3885, 168, 193};
	tft.setTouchCalibrate(calData);

	/*
	std::uint16_t fg = TFT_WHITE;
	std::uint16_t bg = TFT_BLACK;
	if (tft.isEPD()) std::swap(fg, bg);
	uint16_t calibrationData[8];
	tft.calibrateTouch(calibrationData, fg, bg, std::max(tft.width(), tft.height()) >> 3);

	for(int i=0; i<8; i++){
		Serial.print(calibrationData[i]);
		Serial.print(", ");
	}
	*/

	/*Initialize `disp_buf` with the buffer(s). With only one buffer use NULL instead buf_2 */
	lv_disp_draw_buf_init(&draw_buf, buf_1, NULL, screenWidth * 10);

	// Initialize the display
	static lv_disp_drv_t disp_drv;
	lv_disp_drv_init(&disp_drv);

	// Change the following line to your display resolution
	disp_drv.hor_res = screenWidth;
	disp_drv.ver_res = screenHeight;
	disp_drv.flush_cb = disp_flush;
	disp_drv.draw_buf = &draw_buf;
	lv_disp_drv_register(&disp_drv);

	// Initialize the input device driver
	static lv_indev_drv_t indev_drv;
	lv_indev_drv_init(&indev_drv);													// Basic initialization
	indev_drv.type = LV_INDEV_TYPE_POINTER;											// Touch pad is a pointer-like device
	indev_drv.read_cb = touchpad_read;												// Set your driver function
	indev_drv.feedback_cb = feedback_audio;											// Set your driver function
	lv_indev_t *disp_indev = lv_indev_drv_register(&indev_drv); 					// Register the driver in LVGL and save the created input device object

	ui_init();
}

void displayInit()
{
	lv_timer_handler();
}
```

```
CPU Freq: 240 MHz
XTAL Freq: 40 MHz
APB Freq: 80000000 Hz
Free heap memory: 252383 bytes
Free PSRAM: 2092100 bytes
[   881][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type LEDC (9) successfully set to 0x4205f158
[   893][E][esp32-hal-ledc.c:131] ledcAttachChannel(): ledc setup failed!
[   908][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type SPI_MASTER_SCK (34) successfully set to 0x42060ef8
[   921][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type SPI_MASTER_MISO (35) successfully set to 0x42060fdc
[   934][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type SPI_MASTER_MOSI (36) successfully set to 0x420610bc
[   947][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type SPI_MASTER_SS (37) successfully set to 0x420611a0
```

plystdr, 2025-05-05:
Additional output from serial.

```
[  1003][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type LEDC (9) successfully set to 0x4205ff3c
E (1898) ledc: ledc_set_timer_div(685): timer clock conflict, already is 4 but attempt to 11
[  1015][E][esp32-hal-ledc.c:131] ledcAttachChannel(): ledc setup failed!
```

plystdr, 2025-05-05:
I am also using another pwm code to control motor speed. It was working fine before the update.

But it seems it is conflicting with lovyan. When I disable my pwm control code. backlight works as usual.

Does lgfx updated how to assign ledc timer? is there any way to manually set a different timer for lgfx?

```
#include <stdio.h>
#include "driver/ledc.h"
#include "esp_err.h"

#define LEDC_TIMER              LEDC_TIMER_2
#define LEDC_MODE               LEDC_LOW_SPEED_MODE
#define LEDC_OUTPUT_IO          (MOTOR_PIN_SPD) // Define the output GPIO
#define LEDC_CHANNEL            LEDC_CHANNEL_2
#define LEDC_DUTY_RES           LEDC_TIMER_10_BIT // Set duty resolution to 13 bits
#define LEDC_DUTY               (0) // Set duty to 50%. (2 ** 13) * 50% = 4096
#define LEDC_FREQUENCY          (2000) // Frequency in Hertz. Set frequency at 4 kHz

/* Warning:
 * For ESP32, ESP32S2, ESP32S3, ESP32C3, ESP32C2, ESP32C6, ESP32H2, ESP32P4 targets,
 * when LEDC_DUTY_RES selects the maximum duty resolution (i.e. value equal to SOC_LEDC_TIMER_BIT_WIDTH),
 * 100% duty cycle is not reachable (duty cannot be set to (2 ** SOC_LEDC_TIMER_BIT_WIDTH)).
 */

static void PWMControl_Setup(void)
{
    // Prepare and then apply the LEDC PWM timer configuration
    ledc_timer_config_t ledc_timer = {
        .speed_mode       = LEDC_MODE,
        .duty_resolution  = LEDC_DUTY_RES,
        .timer_num        = LEDC_TIMER,
        .freq_hz          = LEDC_FREQUENCY,  // Set output frequency at 4 kHz
        .clk_cfg          = LEDC_AUTO_CLK
    };
    ledc_timer_config(&ledc_timer);

    // Prepare and then apply the LEDC PWM channel configuration
    ledc_channel_config_t ledc_channel = {
        .gpio_num       = LEDC_OUTPUT_IO,
        .speed_mode     = LEDC_MODE,
        .channel        = LEDC_CHANNEL,
        .intr_type      = LEDC_INTR_DISABLE,
        .timer_sel      = LEDC_TIMER,
        .duty           = 0
    };
    ledc_channel_config(&ledc_channel);

    ledc_fade_func_install(0);
}
```

tobozo, 2025-05-05:
hi,

> is there any way to manually set a different timer for lgfx?

timer selection is based on the value of `_cfg.pwm_channel`

```cpp
    ledc_channel.timer_sel  = (ledc_timer_t)((_cfg.pwm_channel >> 1) & 3);
```

so changing the value of `cfg.pwm_channel` to a different value (0...7), or just commenting out its declaration (default pwm_channel = 0, timer sel = 0) should do it

tobozo, 2025-05-05:
another [suggested solution](https://github.com/espressif/esp-idf/issues/13769#issuecomment-2103777186) is to use `LEDC_USE_XTAL_CLK` instead of `LEDC_AUTO_CLK` in your `ledc_channel_config_t` :

```c
   .clk_cfg = LEDC_USE_XTAL_CLK,
```

plystdr, 2025-05-05:
> another [suggested solution](https://github.com/espressif/esp-idf/issues/13769#issuecomment-2103777186) is to use `LEDC_USE_XTAL_CLK` instead of `LEDC_AUTO_CLK` in your `ledc_channel_config_t` :
> 
>    .clk_cfg = LEDC_USE_XTAL_CLK,

Thanks!

`.clk_cfg          = LEDC_USE_XTAL_CLK`

Solved the issue! Closing this.

closed 2025-05-05 by plystdr