lovyan03/LovyanGFX #709: Compile fails with Arduino\libraries\LovyanGFX\src\lgfx\v1\platforms\esp32s3\Bus_RGB.cpp:98:5: error: 'gpio_hal_iomux_func_sel' was not declared in this scope; did you mean 'gpio_hal_func_sel'?

Westcott1, 2025-05-07:
## Environment ( 実行環境 )

- MCU or Board name:  ESP32-S3 + Waveshare 7" panel
- Panel Driver IC:   Panel_RGB
- Bus type:           Bus_RGB
- LovyanGFX version:  1.2.7
- FrameWork version:   ArduinoESP32-S3 v3.3.0-alpha1
- Build Environment:   ArduinoIDE 2.3.5
- Operating System:   Windows 11

## Problem Description ( 問題の内容 )
I have been using this panel for over a year without any problems.
Now the compile fails with
Arduino\libraries\LovyanGFX\src\lgfx\v1\platforms\esp32s3\Bus_RGB.cpp:98:5: error: 'gpio_hal_iomux_func_sel' was not declared in this scope; did you mean 'gpio_hal_func_sel'?

### Expected Behavior ( 期待される動作 )
Compile correctly

### Actual Behavior ( 実際の動作 )
Compile fails with
Arduino\libraries\LovyanGFX\src\lgfx\v1\platforms\esp32s3\Bus_RGB.cpp:98:5: error: 'gpio_hal_iomux_func_sel' was not declared in this scope; did you mean 'gpio_hal_func_sel'?

tobozo, 2025-05-08:
hi,

alpha and beta versions arduino-esp32 aren't supported by LovyanGFX, and esp-idf v5.5 isn't yet in espressif support matrix, so I can't push a commit yet

however end of idf v5.0 support happends at the end of this month, with probable start of idf v5.5 support

[image]

so if you're okay with a little testing, here's a suggested solution:

`lgfx/v1/platforms/esp32s3/Bus_RGB.cpp`

```cpp
#if __has_include(<esp_idf_version.h>)
 #include <esp_idf_version.h>
 #if ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 5, 0)
  #define GPIO_FUNC_SEL(pin) gpio_func_sel(pin, PIN_FUNC_GPIO)
 #else
  #define GPIO_FUNC_SEL(pin) gpio_hal_iomux_func_sel(GPIO_PIN_MUX_REG[pin], PIN_FUNC_GPIO)
 #endif
#endif

namespace lgfx
{
 inline namespace v1
 {

  // ....

  static void _gpio_pin_sig(uint32_t pin, uint32_t sig)
  {
    GPIO_FUNC_SEL(pin);
    gpio_set_direction((gpio_num_t)pin, GPIO_MODE_OUTPUT);
    esp_rom_gpio_connect_out_signal(pin, sig, false, false);
  }

  // ....

```

Westcott1, 2025-05-08:
Thanks!
Reverted to 3.2.0 and it compiles OK.