lovyan03/LovyanGFX #710: ESP32S3 - ST7796, wrong color with the same code, different number of pixels

tanmanh0707, 2025-05-10:
## Environment ( 実行環境 )

- MCU or Board name: ESP32-S3-N16R8
- Panel Driver IC: ST7796 - 320x480
- Bus type: Parallel8
- LovyanGFX version: 1.2.7
- FrameWork version: framework-arduinoespressif32 @ 3.20017.0 (2.0.17)
- Build Environment: PlatformIO - Core 6.1.18 - Home 3.4.4
- Operating System:  Windows

## Problem Description ( 問題の内容 )
This is the minimal block of code that describes my issue.
I draw 2 lines with different number of pixels (1st line has 131 pixels and 2nd line has 130 pixels), the same color code, and the lcd shows 2 lines with different colors.

### Expected Behavior ( 期待される動作 )
It should show 2 lines with the same color

### Actual Behavior ( 実際の動作 )
It shows different colors with the same code.

### Steps to reproduce ( 再現のための前提条件 )

1. I prepare a uint16_t array which is filled with TFT_RED color code.
2. I use **setAddrWindow** w = **131**, h = 1 (one line), **pushPixels** with len = **131**, the lcd shows a BLUE line
3. I use **setAddrWindow** w = **130**, h = 1 (one line), **pushPixels** with len = **130**, the lcd shows a RED line
4. I draw a 50x50 red square to check the color, it shows correct red square

[image]

### Code to reproduce this issue ( 再現させるためのコード )

[GIST](https://gist.github.com/tanmanh0707/762930d1f031cb6d8af402cd0d04e723)

tanmanh0707, 2025-05-30:
> hi,
> 
> does it also occur with drawLine() or pushImage() ?
> 
> _tft.pushImage(0, 0, 131, 1, (const uint16_t *)pBitmap);
> _tft.pushImage(0, 0, 130, 1, (const uint16_t *)pBitmap);
> what if you invert (even first, odd last), do the colors get inverted too?
> 
> _tft.pushImage(0, 0, 130, 1, (const uint16_t *)pBitmap);
> _tft.pushImage(0, 0, 131, 1, (const uint16_t *)pBitmap);

Hi, thank you for your time. There is one point, the picture in the #1 post, you can see the first pixel of second line gets blue.
Below is my testing:

```
  _tft.setColor(TFT_RED);
  _tft.drawLine(0, 0, 131, 1);

  _tft.setColor(TFT_RED);
  _tft.drawLine(0, 10, 130, 11);
```
It shows same color but there is a gap, i don't know why

[image]

```
  _tft.setColor(TFT_RED);
  _tft.fillRect(0, 0, 131, 1);

  _tft.setColor(TFT_RED);
  _tft.fillRect(0, 10, 130, 1);
```
fillRect is working perfectly !!!

```
_tft.pushImage(0, 0, 131, 1, (const uint16_t *)pBitmap);
_tft.pushImage(0, 10, 130, 1, (const uint16_t *)pBitmap);
```
=> Same issue

```
  _tft.pushImage(0, 0, 130, 1, (const uint16_t *)pBitmap);
  _tft.pushImage(0, 10, 131, 1, (const uint16_t *)pBitmap);
```
if I invert, the colors get inverted too. There are also the first 2 pixels get inverted.

[image]

tobozo, 2025-06-02:
Can you confirm you have witnessed a working reference e.g. arduino_gfx, adafruit or other gfx library example sketch drawing lines without showing those symptoms? That will eliminate the hardware failure from the probable causes.

Auditing that bug as if it was a feature may yield new behaviours, let's try random stuff.

What happens when:
  - x:y coordinates are odd
  - tft.setRotation() is applied with a different value
  - tft.setBrightness() is set as low as possible 
  - try these instead of calloc(): `lgfx::heap_alloc_dma(PIXELS_NUM*2)`, `lgfx::heap_alloc_psram(PIXELS_NUM*2)`
  - use `tft.begin()` instead of `tft.init(); tft.fillScreen(TFT_BLACK);`

It would also be interesting to know more about the wiring: is the display connected to the ESP32 using dupont cables, a ribbon cable or a daughterboard?

If these tests don't yield new symptoms, the next thing to check will be the panel init sequence e.g. compare [Panel_ST7796 commands](https://github.com/lovyan03/LovyanGFX/blob/master/src/lgfx/v1/panel/Panel_ST7796.hpp#L40-L87) with those from [lcdwiki](http://www.lcdwiki.com/res/MSP4030_MSP4031/ST7796_Init.txt), maybe gate settings (`0xb6`) and/or power controls (`0xC1, 0xC2`) need some tweaking.

tobozo, 2025-06-02:
I converted TFT_eSPI init sequence, apparently it's using different values for pixel format and DFC, order of the commands is different too, some are repeated twice.

You can try it by replacing [this code block](https://github.com/lovyan03/LovyanGFX/blob/5438181440c71cf30bbdc347b0b1597ae3ebf77d/src/lgfx/v1/panel/Panel_ST7796.hpp#L58-L89) with this:

```cpp
      static constexpr uint8_t list0[] = {
          CMD_CSCON,  1, 0xC3, //Command Set control -> Enable extension command 2 partI
          CMD_CSCON,  1, 0x96, //Command Set control -> Enable extension command 2 partII
          CMD_MADCTL, 1, 0x48, //Memory Data Access Control MX, MY, RGB mode -> X-Mirror, Top-Left to right-Buttom, RGB
          CMD_COLMOD, 1, 0x55, //Interface Pixel Format -> Control interface color format set to 16
          CMD_INVCTR, 1, 0x01, //Column inversion -> 1-dot inversion
          CMD_DFUNCTR, 3, 0x80, 0x02, 0x3B, //Display Function Control -> Bypass, Source Output Scan from S1 to S960, Gate Output scan from G1 to G480, scan cycle=2, LCD Drive Line=8*(59+1)
          CMD_DOCA, 8,  //Display Output Ctrl Adjust
            0x40, 0x8A, 0x00, 0x00,
            0x29,    //Source eqaulizing period time= 22.5 us
            0x19,    //Timing for "Gate start"=25 (Tclk)
            0xA5,    //Timing for "Gate End"=37 (Tclk), Gate driver EQ function ON
            0x33,

          CMD_PWCTR2, 1, 0x06, //Power control2 -> VAP(GVDD)=3.85+( vcom+vcom offset), VAN(GVCL)=-3.85+( vcom+vcom offset)
          CMD_PWCTR3, 1, 0xA7, //Power control 3 -> Source driving current level=low, Gamma driving current level=High

          CMD_VMCTR, 1+CMD_INIT_DELAY, 0x18, //VCOM Control -> VCOM=0.9

          //ST7796 Gamma Sequence
          CMD_GMCTRP1, 14,
            0xF0, 0x09, 0x0b, 0x06, 0x04, 0x15, 0x2F, 0x54, 0x42, 0x3C, 0x17, 0x14, 0x18, 0x1B, //Gamma"+"
          CMD_GMCTRN1, 14+CMD_INIT_DELAY,
            0xE0, 0x09, 0x0B, 0x06, 0x04, 0x03, 0x2B, 0x43, 0x42, 0x3B, 0x16, 0x14, 0x17, 0x1B, //Gamma"-"

          CMD_CSCON, 1, 0x3C, //Command Set control -> Disable extension command 2 partI
          CMD_CSCON, 1, 0x69, //Command Set control -> Disable extension command 2 partII

          CMD_SLPOUT, 0+CMD_INIT_DELAY, 130,    // Exit sleep mode
          CMD_IDMOFF, 0,
          CMD_DISPON, 0,
          0xFF,0xFF, // end
      };
```

tanmanh0707, 2025-06-02:
Hi, unfortunately, when I replaced that code block, the LCD will not be initialized properly, the lcd turns all White and cannot be controlled anymore.

Then I compare the original code and add 2 more value 120 as below picture, the screen is working again.
However, the issue is still there. Also, I noticed that the x coordinate seems to be inverted.

[image]

[image]

tobozo, 2025-06-02:
this value is set to `0x22` in the original init sequence, but set to `0x02` in TFT_eSPI version

[image]

and there's `CMD_COLMOD, 1, 0x55,` where the color format should be set to rgb565 for both the rgb interface and the control interface (=0x55)

[image]