lovyan03/LovyanGFX #711: Inconsistent rotation direction between displays and touch screens

MitchBradley, 2025-05-25:
## Environment ( 実行環境 )

- MCU or Board name:  ESP-WROOM-32 on 2432 2.8" Cheap Yellow Display
- Panel Driver IC:    ILI9341_2 and ST7789
- Touch Driver ICs: CST816S and XPT2046
- Bus type:           SPI (panel and XPT2046) and I2C (CST816S)
- LovyanGFX version:  latest develop
- FrameWork version:   ArduinoESP32 v2.0.3
- Build Environment:   PlatformIO
- Operating System:   Windows

## Problem Description ( 問題の内容 )

I am developing an Open Source product that uses a 2432 2.8" "Cheap Yellow Display".  It supports both a resistive version that uses ST7789 panel with XPT2046 touch and a capacitive version that uses ILI9341_2 panel with CST816S touch.

The user can select one of four screen rotations at run time so the enclosure can be used in different orientations.  This is accomplished by calling display.setRotation(N) with values of N from 0 to 3.

It works correctly with the ST7789/XPT2046 board, but there is a rotation direction problem with the ILI9341_2/CST816S board.  The problem is that the direction of rotation is different between the panel and the touch.  With setRotation(1), the image on the ILI9341_2 panel rotates in the anti-clockwise direction relative to rotation 0, while the touch coordinates rotate clockwise, so the touch point from convertRawXY does not match the image orientation.  Even numbered rotations (0 and 2) work correctly, but odd numbered rotations (1 and 3) do not.

With the ST7789/XPT2046 board, both the panel image and the touch coordinates correctly rotate in the clockwise direction as the argument to setRotation() increases by 1.

This could be related to the fact that the ILI9341_2 panel must be mirrored for right-reading text.  The initial values of offset_rotation are:   ST7789: 0  XPT2046: 2   ILI9341_2: 6  CST816S: 6.

### Expected Behavior ( 期待される動作 )

I expect that setRotation() should change both the panel image and the touch coordinate system in a corresponding manner.

### Actual Behavior ( 実際の動作 )

The ST7789/XPT2046 board behaves as expected, but the ILI9341_2/CST816S board fails under setRotation(1) and setRotation(3), with the touch coordinate system rotated 180 degrees relative to the panel coordinate system.

### Steps to reproduce ( 再現のための前提条件 )

1. Configure an ILI9341_2 + CST816S board - for example Guition JC2432W328C - with offset_rotation = 6 for both the panel and the touch.
2. With setRotation(0), observe that the panel and touch coordinate systems match.
3. With setRotation(1), observe that the panel coordinate system does not match the touch coordinate system.  The panel system is 90 degrees anti-clockwise relative to rotation 0, while the touch system is 90 degrees clockwise relative to rotation 0.

### Code to reproduce this issue ( 再現させるためのコード )

This patch fixes the problem for me.

```patch
modified   src/lgfx/v1/panel/Panel_Device.cpp
@@ -464,7 +464,16 @@ namespace lgfx
     auto r = _internal_rotation;
     if (_touch) {
       auto offset = _touch->config().offset_rotation;
-      r = ((r + offset) & 3) | ((r & 4) ^ (offset & 4));
+
+      // If the panel is mirrored, successive values of the argument to
+      // setRotation() cause anti-clockwise rotation.  We rotate the
+      // touch coordinate system accordingly.
+      uint8_t panel_rotation = r & 3;
+      if ((r & 4) && (r & 1)) {
+          panel_rotation = 4 - panel_rotation;
+      }
+
+      r = ((panel_rotation + offset) & 3) | ((r & 4) ^ (offset & 4));
     }
     bool vflip = (1 << r) & 0b10010110; // r 1,2,4,7

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

lovyan03, 2025-05-25:
Rotating clockwise is correct.
Someone other than me created ILI9341_2, so I don't know the configuration of ILI9341_2.

MitchBradley, 2025-05-25:
So the correct fix would be to modify the ILI9341_2 driver to rotate clockwise?  If so, I will  investigate how to do that?

MitchBradley, 2025-05-25:
I suspect that the problem is in madctl_table[].  If I exchange lines 5 and 7 of that table, the ILI9341_2 rotation is clockwise and touch works without modifying convertRawXY.  That only affects LCDs with mirroring enabled.  The only mirrored LCD in LGFX_AutoDetect_ESP32_all.hpp is Panel_ILI9342_2 for Guition_JC2432W328C, which I added in an earlier PR.

I will submit a PR.

lovyan03, 2025-05-26:
The text is long and I'm not sure what you're saying, but I apologize for possibly misunderstanding what needed to be fixed.
What is the problem in the first place?

MitchBradley, 2025-05-26:
I solved the problem.  It turns out that the JC2432W328C board has an ST7789 panel, not an ILI9341 panel.  I was confused because the Sunton_2432S028 (ST7789) driver in LGFX_AutoDetect_ESP32_all.hpp did not work with the board, but an ILI9341_2-based driver did (mostly) work.  The reason why the the ST7789 driver did not work was because the JC2432W328C board has the backlight on GPIO27, whereas the Sunton board has it on GPIO21.

The Guition board works correctly with the correct backlight pin and the correct touch chip (CST816S).  Mirroring is unnecessary.

The cause of the mirroring is interesting.  When the ILI9341_2 driver is used with the ST7789 chip, the driver init sequence uses command 0xC0 to try and set the gamma VDD.  But on ST7789, command 0xC0 does not control gamma, but instead has bits that invert display directions and colors!

closed 2025-05-26 by MitchBradley