lovyan03/LovyanGFX #713: When compile with Arduino-esp32 and run  in esp32 will panic

NewNoob2002, 2025-06-02:
**Carefully written Issues are more likely to be given priority.**
**丁寧に記述された報告は優先して対応される可能性が高くなります。**

## Environment ( 実行環境 )

- MCU or Board name:  [ESP-32-S3-WROON-1]
- Panel Driver IC:    [ST7789]
- Bus type:           [SPI]
- LovyanGFX version:  [1.2.7]
- FrameWork version:   [ArduinoESP32 v3.2.0|ESP-IDF v5.4.1]
- Build Environment:   [ESP-IDF]
- Operating System:   [Linux]

## Problem Description ( 問題の内容 )
When espidf use Arduino-esp32 as component, the project will painc and halt, but compile just with esp-idf is ok
### Expected Behavior ( 期待される動作 )
both can work
### Actual Behavior ( 実際の動作 )

### Steps to reproduce ( 再現のための前提条件 )

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4200ad5e  PS      : 0x00060730  A0      : 0x8200a928  A1      : 0x3fcc9480  
--- 0x4200ad5e: lgfx::v1::LGFXBase::initDMA() at /home/gtc/Workfolder/lcd_lvgl_s3/components/LovyanGFX/src/lgfx/v1/LGFXBase.hpp:328
---  (inlined by) HAL::Display_Init() at /home/gtc/Workfolder/lcd_lvgl_s3/main/src/HAL_Display.cpp:11

A2      : 0x00000032  A3      : 0x3c065190  A4      : 0x3c06517c  A5      : 0x00000009  
A6      : 0x3c065174  A7      : 0x3fc97fa0  A8      : 0x8200ad5c  A9      : 0x3fcc9460  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x3c065254  A13     : 0x00000008  
A14     : 0x00000000  A15     : 0x3fcc9460  SAR     : 0x00000016  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  
--- 0x40056f5c: memcpy in ROM
--- 0x40056f72: memcpy in ROM

Backtrace: 0x4200ad5b:0x3fcc9480 0x4200a925:0x3fcc94a0 0x4200a8e7:0x3fcc94c0 0x42051a78:0x3fcc94e0 0x4037be76:0x3fcc9510
--- 0x4200ad5b: HAL::Display_Init() at /home/gtc/Workfolder/lcd_lvgl_s3/main/src/HAL_Display.cpp:10
--- 0x4200a925: HAL::HAL_Init() at /home/gtc/Workfolder/lcd_lvgl_s3/main/src/HAL.cpp:11
--- 0x4200a8e7: app_main at /home/gtc/Workfolder/lcd_lvgl_s3/main/main.cpp:45
--- 0x42051a78: main_task at /home/gtc/esp/v5.4.1/esp-idf/components/freertos/app_startup.c:208
--- 0x4037be76: vPortTaskWrapper at /home/gtc/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: 7f99461c1

CPU halted.

[image]

[image]

### Code to reproduce this issue ( 再現させるためのコード )

**Please submit complete source code that can reproduce your problem.**
**あなたの問題を再現できる完全なソースコードを提示してください。**

```cpp
extern "C" void app_main() {
  HAL::HAL_Init();
  lv_init();
  lv_port_init();

  ui_init();
}
void Display_Init(void) {
  // init display
  DISPLAY_PRINTF("Init Display...\n");
  lcd.init();
  lcd.setRotation(1);
  lcd.initDMA();
  lcd.fillScreen(TFT_BLACK);
  lcd.setBrightness(255);
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

tobozo, 2025-06-02:
hi,

what happens if you comment this line out:

```cpp
  lcd.initDMA();
```

NewNoob2002, 2025-06-03:
> have you tried to run cpp code outside `extern C` ?

it need extern C in espidf, because the app_start task in c file, comment extern "C" will can't link

[image]

tobozo, 2025-06-03:
sorry I realized my previous question was stupid :)

Let's reduce the problem to a simple "hello world" sketch with `lcd.init(); lcd.fillScreen(TFT_BLACK); lcd.print("Hello World");` and the use of [Option 1. Using Arduino setup() and loop()](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html#option-1-using-arduino-setup-and-loop) instead of `app_main()`.

jifengshandian, 2025-06-29:
Hello friend, have you solved it? I have the same problem.

[image]

NewNoob2002, 2025-07-01:
> Hello friend, have you solved it? I have the same problem.
> 
> <img alt="Image" width="765" src="https://private-user-images.githubusercontent.com/62104564/460287179-e45116ce-3ca0-483d-9880-dbab9e6aecc5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTEzNTIyMzgsIm5iZiI6MTc1MTM1MTkzOCwicGF0aCI6Ii82MjEwNDU2NC80NjAyODcxNzktZTQ1MTE2Y2UtM2NhMC00ODNkLTk4ODAtZGJhYjllNmFlY2M1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzAxVDA2Mzg1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY2NzBiZTc3ZjdkMzdjNjRhODIwNGMzOWJkZDRhMDE0MjUzNjU3MWY1MjNiMjA1YzlhMTIxNTdiOTA2MTFiMDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.eGb3kQec2Wam0gfaibs21QAuulEGbrPREhHIZZRO2j0">

hello, not yet, but i try another GFX Library, it's ok, this link: https://github.com/moononournation/Arduino_GFX.git

tobozo, 2025-07-01:
@NewNoob2002 please share the source for your `LGFX_ST7789` lcd object, not a screenshot but actual code