lovyan03/LovyanGFX #714: Error with ESP32 3.3.0 alpha1

jaley95630, 2025-06-05:
**Carefully written Issues are more likely to be given priority.**

## Environment

- MCU or Board name:  LilyGo T-Display S3 (ESP32-S3)
- Panel Driver IC:    ST7789
- Bus type:           Parallel8
- LovyanGFX version:  1.2.7
- FrameWork version:   ESP32 by Espressif Systems 3.3.0-alpha1
- Build Environment:   ArduinoIDE - pioarduino compiles but doesn't work (perhaps different issue)
- Operating System:   Windows11

## Problem Description
"Hello World" fails to compile under 3.3.0-alpha1.  Works properly under 3.2.0.

### Expected Behavior
Successful compile.
Displays test both on Serial Monitor and on TFT.

### Actual Behavior
Does not compile.  Error message:
c:\Users\XXXX\OneDrive\Documents\Arduino\libraries\LovyanGFX\src\lgfx\v1\platforms\esp32s3\Bus_RGB.cpp: In function 'void lgfx::v1::_gpio_pin_sig(uint32_t, uint32_t)':
c:\Users\XXXX\OneDrive\Documents\Arduino\libraries\LovyanGFX\src\lgfx\v1\platforms\esp32s3\Bus_RGB.cpp:98:5: error: 'gpio_hal_iomux_func_sel' was not declared in this scope; did you mean 'gpio_hal_func_sel'?
   98 |     gpio_hal_iomux_func_sel(GPIO_PIN_MUX_REG[pin], PIN_FUNC_GPIO);
      |     ^~~~~~~~~~~~~~~~~~~~~~~
      |     gpio_hal_func_sel
exit status 1

Compilation error: exit status 1

### Steps to reproduce

1. Update board to "latest", which at the time of this writing is 3.3.0 alpha 1.
2. Compile the code.

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

**Please submit complete source code that can reproduce your problem.**
```cpp
#include <LovyanGFX.hpp>       // Include the graphics library
#include <lgfx_user/LGFX_T-DisplayS3.hpp>
#include <LGFX_TFT_eSPI.hpp> // makes it TFT_eSPI compatible??

LGFX tft;  // Create object "tft"

void setup() {
  Serial.begin(115200);

  /* Initialize TFT Display, and set non-default items (rotation, font, etc.) */
  tft.init();
}

void loop() {
  Serial.println("Hello World!  Hello Serial!");
  tft.println("Hello World!  Hello LGFX_TFT!");
  delay(5000);
}
```

tobozo, 2025-06-05:
hi,

LovyanGFX doesn't support espressif alpha releases, too many things are unstable, unfixed or half implemented. 
Espressif alpha releases are usually for core developers, unless you were looking for a specific feature and/or you're involved in the testing, it doesn't make sense to use it.

Let's keep this issue open until 3.3.0-rc1, until then you can just rename the function call as the error message suggests.

mhaberler, 2025-06-12:
patching it up like so worked for me:

// Replace:
gpio_hal_iomux_func_sel(GPIO_PIN_MUX_REG[pin], PIN_FUNC_GPIO);
// With:
gpio_iomux_out(pin, PIN_FUNC_GPIO, false);