lovyan03/LovyanGFX #730: RP2040 8bit Para Mode support

XueNianOfficial, 2025-07-13:
Request to Use the PIO Function of the RP2040 to Drive an 8-Bit Parallel Port Screen (ST7789)

[image]

ST7789 2.4 inch display

[ST7789V_SPEC_V1.0.pdf](https://github.com/user-attachments/files/21203087/ST7789V_SPEC_V1.0.pdf)

[T240B8-C24-09-V1+.pdf](https://github.com/user-attachments/files/21203094/T240B8-C24-09-V1%2B.pdf)

https://e.tb.cn/h.hiw8oQtUuTptGDg?tk=elxk4YvCs55

tobozo, 2025-07-13:
Hi,

After a quick research it appears that rp2040 will need to use the PIO interface for parallel communications instead of SPI interface.

I can guide you through the creation of a new bus type in LovyanGFX, but this also requires an acute knowledge for pico sdk low level functions, which I don't have.

Let me know if you're willing to get involved.

XueNianOfficial, 2025-07-14:
I'm sorry, I'm just a senior in high school, and obviously I don't have the time and energy to do this job.